jgyates/genmon #721: Monitor Colors not changing with Status

in2hockey04, 2022-06-10:
[image]

We had our modem/router fail, after replacing and setting ip/mac address reservation for both the serial server and the genpi (to what they were previously) the colors on the interface no longer change with the status of the generator. The buttons will stay that mustard/yellow color (as shown in the attached screenshot) regardless of the generator state. No longer is it Green (when good) or Red (when bad) or Blue (when running). If I turn off communication to the serial server then the buttons will turn back to green; which really doesn't make sense since it's technically in an error state when there's no communication between the genpi and the serial server. This isn't a big deal (at least it doesn't seem to be) because everything seems to be working properly. This is a fantastic product you have and I'm grateful for all the hard work going into this!! Thanks for any and all help!
Model: G0062770
Serial: 3004333374

jgyates, 2022-06-10:
This likely is due to your service being due. If your service currently due (service due alarm) or is due within the next 24 hours then the base state of the generator is set to SERVICEDUE. This state turns the UI yellow. Look on the Maintenance page at your service due times and see if they are currently due or within 24 hours of being due.

Let me know how it goes.

jgyates, 2022-06-10:
FYI, yellow is the color for service due. If this is not the case you can send me your logs and I can see if there are any issues.

jgyates, 2022-06-10:
I got your logs. I loaded your registers up on my simulator and I also get the yellow. It looks like your service is due in one more run hour. You should see this on the Maintenance page:

[image]

The UI will turn yellow if your service is due or if the service is due within one run hour.

in2hockey04, 2022-06-10:
Ah... I missed that I was looking at the date and not the hours. I'll have to check how to reset the service since it's been done. Probably need to do that at the controls I'd image. Thank you!

jgyates, 2022-06-10:
No problem.

closed 2022-06-10 by jgyates

jgyates, 2022-06-10:
You can reset the maintenance from the dealer menu. To access Dealer Menu on the Evolution controller, from the top menu selection (SYSTEM, DATE/TIME,BATTERY, SUB-MENUS) enter (for air cooled models) UP UP ESC DOWN UP ESC UP, then go to the dealer menu and press enter. **For liquid cooled models a level 2 dealer code can be entered, ESC UP UP DOWN DOWN ESC ESC, then navigate to the dealer menu and press enter.** Passcode for Nexus controller is ESC, UP, UP ESC, DOWN, UP, ESC, UP, UP, ENTER.

in2hockey04, 2022-06-10:
Thanks, I was literally just looking through my docs for the super-secret access. lol

jgyates, 2022-06-22:
I added this to the Wiki entry for tips and tricks: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#access-the-dealer-menu