jgyates/genmon #722: Wi-Fi signal strength dissapeared

griffths, 2022-06-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Wi-Fi signal strength gauge was reading 0db and was missing completely from monitor page.  I restarted Genmon and now the gauge has dissipeared from status screen

### Actual Behavior

Gauge is missing

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: QT02224ANAX
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.11

jgyates, 2022-06-15:
if you log into your pi, from the command line, what is the output of this command:

     ip link

and this command

      iw wlan0 link

griffths, 2022-06-15:
sorry, yes is active

griffths, 2022-06-15:
Something else odd...

I thought that there were buttons within the Genmon interface somewhere to restart Genmon and reboot the Pi but maybe that's just my old age active up again.

If there are buttons then please tell me where to look for them as I have been poking around for a while and haven't seen them.

jgyates, 2022-06-15:
you only showed one of the commands I requested:

what is the output of this command

      ip link

The restart, reboot buttons are on the advance page. Double click the gear icon in the upper right of the web interface to access the advanced settings.

genmon gets the signal strength from the "iw" command so this is why the gauge is not working. The "ip link" command will show all network adapters.

griffths, 2022-06-15:
Darn,

I am usually better at following directions...
'

[image]

griffths, 2022-06-15:
OK,

Found the "missing buttons" 

Will wait before reloading Genmon and/or restarting Pi until instructed.

jgyates, 2022-06-15:
reloading genmon will not fix this issue. A reboot of your pi might.

griffths, 2022-06-15:
We are in the second day of a heat wave.  CPU temp is around 150 Fahrenheit

griffths, 2022-06-15:
That restored the missing gauge and missing WLAN stats on the monitor page.

Sorry to bother you.  Should have restarted Pi before opening issue but convinced myself something else was going on when I couldn't locate reboot button in Genmon.

jgyates, 2022-06-15:
no worries. I am glad it is working now. Probably a glitch in the hardware or OS software due to the heat. Stay cool

closed 2022-06-15 by jgyates