jgyates/genmon #724: Genmon + MobileLink Cellular modem?

firestormv1, 2022-06-23:
I had forgotten that part of the install contract for the generator included MobileLink 4G/LTE (Generac part number 71690) .  They just came to install it today however it appears that the cellular device does not use RS485, but uses the RS232 port for communications.  I'm currently on the fence about keeping it, I much rather like the Genmon monitor (+prometheus and grafana) opposed to their proprietary solution.  Unfortunately the maintenance company doesn't understand Genmon and only uses MobileLink for remote diagnosis.

Can someone confirm or refute that the MobileLink cellular uses RS232?  I know the wifi is RS485-based, but I don't know if the cell modem is.  I'm also not opposed to ripping it apart with a screwdriver to find out either. :D 

Alternatively, any suggestions on a mux of sorts so I can split the serial port between Genmon and the MobileLink cellular device or convert Genmon to RS485?

jgyates, 2022-06-23:
Maybe a picture of the cable may help. This page has the pinout of the molex connector.  I have not seen the 4G versions. It would make sense that the controller could use WiFi if internet was available and 4G if there was an outage so having two simultaneous comms makes sense.

firestormv1, 2022-06-24:
Ok, so here's what I found:

Pic 1 -  Cellular on top, wifi on bottom.
Pic 2 - Cellular on left, wifi on right.
Pic 3 -  Back of Cellular MobileLink device.
Pic 4 - Bottom of Cellular MobileLink device.
Pic 5 - Detail of the four wires coming out of the back of the Cellular MobileLink device.

I'm sad to say that it looks like it uses RS232 for the Cellular MobileLink device which is incredibly disappointing. I was hoping it would use RS485 since it doesn't look like you can have both cellular and wifi enabled on the controller.  As it is, I don't think I'll be keeping the service past the initial year maintenance contract as I've gotten quite fond of having Genmon, SNMP, Nagios,  Prometheus, and Grafana keeping eyes on the generator when I'm not at home (Genmon is not exposed to the Internet, I use a VPN to access it).  The only downside is that the dealer requires cellular because even they agree that wifi is too flaky (I would have much preferred an Ethernet communicator which would hopefully have allowed me to keep MobileLink on RS485 and keep Genmon on RS232.

I'm also not sure a mux would work and I don't know if Genmon has RS485 capability. I know there's RS232 to RS485 adapters exist, but not knowing if the controller exposes the modbus connection via 485 like it does via 232, I might just be up a tree until the contract expires.

Oh well, here's counting down until the contract expires.

Thank you for the replies!

[image]
[image]
[image]
[image]
[image]

jgyates, 2022-06-24:
The controller does not support modbus over RS485. I have validated this with other installations.

Using both on RS232 is problematic as both the LTE modem and genmon are expecting to be modbus master devices. Even if you could get a mux type of situation you would likely experience modbus timeouts on one or both master devices.

closed 2022-06-24 by jgyates