jgyates/genmon #726: Genmon sending Blank e-mails

lps-rocks, 2022-06-30:
### Expected Behavior

Genmon send e-mail with contents

### Actual Behavior

E-mails from genmon are blank

### Steps to Reproduce (including precondition)

Configure e-mail notifications, test notifications, get blank emails

### Screenshot or Pictures relating to the problem (if possible)

https://i.hexhost.net/7e5676c3-1dec-49f3-8cef-6386ce1178ef.jpg

### Your Environment

- Genmon Logs: Submitted
- Genmon Version: V1.18.12

Genmon keeps throwing an error also whenever it goes to send a message:
`2022-06-22 18:32:54,328 : Error in MyConfig:ReadValue: MyMail: extend_wait: invalid literal for int() with base 10: '' : myconfig.py:94`

However, extend_wait exists as a number zero in mymail.conf - so i'm puzzled on this one...

jgyates, 2022-06-30:
Hi @lps-rocks 

Make sure you are looking at the file /etc/genmon/mymail.conf for the extend_wait parameter and not the file with the rest of the genmon files (usually in your /home/pi/genmon/conf folder). The file in the conf folder is the original copy and is not used at by genmon. The file in /etc/genmon is used by genmon. This error can be removed by either removing this entry or making it a valid integer. This error has no effect on sending mail.

I am guessing your problem is the security settings you have on your email client. You sent a screen shot but I don't know where this is from. To me, it looks like your email client is blocking the email from genmon.  What email client are you using? You can validate the email is sent correctly by looking at the sent folder on your email account. For example if you are using a gmail account to send emails from genmon, log into the gmail account in a web browser and look at the Sent folder associated with the email account. This should allow you to validate the email is being sent and the issue is with the client's security settings.

Let me know how it goes.

lps-rocks, 2022-06-30:
> Make sure you are looking at the file /etc/genmon/mymail.conf for the extend_wait parameter and not the file with the rest of the genmon files (usually in your /home/pi/genmon/conf folder). The file [...]

Thanks, modifying it there made the errors stop. 

> I am guessing your problem is the security settings you have on your email client. You sent a screen shot but I don't know where this is from. To me, it looks like your email client is blocking the [...]

I've tried viewing the body in multiple platforms, iOS, Outlook, Outlook Web, and they all show no body content in the e-mail. Clicking 'View Source' only shows headers and no body content in this instance. Other emails show both headers and body content.

jgyates, 2022-06-30:
If you do not have any errors in your /var/log/mymail.log then this is the recommended next step: You can validate the email is sent correctly by looking at the sent folder on your email account. For example if you are using a gmail account to send emails from genmon, log into the gmail account in a web browser and look at the Sent folder associated with the email account.

lps-rocks, 2022-06-30:
I'm not using an e-mail account to send, I'm using an MS Exchange server and genmon is allowed to SMTP relay through it. The recipient accounts are MS365 accounts.

lps-rocks, 2022-06-30:
Specifically, we went from version 1.18.08 to 1.18.12 and started having this problem. Looking at the commit history for mymail.py, there were many changes between those two versions. I'm going to try checking out 1.18.08 mymail and then go forward one commit at a time until the issue it shows up.

lps-rocks, 2022-06-30:
Checking out that version for just mymail.py fixed it, I'm currently iterating up through the commits till I find the one that broke the email bodies.

lps-rocks, 2022-06-30:
Found the commit that broke it, fccdd5751fb175063ba7de2c06e0adc296303c0b

lps-rocks, 2022-06-30:
E-mail bodies work with commit e2fe33c0f11c988314523d767ca671030f0e9cfb and prior.

lps-rocks, 2022-06-30:
Looks like it's the usage of as_bytes that's mangling the message most likely due to the warnings the python developers give:

https://docs.python.org/3/library/email.compat32-message.html#email.message.Message.as_bytes

> Note that this method is provided as a convenience and may not always format the message the way you want. For example, by default it does not do the mangling of lines that begin with From that is required by the unix mbox format. For more flexibility, instantiate a BytesGenerator instance and use its flatten() method directly. For example:

lps-rocks, 2022-06-30:
Yeah, there's something weird going on with using a bytes object into `session.sendmail` for the messages. As soon as I change it back to using `msg.as_string()` for that call, it works. I validated that I'm getting the same exact output on both as_string and as_bytes. This may be a python bug.

lps-rocks, 2022-06-30:
I see subsequent commits try and fix this but don't entirely do-so. I'm still investigating.

jgyates, 2022-06-30:
what version of python are you using?

lps-rocks, 2022-06-30:
3.9.2 - I found the issue. MimeMultipart and MimeText default to using a compatible policy that is designed for backwards compatibility which strips carriage returns. I'm going to open a PR. 

I do have to ask, why bother with python2 compatibility since python2 is EOL & you're importing libraries that only exist in python3?

lps-rocks, 2022-06-30:
Ah, okay. Makes sense. 

I found the issue with why the bodies are created out of spec. The default policy that `MIMEMultipart` and `MimeText` use strips the carriage returns in anticipation that `smtplib.sendmail` will add them back. `smtplib.sendmail` does this when passing the body in as a string, but as a bytes objects it's left uncorrected and the malformed components are dropped by an SMTP server.  I've corrected the policy to be SMTP compliant (it's also RFC-compliant, so it shouldn't affect any other sending mechanisms either).

closed 2022-06-30 by lps-rocks

jgyates, 2022-06-30:
Thanks for tracking this down. Great work!

PR #727 (2022-06-30): Fix E-mail bodies in python3 via SMTP
Fixes email bodies being sent directly via SMTP and conforms to RFC standards.

# Pull Request Template

## Description

Fixes #726 when sending messages via SMTP

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

## How Has This Been Tested?

Tested Locally on latest co [... cut]