jgyates/genmon #728: Error moving processed messages to imap folder

rwskinner, 2022-07-01:
Most likely a user error.

### Expected Behavior
After an email is sent, Genmon processes the command and should move the message to the processed folder.

### Actual Behavior
Message is never moved.  

mymail.log shows this.
2022-06-30 23:28:40,776 : Resetting email thread : STORE command error: BAD [b'Error in IMAP command STORE: Invalid arguments (0.001 + 0.000 secs).'] : mymail.py:407

Here is the actual path from my mail program on where I'm trying to move the message to.
/INBOX/Genmon/Processed

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)
[image]

{Please write here}

### Your Environment

- Generator Model: Generac 24kw Evo2
- Genmon Logs: Submit via the About page - Done
- Genmon Version: 1.18.12

rwskinner, 2022-07-01:
The items below fixes it on my system.
Using copy then after the deleted flag, use expunge.

                       # move the message to processed folder
                        **self.Mailbox.copy(num, self.ProcessedFolder) #rws**
                        #self.Mailbox.store(num, '+X-GM-LABELS', self.ProcessedFolder)  #add the label
                        self.Mailbox.store(num, '+FLAGS', '\\Deleted')     # this is needed to remove the original label
                        **self.Mailbox.expunge()**
                    if self.WaitForExit("EmailCommandThread", 15 ):
                        return
                except Exception as e1:
                    self.LogErrorLine("Resetting email thread : " + str(e1))
                    if self.WaitForExit("EmailCommandThread", 60 ):  # 60 sec
                        return
                    break

jgyates, 2022-07-01:
what email provider are you using? gmail or something else. It works for me with gmail. If you can send a screen cap of your IMAP folders that may help.  This is what I use with the default settings:

[image]

[image]

Note in my example above the folders are not in the inbox. In your example you appear to have separate conventions ( slash vs dot and mixed case labels). The folders specified must exist on the IMAP account so the names and path must match exactly.

To test, send an email with the subject "generator: status" will send an reply email with the generator status.

Are you getting any errors now and does genmon respond to email commands? Does the email command get processed and does it get moved to the process folder?

If you are not sending email commands to genmon then you do not need to enable this functionality. If you do want this functionality I recommend that you follow the guidelines [here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-mymailpy). At a minimum you should add an email filter that only accepts specific email commands from specific email addresses and then move the email to the incoming mail folder. Once your filter rule moves the email to the genmon incoming folder specified then genmon will process the email command.

Also, the expunge command deletes the trash folder, which is not the intent of the original code. The original intent is to move the email from the incoming folder to the processed folder. Using this approach you have a record of all commands sent in the processed folder.

rwskinner, 2022-07-01:
The mail provide I have, INBOX is the root and all the folders are subfolders below that.  In order to make any of it work, I must use "." and not "/".

The code I sent you does work perfectly.  The message is processed, the generator returns the response, the message is moved to the processed folder, then the message in genmon is deleted.

Expunge is not affecting my trash can or any other folder.  It's only expunging the messages marked to delete in the current folder selected.

My folder setup in Genmon looks like this.
INBOX.Genmon
INBOX.Genmon.Processed

Messages that are in INBOX.Genmon are properly parsed, a response is generated, it's then moved to the Inbox.Genmon.Processed folder then deleted from the Inbox.Genmon folder.

mymail.log is not showing any errors anymore.

A really good imaplib tutorial is here:  

https://coderzcolumn.com/tutorials/python/imaplib-simple-guide-to-manage-mailboxes-using-python

You system may work for Gmail, but, I was only trying to show you something that should be a little more universal.

BTW, I use Thunderbird for an email client and it also works fine with my providers imap setup.

rwskinner, 2022-07-01:
It's not Gmail.  My provider supplies the email and when I called, they say the use a hosted service called MagicMail.

The imaplib documentation I linked to has yahoo and gmail examples, and their gmail examples all work perfectly fine on our imap server.  

Again, Thunderbirds works totally fine on this server, and the imaplib examples all work perfectly fine on this server, it just didn't like the commands you were sending that I commented out.

Try what I did on your Gmail server and see if it works.  It's 2 lines of code.

 

https://www.magicmail.com/overview.php#:~:text=What%20is%20MagicMail%3F,ISP%2C%20Telco%20and%20business%20markets.

rwskinner, 2022-07-01:
Here is the output from listing folders.

Connection Object : <imaplib.IMAP4_SSL object at 0x0000017A1A27DF10>
Logging into mailbox...
Response Code : OK
Response      : Logged in

Response Code : OK
========= List of Directories =================

(\HasChildren) "." INBOX
(\HasChildren \UnMarked) "." INBOX.Genmon
(\HasNoChildren \UnMarked) "." INBOX.Genmon.Processed
(\HasNoChildren \UnMarked \Sent) "." INBOX.Sent
(\HasNoChildren \UnMarked \Drafts) "." INBOX.Drafts
(\HasNoChildren \Junk) "." INBOX.Spam
(\HasNoChildren \UnMarked \Trash) "." INBOX.Trash
>>>

jgyates, 2022-07-01:
Your updates are checked in. Thanks for tracking down the improvement.

closed 2022-07-01 by jgyates

rwskinner, 2022-07-01:
Thanks, however the invalid aurguments still exist from this line.

 self.Mailbox.store(num, '+X-GM-LABELS', self.ProcessedFolder)  #add the label

that can be changed to:
 self.Mailbox.copy(num, self.ProcessedFolder) 

That should also work fine in Gmail as well as other servers.

jgyates, 2022-07-01:
oops, try it now