jgyates/genmon #729: Set Exercise Time not savings new value

bantaj, 2022-07-01:
Unable to Set Exercise Time on maintenance page.  I found a closed ticket/bug that said this was due to a python issue but I am running the latest version as far as I can tell.

### Expected Behavior

Changing the day of week and/or time and clicking Set Exercise Time should update on the Generac and persist after refreshing the Maintenance page

### Actual Behavior

The exercise time is not updated and if I refresh the page, I see the old/original values.

### Steps to Reproduce (including precondition)

From the maintenance page, change the day of week and time
Click Set Exercise Time
Refresh the page

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generic Air Cooled
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.12

bantaj, 2022-07-01:
I had "Enhanced Exercise Time" disabled and when enabling it (and confirming it was changed with a refresh), it does not fix this issue.  I noticed it now shows the weekly/biweekly/monthly drop down but choosing a new day or time and clicking "Set Exercise Time" does not save those changes.

bantaj, 2022-07-01:
Bingo! 
Setting **uselegacysetexercise = True** fixed the issue.

Will this customization of genmon.conf be overwritten if I upgrade to future versions?

jgyates, 2022-07-01:
No, the /etc/genmon folder is not update when you upgrade to the latest version.

I am glad it is working. I updated the known issues [wiki page](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues)

closed 2022-07-01 by jgyates