jgyates/genmon #733: SMTP errors out with "STARTTLS extension not supported by server"

danodemano, 2022-07-07:
### Expected Behavior

SMTP should work

### Actual Behavior

```
2022-07-06 21:54:03,997 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-06 21:54:03,999 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-06 21:56:04,009 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-06 21:56:04,010 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-06 21:58:04,018 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-06 21:58:04,024 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-06 22:00:04,030 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-06 22:00:04,031 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
```

### Steps to Reproduce (including precondition)

Trying to test the SMTP settings fails, as does actual emails. I'm honestly not sure when it stopped working. Last email I have is from 5/18. SMTP settings are still good, I've been getting emails from other servers as of 10A today.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: Generac 22kw
- Genmon Logs: Above
- Genmon Version: V1.18.12

jgyates, 2022-07-07:
The first thing to suggest is to update your genmon to the very latest. There were some changes in the past week that ***may*** correct this issue. These changes were after the version number change to V1.18.12 so you will not see the version number update. Then let genmon restart and repost your mymail.log details.

jgyates, 2022-07-07:
The last update was dealing with SMTP relay. This is mostly what legacy port 25 is use for so this may resolve your issue.

danodemano, 2022-07-07:
Thanks for the quick reply! Just did an update and it's still doing the same thing:

```
2022-07-07 08:10:07,176 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-07 08:10:07,177 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-07 08:12:07,182 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-07 08:12:07,183 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-07 08:14:07,190 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:503
2022-07-07 08:14:07,192 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-07 08:14:17,730 : Error SMTP Init : SSL:<False>: STARTTLS extension not supported by server. : mymail.py:513
2022-07-07 08:14:17,752 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
```

jgyates, 2022-07-08:
@danodemano 

I got your logs and your email. Looks like the update did the trick. Let me know if you have any other issues or questions.

jgyates, 2022-07-08:
ok. Sorry, I guess I am getting this issue confused with someone else who sent me their logs and email. 

Based on the error message in your logs (STARTTLS extension not supported by server), if your server (or relay) does not support STARTTLS, then I would try to set the "Disable TLS" setting to On in the email settings on the Genmon setting page. Let me know how that goes.

danodemano, 2022-07-08:
All good mate, I really appreciate all the work you put into this project! 

The relay does not, it's a local server which then uses TLS outbound to my ISPs server. I set the "Disable TLS" setting and the fixed the issue. Thanks so much. And keep on being awesome!!

rwskinner, 2022-07-08:
> @danodemano
> 
> I got your logs and your email. Looks like the update did the trick. Let me know if you have any other issues or questions.

I sent logs and an email stating the update fixed the logs being rejected.  So yes, the update fixed my issue.  rwskinner

jgyates, 2022-07-08:
@danodemano glad to hear it is working.

@rwskinner , sorry for the confusion on my part.thanks for the logs and follow up.

rwskinner, 2022-07-08:
Thank you for all the hard work and quick responses!  A+++  Nice project.