jgyates/genmon #734: Modbus mapping question for custom controller

marklitwin, 2022-07-10:
Awesome software, being 3 days into the deployment it's apparent you put a ton of effort into this!

...and apologies for the question, I'm sure these get quite annoying.  

### Expected Behavior
Early in the process of trying to connect my Briggs generator to genmon.  No problem connecting via modbus and making progress with the mapping, but a bit confused how to use the custom json file.

Appears 300018 is the address for utility voltage with a .1 multiplier.  However entering the values into the file hasn't been successful and results in a modbus error in the log. 
  
[image]

### Actual Behavior
"Modbus exception"

### Steps to Reproduce (including precondition)

Tried a few values in the mask, no success assuming that is my issue.

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

- Generator Model: Briggs Fortress 040609 GC-1032 controller
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.12

jgyates, 2022-07-10:
Hi @marklitwin,

I assume the screen shot above is the values of the modbus registers? So reg 300018 (after removing the hi 3 is register 0x12) has a value of decimal 2424? Is this a correct assumption? I guess I need more context on the data you sent above.

What is the screen shot from? 
Is the above example JSON you sent from the gauge array. 

If you like you can send you modbus register definitions and that may allow me to make some suggestions. You can attach it to this thread or send it to generatormonitor.software@gmail.com

In the above JSON example you have a mask of 0F0000000 which only allows 4 bits to be used. For example if you have the hex value of 0x12345678 and applied the mask of 0x0F000000 then you would get 0x02000000. I would assume for a utility voltage, if this was a 32 bit value you would want a mask of FFFFFFFF.

Another thing to validate is that you have the register values included in the  "base_registers" list at the top of the JSON. The way the software works is this:

1) The base_registers are read in one thread. The software can read multiple bytes (in multiples of 2) from modus.
2) when the web interface parses the data read from the base registers it uses the other JSON lists to determine how to parse the data. For example the gauges list defines the gauges with a label and a modbus register from the base registers. The mask field will ignore the bits (mask is in hex) that are zero of the value read from base registers.

You have have seen this already but this is the doc for the custom controller JSON file:

https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

I can help you get this working. Hopefully once we get it going you would allow your JSON file to be added to the repository.

jgyates, 2022-07-11:
To better support this controller I have added a new option to the advanced settings page named "Use Modbus FC4 instead of FC3"

jgyates, 2022-07-13:
I am closing this thread for now however we will update this as needed as we progress on supporting this controller.

closed 2022-07-13 by jgyates

jgyates, 2022-07-28:
The Briggs and Stratton GC-1032 is now supported. More info here: https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

jgyates, 2023-07-12:
I have added the ability to allow custom controllers to be used with the SNMP add on. More info is [here](https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types#snmp-addon-program). Let me know if anyone needs any assistance in creating the JSON file to support SNMP your controller.