jgyates/genmon #736: System In Alarm : UNKNOWN ALARM: 00000072, but everything's fine.

jimison, 2022-07-13:
Greetings! Brand new setup (generator arrived in May, manufactured 11/2021, RPi and GenMon installed last week), and the only issue seems to be this unknown alarm. Please see details below, and let me know if I can send you any additional information. 

### Expected Behavior

Green Status in GenMon, as generator status is Green.

### Actual Behavior

Red Status in GenMon, with System in Alarm Switch State: UNKOWN ALARM: 00000072

### Steps to Reproduce (including precondition)

Start GenMon or allow self start after reboot. Red Status with UNKOWN ALARM: 00000072 is immediately present.

### Screenshot or Pictures relating to the problem (if possible)

[image]

[image]

### Your Environment

- Generator Model: G0072101 (24kW Air Cooled)
- Genmon Registers: Submitted 7/12/2022, 9:51pm MDT
- Genmon Logs: Submitted 7/12/2022, 9:51pm MDT
- Genmon Version: V1.18.12

jgyates, 2022-07-13:
I checked in a change that ***should*** make this alarm condition not be "UNKNOWN". Just update from the About page, once it restarts then you can clear the alarm condition at the controller.

closed 2022-07-13 by jgyates

jimison, 2022-07-13:
Status Update:

Updated Genmon, and the UNKNOWN has now changed to Overload Remove Load as you stated. However, I think there's an additional issue, as the Generator does not display that it's in alarm. The Green ready state for Auto mode on the controller is lit. 

If I hit the auxiliary shutoff switch next to the controller or behind the lid, it alarms as expected on the Controller, and Genmon reports the following:
Current Alarm: Emergency Stop
System In Alarm: 0001:00080072

10-15 seconds later, Genmon re-reports with a slight change:
Current Alarm: Emergency Stop
System In Alarm: 0001:00080034

I set the switch back to run, and the option to clear the alarm is presented. I hit escape and enter, and the alarm clears. I hit Auto, and the controller goes Green again and I'm good to go. 

But, Genmon says otherwise. Genmon reports an alarm:
Current Alarm: Overload Remove Load
System In Alarm: 0001:00090072

30 seconds later, Genmon again re-reports with a slight change:
Current Alarm: Overload Remove Load
System In Alarm: 0001:00000072

The Alarm Log on the controller only shows "Auxiliary Shutdown", unless I try to clear the alarm with the auxiliary shutdown switch still off, which then throws "Rotation Warning". This alarm never presents to the display, I'm guessing because it's a catch between the adjacent "Auxiliary Shutdown" alarms. The genmon.log file doesn't show any alarm besides "Alarm : Reg 0001 = 00000072".

[image]

jgyates, 2022-07-14:
Please send me your latest logs.

Questions:

Is "Auxiliary Shutdown" displayed on the controller alarm log or on the top level display?  It could be that 72=Auxiliary Shutdown and is not an alarm but a status.  Does the top level LCD ever say Overload Remove Load or does it say Auxiliary Shutdown when register 0001 is xxxx0072?

Do you have multiple entries for both Emergency Stop and Overload Remove Load in your alarm log? If you do not see an Overload Remove Load entry in the log every time you got a 72 code then that means 72 is not Overload Remove Load Alarm.

From the software's perspective, it monitors register 0001 for changes in the generator status. The upper half of this register is engine state and the lower half is switch state. for the switch state if there is any value other than xxxx0000, xxxx0007, xxxx0006 or xxxx0017  (auto, off, manual and two wire start), then it is an alarm and each alarm has at unique value for the lower half of register 0001. If a value other than xxxx0000,xxxx0006, xxxx0007 or xxxx0017 then it is an active alarm.

The upper part of 0001 is the engine state. 0000xxxx is ready, 0009xxxx is stopped, 0008xxxx is stopped in alarm. 

00090072 is Stopped with alarm or status 72
00080034 is stopped in alarm with alarm 34

34 is ESTOP and until this thread I have never seen and the value 72 before today. 

If you are telling me that the controller does not show an alarm condition when genmon is reading 72 then it may be that the value 72 should be a status (not an alarm) and should display "Auxiliary Shutdown" for the switch status. Can you validate this?

BTW, the alarm values of register 0001 are not the same values the software uses for the alarm log. The register 0001 values only show the current alarm. Different codes are used for the alarm log.

jgyates, 2022-07-14:
also, it is probably worth scrolling thru your alarm log on the controller and making sure it matches the alarm log in genmon. I am guessing it does but I want to validate this on your new (ish) controller.

jgyates, 2022-07-14:
I checked in a change that makes 72 a status instead of an alarm. You can update form the About page and let me know if that corrects your issues.

rwskinner, 2022-07-14:
I have basically the same machine.  Is there anything I can test here to help you out or duplicate it?  When I trip either one of the Aux shutdown switches, mine show Emergency Stop in the logs.

Mine is not running but is in Auto Standby and register 1  x0000 = 0

Switch, Engine and Alarm Status
(0001)
HEX: 00000000     DEC: 0 \| HI:LO: 0:0

Built in May 22.
Maintenance :
    Model : G0072101
    Generator Serial Number : 301142****
    Controller Detected : Evolution 2.0, Air Cooled
    Nominal RPM : 3600
    Rated kW : 24 kW
    Nominal Frequency : 60 Hz

jgyates, 2022-07-14:
Yes, to me this is the unique thing: pressing the aux stop button triggers alarms other than just Emergency Stop.

@jimison, let me know if the latest changes work for you.

jimison, 2022-07-14:
Engine: Switch State in Genmon still shows "Auxiliary Shutdown", but not sure what it's supposed to show when not running. Shouldn't it be "Auto" to match the Controller? 🤷‍♂️ 

[image]

jgyates, 2022-07-14:
@jimison 

That is good to know about the log entry for No Rotation Warning, I will try to fix this with the logs you sent.

This issue here is that the switch state register is showing a status other than auto. Modbus register 0001 shows a state. There is always one and only active state. The lower half of register is the switch state and the upper half is the engine state. Your generator controller is showing the the switch state as 72. Since this is a new installation, can you give some history and possibly a photo on the wiring of the auxiliary shutdown switch? 

Regarding the change in what you are seeing with genmon between the version. Based on what I am seeing both of the version is completely expected. in the initial version I had a 72 in register 0001 as an alarm. in the update I classify 72 in register 0001 as a switch state. Both of these show up because your controller is showing this in the switch status register in your controller.

The logs (alarms, start/stop and maintenance) take a few moments to fully populate once you restart genmon. When genmon starts it reads all of the status registers (about 80), then it reads all of the log registers (about 150) so you need over 200 modbus transactions to occur before genmon is fully displaying all of the possible information. I would guess that if you did not see this full logs before then it is because you were looking at it too early.

The way this is looking to me is that one of the following is occurring:

1) you have something strange going on with either your wiring or controller. The alarms other than Emergency Stop when pressing the Aux Stop button are unusual. Even if this is modbus status is a state and not an alarm then it should not be persistant.
2) There is a bug in the Evo2 firmware that does not clear this condition so the register 0001 status remains. I know that there is one other bug in the Evo2 firmware related to modbus status.

In theory I could make the software ignore the value of 72 but I can not guarantee that this would not cause other problems. When this register reads 72 it is impossible for the software to determine the state of the switch until this value is cleared.

When you test this, is the generator running, then you press the aux stop? or is the generator stopped when you do this? Can you give the exact steps to reproduce and possibly @rwskinner can try to duplicate?

jgyates, 2022-07-14:
I checked in an update that should fix the alarm log for No Rotation Warning.

jimison, 2022-07-15:
Upgraded, and No Rotation Warning now displays in the Alarm Log as expected and matches the Controller.

Generator is not running in any of these instances. Just switching from Auto to Off or Auto to Aux Bypass with its ESTOP alarm. 

Generator was setup initially with MobileLink. Not sure of the original firmware version, but it updated shortly after. My MobileLink app on my phone shows the current firmware version at 1.15.176. When I first connected it to Utility Power, I didn't have a battery installed in the Generator yet, so my first alarms were for low battery. Once the battery was installed, the only alarms or statuses in the MobileLink log and on the Generator's Controller are ESTOP and Switched Off or Ready To Run.

I'm using the Standard Hat with Fan and its included 1M Molex cable, ordered from PintSize.me a couple of weeks ago, on a Raspberry Pi 3b+ running Debian 11 64-bit..

[image]

[image]

I'll try to get photos of the Aux Switch wiring tomorrow.

jgyates, 2022-07-15:
It is difficult to tell exactly what your results are: "Upgraded, and No Rotation Warning now displays in the Alarm Log as expected and matches the Controller."

When you say it displays the alarm log as expected do you mean that rotation warning is displaying or not? I tested this with your registers that you sent so. Also, please make sure your communication stats on the Monitor page show at ***least*** 300 packets before looking at the logs or sending registers. Genmon needs time to read all of the registers from the controller.

jgyates, 2022-07-15:
@rwskinner, can you test the above as described by @jimison and see if you get any alarm or persistent switch state conditions? Test with the latest form genmon and let us know your firmware version on the maintenance page.

"Generator is not running in any of these instances. Just switching from Auto to Off or Auto to Aux Bypass with its ESTOP alarm."

jgyates, 2022-07-15:
ok. Thanks for the clarification. The No Rotation Warning in the alarm log makes sense in that it firmware may be "expecting" the generator is running when it tries to shut off via the aux shutoff. Ideally the firmware would realize that the generator is not running and not do anything. Let's see if @rwskinner gets the 72 status on register 0001 if he tries to duplicate.

rwskinner, 2022-07-16:
Mine seems to work properly if I tried to duplicate it properly.

Taking the generator out of Auto by pressing Off  shows correctly in GenMon.
Both remote stop switches show Aux. Shutdown Alarm on the generator and GenMon shows Stopped, Alarm, Emergency Stop.

    Alarm Log : 
        07/16/22 08:21:38 ESTOP Pressed : Alarm Code: 2800
        07/16/22 08:20:52 ESTOP Pressed : Alarm Code: 2800

    Run Log : 
        07/16/22 08:21:45 Stopped - Auto 
        07/16/22 08:20:16 Switched Off 

Taking out of Auto:
Engine : 
        Switch State : Manual
        Engine State : Off - Ready
        Battery Voltage : 12.40 V

Hitting external switch:
Current Alarm: Emergency Stop
System In Alarm: 0001:00080007

Status : 
    Engine : 
        Switch State : Off
        Engine State : Stopped in Alarm
        System In Alarm : Emergency Stop

jgyates, 2022-07-16:
@rwskinner , What is your firmware version. You can find that on the maintenance page.

rwskinner, 2022-07-16:
Sorry for not including that.  Firmware is v1.15  Hardware is v1.03

jgyates, 2022-07-16:
@jimison 

your versions are: firmware 1.15 and hardware 1.02

To me this looks like either there is a hardware difference, or you have an issue with your wiring of your controller, or a defective controller.

At this point I am not sure how to handle this value of register 0001. I could possible make an option that would ignore this value (i.e. treat 72 the same as zero), but that would solve this issue for you but possibly cause you not to get notified if condition 72 occurred for another issue on your generator. Let me know what you think.

jimison, 2022-07-17:
Do nothing - It's fixed! It was the Controller.

It occurred to me to try doing a reset of the Controller itself after installing Genmon. I had only reset it during the install and never since. Obviously I never saw the anomalous register value until I had Genmon running, so I never had a reason to try a reset before...

For reference, I pulled the 3 fuses (N2, N1, and T1) in the transfer switch, the 8-pin Molex cable to Genmon/Pi, and the 7.5 amp fuse in the Controller, and the Controller went blank, I assume due to power loss. After 10 seconds or so, I reinstalled the 7.5 amp fuse, which booted up the Controller. While it was booting, I reconnected the 8-pin Molex cable, and reinstalled the 3 fuses in the transfer switch. The Controller stated it connected to my WiFi and then asked for timezone again for some reason, so I set that, and entered the System menu option. Everything looked good on the Controller and in Genmon (I forgot to check the Registers at that time - oops), so I tripped the Aux Bypass Switch to trigger the alarm, switched it back, and cleared the alarm and entered Auto / Ready To Run. Genmon now shows 00000000 for Register 0001 and Auto for the Switch State.

SO, the 00000072 is gone, and obviously we still don't know what that translates to for a Switch State. If somehow someway I'm able to get it back, I'll try to duplicate the steps or conditions that brought it out of hiding, then report back.

Thanks so much for the help and input! I look forward to helping contribute in the future.

jgyates, 2022-07-17:
OK. I am glad it is working for you know. This is good info to know.