jgyates/genmon #737: Web UI randomly becomes unresponsive

mdetweiler, 2022-07-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Web UI is available all of the time

### Actual Behavior

Web UI randomly, but consistently becomes unresponsive.  The other services are up and running as I get the expected email alerts.

I end up restarting the system through ssh to restore web access.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: General 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: Latest available

jgyates, 2022-07-17:
I have seen an issue like this before however it was due to security related software built into the router that was causing the issue. Programs like [AIProtection](AiProtection) will make attempts to block ports that it things are harmful (e.g. not aware of). This thread has info on this: https://github.com/jgyates/genmon/issues/271. If you have an ASUS router with AIProtection then disable this. There are also some other issues with the build int DDNS software that Asus uses but the thread has more info. 

Also, I have not received your logs. The only logs I have received on the 16th were on another issue. Is your outbound email setup and working? This is required for sending the logs via the About page.

Let me know what router you are using and if you are able to send your logs. 

also, are you trying to access genmon via the internet (outside of your local network)? If yes, you way want to use a VPN or similar service like https://tailscale.com/ or https://www.cloudflare.com/products/zero-trust/access/ to access genmon instead of exposing ports on your router. This thread has more info on remote access: https://github.com/jgyates/genmon/issues/707

mdetweiler, 2022-07-18:
Thanks for the quick reply.  Sorry I didn't send the logs but you should see them now.  I run OpenWRT on my router with nothing fancy going on.  I DO ALSO HAVE PiHole running for internal DNS, but this problem was happening before that was ever setup.  In fact, I've had this issue for year I think.  I can access the pi with SSH all of the time, but the web UI doesn't respond randomly until I restart it.  I don't look at the UI that often so I don't know how long it stays up before having issues, but my guess it about a week.  I haven't done much to debug it on my own though.

mdetweiler, 2022-07-18:
I for got to answer that I have the same issue with access from inside and outside.  I would suspect something with my network, but I have other similar systems running without issues like Zoneminder and AqualinkD.  Different webservers for each.  The difference with genmon is the nonstandard port I guess.

jgyates, 2022-07-18:
I got your logs, thanks. Nothing looks unusual. Are you accessing genmon from the internet? I am assuming from your last answer that the answer is yes. [This is not recommended](https://github.com/jgyates/genmon/wiki#connectivity). The web server used is the one built into the flask libraries and is in no way built to withstand probing that takes place on the unprotected internet. Even if you are using passwords that still leaves your genmon web server open to exploit via denial of service to the internal web server. Try disabling the external web access and see if that corrects your issue. 

The only way I recommend accessing genmon remotely is VPN or one of the above services in the links I provided earlier. In theory this project could use apache but that is a learning curve I have not tackled yet.

mdetweiler, 2022-07-18:
Interesting.  I'll give it a shot to see if that fixes it and then I can work on firewall tweaks if necessary.

restauffer, 2022-07-21:
If you look at issue #324 you will see I have the same problem. I did enough testing to determine the problem only occurs when using certs for SSL. It also occurs even if outside (Internet) access is eliminated. It has happened with now three different routers. The only thing I could find to stop it was to stop using SSL and certs. That was not an acceptable solution for my instance. In the end I started a CRON job that restarts the webserver on a regular basis. This works fine with the minor annoyance being that if you are logged on and the CRON job runs you need to log on again. If you can stop using certs and turn to VPN for outside access as suggested, I think you will find the problem goes away.

jgyates, 2022-07-22:
I am going to close this issue for now but feel free to post to this thread with followup questions or information.

Also, there is a longer term fix for this that is on my radar (move to apache instead of the internal flask server) however I have not had the time to implement this as it is not a trivial change and it also effects the genmon update process.  I  need to figure out a solution to make sure the migration works with the genmon update process. Until then, if you run into an issue with the built in server the only solution I can offer is to use VPN.

closed 2022-07-22 by jgyates