jgyates/genmon #738: Genmon unable to communicate only during real outage, fine otherwise

jmActifio, 2022-07-17:
Environment:
Generator Model: 22KW
Genmon Version: 1.18.12
Physical: Pi 4 with pintsize.me HAT, powered from the generator battery via HAT's buck converter

Description:
Genmon has been working great for a long time.  I have tested it during simulated power outages by opening the service disconnect breaker at the transfer switch and everything has always worked fine.  We had our first REAL power outage last night, and the last Genmon status was that utility voltage was 25V, outage status was "yes", and engine status was in startup delay.  After that, Genmon lost ALL MODBUS COMMS WITH THE CONTROLLER.  Never got a "cranking" status.  I restarted Genmon, and it failed to connect to the controller entirely with the error "WARNING: Initilization not complete after 30 seconds, possible communication failure. Check cabling.".  The modbus log file showed continuous "Error Receiving slave packet".  The Generac WiFi module remained functional the entire time and communicating with MobileLink.  As soon as utility power came back, Genmon immediately started communicating again!

The outage happened in the middle of the night, so I couldn't do any extensive physical troubleshooting.  The following morning, I put the generator into stop mode, opened the service disconnect breaker, and proved again that Genmon worked just fine during an extended test where there was no utility power and the generator had not yet started.

Issue:
Why would a real utility failure be any different than opening the service disconnect at the transfer switch, and why would this cause Genmon to loose its connection with the controller?

grsthegreat, 2022-07-17:
is the house wifi on a battery backup system ?

jgyates, 2022-07-17:
I have not received any logs on this issue so I am assuming that your pi lost power since you got the "WARNING: Initialization not complete after 30 seconds". 

There should not be a difference in a utility power outage vs opening the breaker at the transfer switch. 

There could have been a surge on the line if there was lightening.

jmActifio, 2022-07-17:
Yes, all network is on UPS power and network comms were unaffected.  Only MODBUS was down.

There was no lightning.  The outage was caused by a car crash.  I have a Type 1 SPD on the transfer switch and a Type 2 SPD on the main panel.

I did send logs just before posting.

Still mystified.

skipfire, 2022-07-17:
Was this the only non-testing/utility outage you have had since installing?

Did you do any troubleshooting such as unplugging and replugging cables before flipping the service breaker?

The generator, and thus the HAT & Pi, don't know a difference between the car crash and you flipping the service disconnect breaker as all the sensors connect after that breaker.  Noise from the engine, output codes, etc; would all be the same in both cases.

andrewvanlagen, 2022-07-18:
Same issue after last update

jmActifio, 2022-07-18:
Yes, this was the only time I've had a real outage since installing.  No, it happened in the middle of the night so the extent of troubleshooting was limited to what could be done from bed! -:)

The ONLY thing impacted was the modbus connectivity between controller and the Pi.  And the moment utility power was restored, so was the modbus connectivity.  I realize this makes no sense as there should be zero difference between opening the service disconnect and a real outage, but given how it came back instantly with utility power it seems way too coupled to be coincidental.

@[andrewvanlagen](https://github.com/andrewvanlagen) When you say you had the same issue after the last update, do you mean the update to Genmon or controller firmware?

andrewvanlagen, 2022-07-18:
Genmon update

jgyates, 2022-07-18:
Nothing in the serial or modbus related code has changed in quite sometime. I did add one option to use modbus function 4 instead of 3 but that you have to explicitly enable via a setting on the advanced page.

@andrewvanlagen can you send your logs and give some additional data. Just looking for something to help find an issue of some sort. Did you get the error "WARNING: Initialization not complete after 30 sec" in your logs or web interface?

jmActifio, 2022-07-18:
@jgyates just to be clear, I only got the initialization error after rebooting the Pi during the outage.

I also see " Error: /dev/serial0:device reports readiness to read but returned no data" in the serial log at the same time.  I assume you have my logs.

jgyates, 2022-07-18:
ok. So your pi is not rebooting, just serial comms are out. Do any of these stats on the monitor page show active immediately after this occurs? If you could ever duplicate then this would be something to look at. Your battery voltage may drop, which could cause a power drop in the pi supply voltage.

        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK

jmActifio, 2022-07-18:
Thanks.  I will check that if I am ever able to reproduce it.  Again, I did intentionally disable the auto start and open the service disconnect, and I had no problems running off the generator's battery for quite some time with no appreciable voltage drop.  In the failure case, serial comms appear to have been lost within a second of the utility power going down.

jmActifio, 2022-07-18:
@jgyates I also checked the kernel logs for a low voltage message, which I believe is supposed to be logged and found none.  The only real telling thing I can find in all the logs is in the modbus logs, just thousands upon thousands of this:

2022-07-17 04:02:44,633 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2847
2022-07-17 04:02:44,634 : Error Receiving slave packet for register 0008
2022-07-17 04:02:45,948 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2851
2022-07-17 04:02:45,949 : Error Receiving slave packet for register 000a
2022-07-17 04:02:47,259 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2855
2022-07-17 04:02:47,259 : Error Receiving slave packet for register 000b
2022-07-17 04:02:48,571 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2859
2022-07-17 04:02:48,572 : Error Receiving slave packet for register 000e
2022-07-17 04:02:49,878 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2863
2022-07-17 04:02:49,878 : Error Receiving slave packet for register 000f
2022-07-17 04:02:51,191 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2867
2022-07-17 04:02:51,192 : Error Receiving slave packet for register 0010
2022-07-17 04:02:52,503 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2871
2022-07-17 04:02:52,504 : Error Receiving slave packet for register 0011
2022-07-17 04:02:53,819 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2875
2022-07-17 04:02:53,820 : Error Receiving slave packet for register 0012
2022-07-17 04:02:55,131 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 2879

jmActifio, 2022-07-18:
@jgyates Agree, but this burst of exception packets is exactly timed with the utility outage.  Prior to, and as soon as the utility power was restored, we see around 10-12 exception packets per hour.  During the utility outage, it went to about 2 per SECOND and appeared no modbus commands were getting through to the controller at all.  This has NOT occurred during simulated outages by opening the service disconnect breaker.  There was something different about this real outage, that I can not explain.

jgyates, 2022-07-18:
That is interesting. Hopefully this is not something new in terms of generac making things hard for third parties to read modbus. I hope this is just a bug in the Evo2 firmware. I know there are some existing modbus related anomalies with Evo2.

Another theory is that this behavior is caused by a race condition of some type in the modbus encryption interaction between genmon and the controller. This could cause it only to occur on rare occasions. 

Did you try restarting genmon when this occured or did you just restart the pi to get it working again? I may be able to add some code to detect an increase in modbus exceptions and possibly "reset" the encryption. It should be resetting every few minutes anyway by design.

jmActifio, 2022-07-18:
@jgyates I rebooted the entire Pi during the outage and it just made things worse.  Genmon could not longer even identify the controller and it was as if the controller wasn't even plugged in.  I fully expected that the cabling had somehow fallen off.  Yet as soon as utility power was restored, the serial comms came back on their own without me doing anything at all and it was if nothing ever happened.  I have tried everything I can think of to reproduce this, and have not been able to.

The last messages Genmon got from the controller before going dark was that the engine startup delay had started, outage was detected, and utility voltage was 25V.  So sometime between those messages and cranking state, Genmon lost all serial comms which nothing could restore until the utility power came back.

If you have any ideas on how to reproduce this or anything else I can try, I'm game to help.  There was a recent controller firmware push back in May, which does concern me slightly as you pointed out, that maybe Generac is trying to make our lives difficult.  But the controller would have no way of distinguishing a real outage from me opening up the service disconnect.

davisgoodman, 2022-07-18:
@jgyates @jmActifio I've been following the thread and seems the only difference is the "25V" left on the utility voltage reading when the real outage happens. It`s strange that there is still a reading of 25V when the real outage happens. Is it possible that there would be some kind of ground loop or bad ground somewhere in the path? 

Every real outage I got here, and believe me I had many, utility reading was always "0 Volts." 

My two cents....

jgyates, 2022-07-18:
@davisgoodman , good idea. That is certainly possible and a ground issue could cause the pi or the controller to have issues with comms.

skipfire, 2022-07-18:
I have had non-zero utility readings during outages in the past, usually during the earlier part of the outages.  When the lines get ripped off the poles but not completely broken I tend to see abnormal but present voltages on the utility lines.  Saturday we had an outage that did a lot of bouncing between 200 and 220v for about 15 minutes before it went out completely and it was a downstream failure from me.  A previous upstream failure left me seeing voltages between 40 and 50 for a period of time as well.  When I have seen these they lasted 15 minutes or less, not hours.  I believe this happens when there are incomplete discharges to the dirt (physical ground) from downed lines.

jmActifio, 2022-07-18:
Well, this outage was caused by a car crash, so definitely a good chance of downed lines.  I'm also not sure the 25V was what the actual final voltage reading was.  It may just been the last instantaneous snapshot Genmon got before serial comms were lost, and if it could have gotten subsequent data, it may very well have gone to zero.  If I open the service disconnect, it's most definitely zero voltage N1-N2.

The Pi is powered directly from the generator's battery and network comms are WiFi, and there are no external electrical connections beyond the standard ones.

I agree it smells like it could be a ground issue, but I have no leads on what it could be and how it would only manifest in a real outage vs opening the service disconnect breaker.  Any suggestions?

skipfire, 2022-07-18:
With the splitter, the Pi and the WiFi share a ground.  That said, I don't know how the RS485 connection for the WiFi compares to the RS232 and if the RS232 could be impacted by ground issues without the RS485 being impacted as well.

lmamakos, 2022-07-18:
I've had an outage where the line voltage measured at the generator dropped from the  usual 250VAC to ~100VAC or so, with a bit of a decline over time.  My electrical service is near the edge of the distribution infrastructure and there's only a single high-voltage conductor on the pole outside my home.   The cause of my outage was a phase-to-phase fault somewhere upstream that tripped off some protection devices on some of the phases, but leaving one or more still energized.  Somewhere between there and me, the 3 phases on those poles turned into a single phase and that fault dropped my line voltage to this weird intermediate voltage.   I called in to report that; and just a few minutes later, the voltage dropped to zero (almost as if that wasn't an intended state!)   I'm sure many 3-phase motors at some farms around me were unhappy having one or more of their phases missing..

Of course when that all first happened, the generator started up and ATS flipped over.  Happily, I could still watch WTF was going on with the utility side by way of the generator and genmon.

lmamakos, 2022-07-18:
I'd echo the thought that @jmActifio made about this being a ground problem, maybe a ground loop or lots of noise on the ground path used by the serial interface between the controller and genmon.

I was really concerned about what the ground return was for the serial interface when I was planning my genmon installation.  Do you run a ground between the controller's MODBUS interface and the Raspberry Pi?  And if you do, do you also have a ground running to the battery?  And what do you think happens when there's a power failure and the starter on the generator kicks in and now there's 10's of amps of current (and voltage drop) between the battery terminal and the ground connection within the generator's DC power bus and supply.  Is there also going to be a surge on the battery - raspberry pi - generator controller path if that's connected?

I decided that I didn't really know the answer to this question so I built a board that had galvanic isolation of the serial interface between the Raspberry Pi and the MODBUS connector on the generator controller.   I used this fancy part as an alternative to optoisolators, but something to avoid either a ground loop or having the serial port ground be via the battery.

Just a thought.  You can see what I did at https://github.com/lmamakos/RPi-pHat-Thermocouple/blob/genmon/RPi-pHat-Thermocouple.pdf at the lower left portion of the schematic.

andrewvanlagen, 2022-07-19:
here my modbus errors

2022-07-17 19:52:54,653 : Error Receiving slave packet for register 0059
2022-07-17 20:01:08,267 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57816867
2022-07-17 20:01:08,269 : Error Receiving slave packet for register 0055
2022-07-17 20:09:21,881 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57832608
2022-07-17 20:09:21,882 : Error Receiving slave packet for register 0055
2022-07-17 20:17:35,510 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57848354
2022-07-17 20:17:35,511 : Error Receiving slave packet for register 005a
2022-07-17 20:25:49,159 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57864097
2022-07-17 20:25:49,160 : Error Receiving slave packet for register 0001
2022-07-17 20:34:02,790 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57879838
2022-07-17 20:34:02,791 : Error Receiving slave packet for register 0001
2022-07-17 20:42:16,405 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57895582
2022-07-17 20:42:16,406 : Error Receiving slave packet for register 0009
2022-07-17 20:50:30,051 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57911309
2022-07-17 20:50:30,052 : Error Receiving slave packet for register 0001
2022-07-17 20:58:43,671 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57926956
2022-07-17 20:58:43,672 : Error Receiving slave packet for register 0001
2022-07-17 21:06:57,285 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57942611
2022-07-17 21:06:57,286 : Error Receiving slave packet for register 0052
2022-07-17 21:15:10,897 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57958316
2022-07-17 21:15:10,898 : Error Receiving slave packet for register 000a
2022-07-17 21:23:24,555 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57973999
2022-07-17 21:23:24,557 : Error Receiving slave packet for register 0053
2022-07-17 21:31:38,197 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 57989681
2022-07-17 21:31:38,198 : Error Receiving slave packet for register 0057
2022-07-17 21:39:51,816 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 58005343
2022-07-17 21:39:51,817 : Error Receiving slave packet for register 0001
2022-07-17 21:48:05,437 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 58021025
2022-07-17 21:48:05,438 : Error Receiving slave packet for register 0208
2022-07-17 21:56:19,055 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 58036726
2022-07-17 21:56:19,056 : Error Receiving slave packet for register 0001
2022-07-17 22:04:32,687 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 58052471
2022-07-17 22:04:32,688 : Error Receiving slave packet for register 05f1
2022-07-17 22:12:46,307 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83 , Sequence: 58068201
2022-07-17 22:12:46,309 : Error Receiving slave packet for register 05f1

jgyates, 2022-07-19:
@andrewvanlagen, your logs look normal for an Evolution 2.0 controller. As expected the modbus errors occur about 8 minutes apart. It looks like between each modbus exception there are about 15,600 good packets based on the sequence number which increases every packet requested from the controller. See issue 15 here: https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

You would need to match the time and date in your log file with the time that you see the symptoms occur for this to show the same behavior that @jmActifio is seeing.

jgyates, 2022-07-22:
I am going to close this issue for now but feel free to post here with follow up questions or any additional data.

closed 2022-07-22 by jgyates

andrewvanlagen, 2022-10-01:
Thanks
Andrew Van Lagen

Closed #738<https://github.com/jgyates/genmon/issues/738> as completed.