jgyates/genmon #739: Odd Test Email Error

alphaa211, 2022-07-24:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Emails are sent.

### Actual Behavior

Not sent since last reboot, several reboots, and power cycles of pi.

### Steps to Reproduce
Intermittent.  Nothing in particular seems to initiate it.

### Screenshot or Pictures relating to the problem 
[image]

### Your Environment

- Generator Model: G0070330
- Genmon Logs: submitted
- Genmon Version: Latest

jgyates, 2022-07-24:
Hi @alphaa211 

I have never seen that message before.
A few questions.

1) You mentioned this is intermittent. Has this worked before?
2)I have not received any logs. Email has to work for the logs to be sent so if you are having any issues with email the logs may not be sent. Log into the pi and look at the file /var/log/mymail.log and copy and paste any errors to this thread.
3) Log into the pi and type these commands and copy and paste the output to this thread.

     python -V

     pip3 freeze

jgyates, 2022-07-24:
I noticed that you were using the older SSL port 465 setting instead of the more recent port 587. Do you get the same results with port 587 and disabling the "Use SSL" setting. This will use TLS encryption. I have tested both and they do work on my end with gmail with the latest code. 

Also, if you have not already please update your code on the about page. Even if you do not show and update available as there have been some updates since the last version change.

alphaa211, 2022-07-24:
> Hi @alphaa211
> 
> I have never seen that message before. A few questions.
> 
>     1. You mentioned this is intermittent. Has this worked before?
>        2)I have not received any logs. Email has to work for the logs to be sent so if you are having any issues with email the logs may not be sent. Log into the pi and look at the file /var/log/mymail.log and copy and paste any errors to this thread.
> 
>     2. Log into the pi and type these commands and copy and paste the output to this thread.
>        python -V
>        pip3 freeze

Have never seen this error message but ever since gmail less secure app change Genmon notifications have been intermittant.  they will work for a few weeks then stop working.  A reboot of the pi sometimes helped.

After python -V there was a page of install looking items then I accidentally closed the app.  Doing it again on a fresh connection yielded Python 2.7.13

Linux GenMonPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Jul 24 18:25:00 2022 from 192.168.1.22
pi@GenMonPi:~ $ python -V
Python 2.7.13
pi@GenMonPi:~ $ pip3 freeze
astroid==1.4.9
asttokens==1.1.13
automationhat==0.2.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2021.10.8
chardet==4.0.0
chump==1.6.0
click==6.6
colorama==0.3.7
colorzero==1.1
configparser==4.0.2
crcmod==1.7
cryptography==1.7.1
docutils==0.13.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
geojson==2.5.0
gpiozero==1.5.0
idna==2.10
isort==4.2.5
itsdangerous==0.24
jedi==0.10.2
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
lazy-object-proxy==1.2.2
ldap3==2.9.1
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
mypy==0.470
numpy==1.12.1
oauthlib==2.0.1
paho-mqtt==1.6.1
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.5
pigpio==1.38
Pillow==4.0.0
ply==3.11
psutil==5.8.0
pyasn1==0.4.8
pycrypto==2.6.1
pycryptodomex==3.12.0
pygame==1.9.3
Pygments==2.2.0
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.7.1
pylint==1.6.5
pyOpenSSL==16.2.0
pyotp==2.3.0
pyowm==2.10.0
pyperclip==1.5.27
pyserial==3.2.1
pysmi==0.3.4
pysnmp==4.4.12
python-apt==1.4.3
pytz==2021.3
pyxdg==0.25
rainbowhat==0.1.0
requests==2.25.1
requests-oauthlib==0.7.0
roman==2.0.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.12.0
skywriter==0.0.7
smbus==1.1.post2
sn3218==1.2.7
spidev==3.3
thonny==3.1.0
thonny-pi==1.1
touchphat==0.0.1
twilio==6.63.2
twython==3.4.0
typed-ast==0.6.3
unicornhathd==0.0.4
urllib3==1.26.7
Werkzeug==0.11.15
wrapt==1.9.0
pi@GenMonPi:~ $

alphaa211, 2022-07-24:
> I noticed that you were using the older SSL port 465 setting instead of the more recent port 587. Do you get the same results with port 587 and disabling the "Use SSL" setting. This will use TLS [...]

It gives the red error message with both 465 with SSL or 587 without SSL.

I had already updated to the latest version, still gave me the error message, did so again, and tried both 465/587 and still no change.

Updated my pi, no  change.

jgyates, 2022-07-25:
What is the output of this command (use your pi's IP address instead on the one below):

     http://192.168.1.10:8000/cmd/support_data_json

Also, are you able to log into the pi and show the errors in the file /var/log/mymail.log with this command:

       cat /var/log/mymail.com

jgyates, 2022-07-25:
sorry, that should be 

        cat /var/log/mymail.log

another thing to try is to log into the pi and get into the genmon directory and run this command:

      cd genmon
      ./genmonmaint.sh -r

This will update all of the libraries that genmon uses to the latest version.

If that does not fix the problem I would need to see the support data and mymail.log output.

alphaa211, 2022-07-25:
> { "Program Run Time": "Generator Monitor running for 2:56:34.", "Monitor Health": "OK", "Controller Selected": "generac_evo_nexus", "StartInfo": { "version": "V1.18.12", "sitename": "Maple Ct", "python": "3.5.3", "platform": "linux", "os_bits": "32", "zone": "CDT", "NominalBatteryVolts": "12", "fueltype": "Propane", "FuelConsumption": true, "RemoteCommands": true, "RemoteTransfer": true, "ResetAlarms": false, "SetGenTime": true, "Linux": true, "nominalRPM": "3600", "FuelCalculation": false, "FuelSensor": false, "phase": "1", "RaspbeerryPi": true, "nominalfrequency": "60", "model": "Generic Air Cooled", "UtilityVoltage": true, "AckAlarms": false, "nominalKW": "10", "PowerGraph": true, "RemoteButtons": false, "Firmware": "V1.16", "WriteQuietMode": false, "ExerciseControls": true, "Actual": "Evolution, Air Cooled", "Hardware": "V1.06", "Controller": "Evolution, Air Cooled" }, "Comm Stats": [ { "Packet Count": "M: 336217, S: 336216" }, { "CRC Errors": "0 " }, { "CRC Percent Errors": "0.00%" }, { "Timeout Errors": "0" }, { "Timeout Percent Errors": "0.00%" }, { "Modbus Exceptions": 0 }, { "Validation Errors": 0 }, { "Sync Errors": 0 }, { "Invalid Data": 0 }, { "Discarded Bytes": "0" }, { "Comm Restarts": "0" }, { "Packets Per Second": "63.47" }, { "Average Transaction Time": "0.0308 sec" } ], "PlatformStats": [ { "CPU Temperature": "127.76 F" }, { "Pi Model": "Raspberry Pi 3 Model B Rev 1.2" }, { "Pi CPU Frequency Throttling": "Has occurred. " }, { "Pi ARM Frequency Cap": "OK" }, { "Pi Undervoltage": "Has occurred. " }, { "CPU Utilization": "3.81%" }, { "OS Name": "Raspbian GNU/Linux" }, { "OS Version": "9 (stretch)" }, { "System Uptime": "2:57:25" }, { "Network Interface Used": "eth0" }, { "System Time": "Sunday July 24, 2022 21:29:21" } ], "Data": { "Registers": { "Num Regs": "247", "Not Changed": "329106", "Changed": "6863", "Total Changed": "0.02", "Base Registers": [ { "0000": "0009" }, { "0009": "00f3" }, { "0053": "0000" }, { "05f1": "0000" }, { "0001": "00000000" }, { "0052": "0000" }, { "005e": "00000000" }, { "0008": "0000" }, { "000b": "000000a4" }, { "0058": "0000" }, { "002c": "1030" }, { "0010": "0016" }, { "0007": "0000" }, { "05f5": "0000" }, { "0258": "0000" }, { "001b": "4cb6" }, { "001a": "00ad" }, { "002e": "0002" }, { "020f": "0000" }, { "05ee": "004d" }, { "020e": "0000" }, { "000f": "0718" }, { "05f4": "0000" }, { "001f": "4f90" }, { "002b": "0005" }, { "0247": "0000" }, { "0246": "0000" }, { "0241": "0000" }, { "005a": "0000" }, { "0032": "4000" }, { "0209": "0000" }, { "0035": "0000" }, { "0057": "0000" }, { "023a": "0000" }, { "0245": "0000" }, { "000d": "0000" }, { "001c": "0000" }, { "0059": "0000" }, { "004c": "0000" }, { "0020": "0000" }, { "0034": "0000" }, { "05ed": "0064" }, { "023b": "0000" }, { "0235": "0000" }, { "0237": "0000" }, { "0244": "0000" }, { "0037": "11e2" }, { "0039": "0000" }, { "0051": "0000" }, { "0212": "0000" }, { "023e": "0000" }, { "0249": "0000" }, { "0054": "0000" }, { "0056": "0000" }, { "0033": "0050" }, { "0236": "0000" }, { "0021": "0000" }, { "020a": "0000" }, { "0208": "03f8" }, { "024a": "0000" }, { "003a": "00000000" }, { "020d": "0000" }, { "025a": "0000" }, { "05f7": "05c8" }, { "0006": "0200" }, { "0011": "00ab" }, { "002f": "0000" }, { "05f3": "00c8" }, { "0019": "000f" }, { "0238": "0000" }, { "0243": "0000" }, { "002a": "6a74" }, { "0005": "1030" }, { "005d": "0000" }, { "05fa": "0000" }, { "0213": "0000" }, { "000a": "0086" }, { "0022": "4b49" }, { "0248": "0000" }, { "020b": "0000" }, { "000e": "151d" }, { "001d": "0000" }, { "0036": "0000" }, { "0012": "0000" }, { "05f6": "05ad" }, { "05f2": "4b49" }, { "0038": "0001" }, { "001e": "0175" }, { "023d": "0000" }, { "0242": "0000" }, { "005c": "0000" }, { "003c": "0000" }, { "002d": "0000" }, { "0055": "0000" }, { "0239": "0000" }, { "020c": "0000" }, { "01f4": "33303030373839363736" } ], "Log Registers": { "Logs": { "Alarm Log": [ "03e8:214f1a0a070416130000", "03ed:214e020c072a16100000", "03f2:214d0a0f053616060000", "03f7:214c3209022316150000", "03fc:144b0c0b0b1c15070000", "0401:044a0b0b0b1c150705e1", "0406:2149100a0a1915130000", "040b:2148050d090a15050000", "0410:2147340e091815030000", "0415:2146130f081b15100000", "041a:21453101072f150a0000", "041f:214430010723150a0000", "0424:2143170d061a150b0000", "0429:21421e09033b150a0000", "042e:21412a0e023b151a0000", "0433:21400d0a0c2e14160000", "0438:143f1c0e0c2114150000", "043d:213e090f0a3814170000", "0442:213d33080a2214080000", "0447:213c3b0f0702140d0000", "044c:213b120f071014080000", "0451:213a130d070214080000", "0456:0039120d072e14080578", "045b:2138240c071114080000", "0460:21370d0b043814110000", "0465:21361d10020014150000", "046a:2135100e020014110000", "046f:2134010a0222140e0000", "0474:213304100b02131b0000", "0479:2132270b0839130a0000", "047e:2131260b050013010000", "0483:04301c100139130f05e1", "0488:042f1c100129130f05e1", "048d:042e1c10011d130f05e1", "0492:042d18100123130f05e1", "0497:042c18100114130f05e1", "049c:042b1710013b130f05e1", "04a1:042a1410010a130f05e1", "04a6:21291b0c0b0112170000", "04ab:2128050d072312140000", "04b0:2127290b072512140000", "04b5:2126000f072212130000", "04ba:21252e0e052312110000", "04bf:21240f10041312020000", "04c4:21232b0f042312020000", "04c9:2122330e031212160000", "04ce:2121310e032112160000", "04d3:0220310e03121216044c", "04d8:211f1a0e031612160000", "04dd:211e1e0c023812130000" ], "Service Log": [ "04e2:3f09310c0a151508", "04e6:3c08281403291517", "04ea:1607101303011517", "04ee:3e0637090321141c", "04f2:18050 [... cut ...] 18100114130f05e1", "049c": "042b1710013b130f05e1", "04a1": "042a1410010a130f05e1", "04a6": "21291b0c0b0112170000", "04ab": "2128050d072312140000", "04b0": "2127290b072512140000", "04b5": "2126000f072212130000", "04ba": "21252e0e052312110000", "04bf": "21240f10041312020000", "04c4": "21232b0f042312020000", "04c9": "2122330e031212160000", "04ce": "2121310e032112160000", "04d3": "0220310e03121216044c", "04d8": "211f1a0e031612160000", "04dd": "211e1e0c023812130000" }, "Strings": {}, "FileData": {} }

alphaa211, 2022-07-25:
> 

pi@GenMonPi:~ $     cat /var/log/mymail.log
2022-07-24 09:55:40,831 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 09:57:40,837 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 09:59:40,838 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:01:40,840 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:03:40,843 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:05:40,845 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:07:40,846 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:09:40,848 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:11:40,850 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:13:40,852 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:15:40,854 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:17:40,856 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:19:40,858 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:21:40,861 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:23:40,863 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 10:25:40,865 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561

alphaa211, 2022-07-25:
> sorry, that should be
>
> ```
>     cat /var/log/mymail.log
> ```
>
> another thing to try is to log into the pi and get into the genmon directory and run this command:
>
> ```
>   cd genmon
>   ./genmonmaint.sh -r [...]

pi@GenMonPi:~/genmon $   ./genmonmaint.sh -r
Updating libraries....
Requirement already up-to-date: crcmod in /usr/local/lib/python3.5/dist-packages
Requirement already up-to-date: configparser in /usr/local/lib/python3.5/dist-packages
Collecting pyserial
  Downloading https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90kB)
    100% |████████████████████████████████| 92kB 1.4MB/s
Installing collected packages: pyserial
  Found existing installation: pyserial 3.2.1
    Not uninstalling pyserial at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed pyserial-3.5
Collecting Flask
  Downloading https://files.pythonhosted.org/packages/e8/6d/994208daa354f68fd89a34a8bafbeaab26fda84e7af1e35bdaed02b667e6/Flask-1.1.4-py2.py3-none-any.whl (94kB)
    100% |████████████████████████████████| 102kB 1.4MB/s
Collecting Jinja2<3.0,>=2.10.1 (from Flask)
  Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
    100% |████████████████████████████████| 133kB 1.5MB/s
Collecting click<8.0,>=5.1 (from Flask)
  Downloading https://files.pythonhosted.org/packages/d2/3d/fa76db83bf75c4f8d338c2fd15c8d33fdd7ad23a9b5e57eb6c5de26b430e/click-7.1.2-py2.py3-none-any.whl (82kB)
    100% |████████████████████████████████| 92kB 1.9MB/s
Collecting Werkzeug<2.0,>=0.15 (from Flask)
  Downloading https://files.pythonhosted.org/packages/cc/94/5f7079a0e00bd6863ef8f1da638721e9da21e5bacee597595b318f71d62e/Werkzeug-1.0.1-py2.py3-none-any.whl (298kB)
    100% |████████████████████████████████| 307kB 798kB/s
Collecting itsdangerous<2.0,>=0.24 (from Flask)
  Downloading https://files.pythonhosted.org/packages/76/ae/44b03b253d6fade317f32c24d100b3b35c2239807046a4c953c7b89fa49e/itsdangerous-1.1.0-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from Jinja2<3.0,>=2.10.1->Flask)
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-1.1.1-cp35-cp35m-linux_armv7l.whl
Installing collected packages: MarkupSafe, Jinja2, click, Werkzeug, itsdangerous, Flask
  Found existing installation: MarkupSafe 0.23
    Not uninstalling markupsafe at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: Jinja2 2.8
    Not uninstalling jinja2 at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: click 6.6
    Not uninstalling click at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: Werkzeug 0.11.15
    Not uninstalling werkzeug at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: itsdangerous 0.24
    Not uninstalling itsdangerous at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: Flask 0.12.1
    Not uninstalling flask at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed Flask-1.1.4 Jinja2-2.11.3 MarkupSafe-1.1.1 Werkzeug-1.0.1 click-7.1.2 itsdangerous-1.1.0
Requirement already up-to-date: pyowm==2.10.0 in /usr/local/lib/python3.5/dist-packages
Requirement already up-to-date: requests<3,>=2.20.0 in /usr/local/lib/python3.5/dist-packages (from pyowm==2.10.0)
Requirement already up-to-date: geojson<3,>=2.3.0 in /usr/local/lib/python3.5/dist-packages (from pyowm==2.10.0)
Collecting certifi>=2017.4.17 (from requests<3,>=2.20.0->pyowm==2.10.0)
  Downloading https://files.pythonhosted.org/packages/0d/e0/481445ea9f9a8ede3f6f373566c7fb20de435203e4828f3738906013ecfa/certifi-2022.5.18-py3-none-any.whl (155kB)
    100% |████████████████████████████████| 163kB 1.3MB/s
Requirement already up-to-date: chardet<5,>=3.0.2 in /usr/local/lib/python3.5/dist-packages (from requests<3,>=2.20.0->pyowm==2.10.0)
Requirement already up-to-date: idna<3,>=2.5 in /usr/local/lib/python3.5/dist-packages (from requests<3,>=2.20.0->pyowm==2.10.0)
Collecting urllib3<1.27,>=1.21.1 (from requests<3,>=2.20.0->pyowm==2.10.0)
  Downloading https://files.pythonhosted.org/packages/ec/03/062e6444ce4baf1eac17a6a0ebfe36bb1ad05e1df0e20b110de59c278498/urllib3-1.26.9-py2.py3-none-any.whl (138kB)
    100% |████████████████████████████████| 143kB 1.4MB/s
Installing collected packages: certifi, urllib3
  Found existing installation: certifi 2021.10.8
    Uninstalling certifi-2021.10.8:
      Successfully uninstalled certifi-2021.10.8
  Found existing installation: urllib3 1.26.7
    Uninstalling urllib3-1.26.7:
[... 69 lines cut ...]
Requirement already up-to-date: pyasn1 in /usr/local/lib/python3.5/dist-packages
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Requirement already up-to-date: smbus in /usr/local/lib/python3.5/dist-packages
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Collecting psutil
[... cut ...]
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Collecting pyotp
  Downloading https://files.pythonhosted.org/packages/a5/cd/9114b5116a9bfbf7133e79f2f12c31a8611a689849a1a990fe5586c08526/pyotp-2.6.0-py2.py3-none-any.whl
Installing collected packages: pyotp
  Found existing installation: pyotp 2.3.0
    Uninstalling pyotp-2.3.0:
      Successfully uninstalled pyotp-2.3.0
Successfully installed pyotp-2.6.0
Done.
pi@GenMonPi:~/genmon $

rwskinner, 2022-07-25:
For grins, in settings, change Sender Name to something different then GenMon and see if it helps.
When mine said was set to GenMon, I was getting weird errors also.  I changed it to Skinner Generator, and the errors went away.  I'm not sure if it had squat to do with it, but it's worked every since then.

jgyates, 2022-07-25:
I checked in a change that ***may*** help. It will either solve your issue or give more information in the mymail.log file. Update your software from the about page and let me know how it goes.

alphaa211, 2022-07-25:
So after updating using:

  ./genmonmaint.sh -r

I can't even access Genmon with my desktop. I can log into the pi via SSH via my desktop though.

jgyates, 2022-07-25:
have you tried to restart genmon?

        cd genmon
        ./startgenmon.sh restart

or you could power cycle your pi and that would do the same thing. When you update the libraries you should restart genmon. Sorry I forgot to mention that.

alphaa211, 2022-07-25:
> have you tried to restart genmon?
> 
> ```
>     cd genmon
>     ./startgenmon.sh restart
> ```
> 
> or you could power cycle your pi and that would do the same thing. When you update the libraries you should restart genmon. Sorry I forgot to mention that.

Power cycling didn't do it.  The restart command did.  I got:

onPi:~ $ cd genmon
pi@GenMonPi:~/genmon $   ./startgenmon.sh restart
Restarting genmon python scripts
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Stopping....
Stopping genexercise.py
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
Starting /home/pi/genmon/genexercise.py
pi@GenMonPi:~/genmon $

HOWEVER, the same weird red error message pops up using 465 SSL or 587 without SSL.   I did update to the newest version with no change.

jgyates, 2022-07-25:
Did you update from the about page?

If you did please send the output of this command:

       cat /var/log/mymail.log

alphaa211, 2022-07-25:
> Did you update from the about page?
> 
> If you did please send the output of this command:
> 
> ```
>    cat /var/log/mymail.log
> ```

I did update from the about page.

2022-07-24 21:36:51,618 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:38:51,620 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:42:00,432 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:44:00,437 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:46:00,439 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:48:00,441 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:50:00,443 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 21:54:30,155 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 22:01:28,023 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 22:03:28,030 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 22:05:16,566 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 22:06:32,540 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:06:32,541 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:08:32,543 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:08:32,544 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:10:32,546 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:10:32,548 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:12:32,550 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:12:32,551 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:14:32,553 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:14:32,555 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:16:32,557 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:16:32,558 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:17:06,345 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:17:06,346 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
pi@GenMonPi:~/genmon $

jgyates, 2022-07-25:
OK that is helpful. Try updating one more time from the about page and send me the same info

alphaa211, 2022-07-25:
It worked; both with 465 SSL & 587 without.

6,566 : Error in SendMailThread, retrying (2): object of type 'EmailPolicy' has no len() : mymail.py:561
2022-07-24 22:06:32,540 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:06:32,541 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:08:32,543 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:08:32,544 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:10:32,546 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:10:32,548 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:12:32,550 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:12:32,551 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:14:32,553 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:14:32,555 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:16:32,557 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:16:32,558 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:17:06,345 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:17:06,346 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:19:06,347 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:19:06,349 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:20:34,653 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:20:34,654 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:22:34,656 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:22:34,658 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-07-24 22:24:34,660 : Error in email init: object of type 'EmailPolicy' has no len() : mymail.py:465
2022-07-24 22:24:34,662 : Error in SendMailThread, sendEmailDirectMIME failed, retrying

jgyates, 2022-07-25:
ok. so I assume these errors in your log are from before you did the update.

 Let me know if you are still having any issues. I will close this issue. I am glad it is working now.

closed 2022-07-25 by jgyates

alphaa211, 2022-07-25:
What is the command to have the pi start genmon automatically whenever it boots?  It appears it isn't doing so but when I use the restart command via SSH it starts Genmon.

jgyates, 2022-07-25:
There is not a single command that does this. Genmon starts at boot when a line is added to the cron tab. You will have to give me more info on the “restart” command you are using. Are you talking about the “./startgenmon.sh restart” command or something else?

If you type this:

        sudo reboot

The pi will reboot. If genmon does not start then your cron tab entry may be missing. 

You can run the install again to update your cron tab. First stop genmon

     ./startgenmon.sh stop

Then run the install script:

     ./genmonmain.sh -i  

Then reboot and see if genmon starts automatically. note that this will take a few minutes to restart.

alphaa211, 2022-07-25:
Is the install script correct?

pi@GenMonPi:~/genmon $  ./startgenmon.sh stop
Stopping genmon python scripts
Stopping....
Stopping genexercise.py
Stopping genserv.py
Stopping genmon.py
pi@GenMonPi:~/genmon $  ./genmonmain.sh -i
-bash: ./genmonmain.sh: No such file or directory
pi@GenMonPi:~/genmon $ cd
pi@GenMonPi:~ $  ./genmonmain.sh -i
-bash: ./genmonmain.sh: No such file or directory
pi@GenMonPi:~ $ ./genmonmain.sh -i
-bash: ./genmonmain.sh: No such file or directory

jgyates, 2022-07-25:
./genmonmaint.sh -i

alphaa211, 2022-07-25:
What is the best version of Python to be using with Genmon at the moment?  Do I need to worry about:

usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509

alphaa211, 2022-08-05:
I hadn't updated the pi since 2018 when I installed it and had a hitch going from 10 to 11.  I decided it was easiest to perform a complete fresh install.  I updated my pi firmware, reformatted my SD card, installed the latest recommended pi OS from the official website, reinstalled Genmon, and dropped my backup file in the proper folder per the Wiki.  It worked great, no issues, pi runs faster, Genmon is very quick with alerts, I have had no problems since, and it didn't take very long to do.  Thanks for your assistance and all your work on such a great project.

jgyates, 2022-08-05:
I am glad it is working for you now. Thanks for the update.