jgyates/genmon #740: Erroneous low fuel alerts

aric8456, 2022-07-26:
### Expected Behavior

Genmon throws alert when actually low on fuel

### Actual Behavior

I have a generac 14kW (exact model isn't at issue here) with 2x 100g tanks monitored via TankUtility. It has been working great until the last month or so when I keep getting low propane alerts from genmon despite having over 70% fuel remaining.

### Steps to Reproduce (including precondition)

n/a

### Screenshot or Pictures relating to the problem (if possible)

[image]

[image]

[image]

### Your Environment

- Generator Model: Evolution 2.0, Air Cooled
- Genmon Logs: Submitted
- Genmon Version: 1.18.13

jgyates, 2022-07-26:
Hi @aric8456, I received your logs. 

Can you send a screen shot of your maintenance page? this page should contain the tank level of each tank.

What is the Poll Frequency in the settings for the Tank Utility add on?

Can you describe when and how often you receive the email notices (i.e. only when you restart genmon, once a day, weekly, etc)?

aric8456, 2022-07-26:
[image]

Poll Frequency is "60"

It's when the genmon reboots as well as after the generator either turns off from load or weekly excersize

jgyates, 2022-07-26:
you mentioned you have " 2x 100g" tanks. It looks like you only have one tank reading instead of two. do you have 2 tanks or 1? Please show your settings for the tank utility add on.

aric8456, 2022-07-26:
Not being a plumber nor an electricican, the way I understand it is, the two tanks act as one. Tank Utility has a single "tank" that it monitors

[image]

[image]

[image]

jgyates, 2022-07-26:
I will start by saying I don't have any idea how your two tanks are connected, however tank utility works by reading the gauge dial of a tank float gauge. Since each tanks has a float gauge I would guess that your tank utility is only reading one tank. 

This video shows the internals of a float gauge and sensor: https://www.youtube.com/watch?v=ix0XxbnBDhg&ab_channel=DpsTV

According to genmon your tank utility monitor is reading 75% full. Look at the gauge on both tanks (if you have two gauges), what are the readings on the physical gauge?

You gentankutil.log log file contains one or two of these errors each day:

2022-07-26 03:06:14,867 : Error in tankutility:Login: HTTPSConnectionPool(host='data.tankutility.com', port=443): Max retries exceeded with url: /api/getToken (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb5cea5c8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')) : mytankutility.py:45

I am not sure if this is causing your issue but it is noteworthy. At the moment the configuration of your tanks (2 tanks vs 1 sensor) and the error in you logs are the only things I see that are unusual.

One thing that we can try is to enable the debug option for the tank utility add on. To do this do the following:

1) Update to the latest code ( I added some additional debug output to the log)
2) manually edit the file /etc/genmon/gentankutil.conf, to do so log into the pi and use this command:

      sudo nano /etc/genmon/gentankutil.conf 

Then add the following line to the end of the file

      debug=True

save the file by typing Ctrl+X, then Y, then Enter

3) restart genmon

       cd genmon
       ./startgenmon.sh restart

Once genmon has restarted, make a note if you see the email warning about the fuel level. Then wait 2 minutes then send your logs from the about page again. This will allow the data sent by tank utility to be captured in the log.

Once you are done sending your logs then you can edit the conf file and set debug=False, save the file and restart genmon. 

Let me know if you have questions.

jgyates, 2022-07-26:
Also, the code should only check the fuel level (and send email messages if low) every 10 minutes with the first being 10 minutes ***after*** genmon has started. So you may want to let genmon run for at least 10 minutes.

Also, if you are receiving this email when genmon exercises or starts then do a manual start let it run and then stop it, see if you get the email, then send your logs (all of this with the debug=True enabled would be more helpful).

jgyates, 2022-07-26:
I checked in one additional change that will add the exact fuel percentage to the notice email body.

aric8456, 2022-07-28:
Well....somehow it just started working randomly, I can't replicate the behavior at all anymore. Sharing this just for info.

[image]

closed 2022-07-28 by aric8456