jgyates/genmon #741: Feature Request - Mopeka Tank Level Sensor

rwskinner, 2022-07-28:
The Mopeka Tank check sensor is very popular bluetooth sensor that is widely used in the RV industry, and also stationary tanks.
Since lots of tanks are fairly close to the generator, is there any chance of reading this sensor via the Pi's BlueTooth?

There is a lib already here, but honestly, I'm not fluent enough to try to implement it.

https://github.com/spbrogan/mopeka_pro_check

It's an ultrasonic sensor that goes on the bottom of any tank and very accurately measures the actual level.

jgyates, 2022-07-28:
that sounds like a good idea. It should be relatively simple as long as the provided library works as advertised. Unfortunately I don't have a mopeka tank sensor so unless someone wants to provide one or I would need to have someone willing to perform testing and be able to send logs back so I can see and debug what is going on with the code. In the interim I will add this to the feature list and update this thread when this is complete.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

rwskinner, 2022-07-28:
I have one on my 250 gallon tank I can test with.  I would be more than happy to do so.  Just let me know.

jgyates, 2022-07-28:
ok. I will let you know once I have something to test.

rwskinner, 2022-07-29:
What's your address and I can ship you one.  That would be much easier than blind testing.  You can email it to me privately if you wish.

jgyates, 2022-07-29:
I actually bound one last night, I figured I could use it on my gas grill. Thanks for the offer though.

I looked at the library you referenced. One slight issue is that the code will report the results of the tank in millimeters so for any tank to get a percentage full reading we would need to know the minimum millimeter reading and the maximum millimeter reading (or in inches as that is easy to convert).

This code:

https://github.com/spbrogan/sensor.mopeka_pro_check/blob/main/custom_components/mopeka_pro_check/const.py#L31

has minimum and maximum for 20, 30, 40 and 100 lb tanks, however I am not sure how to handle any other size tanks other than getting feedback from users with a full tank and letting people put these values in themselves.

We could compare the reading the pi reads from the sensor and compare it to your tanks reading on your smart phone, then extrapolate the tank depth max (assuming the minimum is the same as the link I posted).

rwskinner, 2022-07-29:
Dang,  I was going to get Amazon to ship you one for all the trouble.

The 20, 30 40, and 100# tanks are vertical and therefore their capacity is easy to calculate since it's linear.  I can get you the tank dimensions then it's just a cubic inches to gallons conversion depending on the level.

On the horizontal tank, for the 250, 500, and 1000,  you still need the dimensions to get capacity, but the formula is not linear since there is more surface area in the center of the tank than at the low side and high side.  I had to do this for horizontal oil tanks, and I have the code I used which proved pretty accurate, but its in pascal.  I'm sure there are python horizontal tank formulas out there.

I can google and get the standard tank dimension for you.  Tank Check app uses std tank sizes, but allows you to use a custom tank size.

jgyates, 2022-07-29:
Sounds good, any help on the dimensions would be useful. I can convert pascal to python if needed.

jgyates, 2022-08-02:
I received the sensor today and proceeded to do some testing. The good news is that I can get readings from the sensors via the app and the python library in the link above. The bad news is that by default the pi maps the on board serial port and the bluetooth device to the same device name (/dev/serial0). There are ways around this but it requires some digging to figure out the exact solution. I am going to think on this for a bit and see if a maintainable solution presents itself. In the interim I will close this issue and update it when I have some additional data.

closed 2022-08-02 by jgyates

jgyates, 2022-08-04:
I have created an add on for Mopeka Pro sensors. The wiki has more info:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional

Let me know if you have any questions.

If you use a 100, 250 or 500 gallon tank let me know. Presently only the vertical tanks are support as I don't have any info on the tank millimenter max and min values and how they scale for the horizontal tanks. The /genmon/OtherApps/mopeka_utility.py program will take a sensor reading if the SYNC button is pressed while the program is running. It will display the millimeter depth reading on the console output.

rwskinner, 2022-08-04:
My apologies, I stated I would get that for you and I got sidetracked.  
Propane Tank Sizes  https://www.amerigas.com/about-propane/propane-tank-sizes

I'll either dig up my code for the volume of horizontal tanks or find one handier.

skipfire, 2022-08-04:
I believe this is what you need but not 100% sure I got the formula converted to code correctly at the acos.  Originally it is inverse cos((r-h)/r).  I believe acos is inverse cos in radians.

m = measured height from top of cylinder to top of fuel
h = 2r - m
a = math.degrees(math.acos((r−h)/r)) * r2 − (r−h) * math.sqrt(2rh−h^2)
v = a*l
If inches, gal = v/231

jgyates, 2022-08-04:
hmm, it has been much too long since i used trig. what value is r and l? Maybe a key for the variable meanings would be helpful in me digesting this.

skipfire, 2022-08-04:
Sorry about that.
d = diameter
r = radius = 1/2 diameter
l = length
h = height of fuel
a = filled area of a circle
v = filled volume of the cylinder

The full volume of the cylinder would be math.pi * r^2 * l.

rwskinner, 2022-08-04:
Function SegmentArea (Depth, Dia : Single) : Single;
Var Radius : Single;
    Temp : Single;
    ang, arcl, seg : Single;
    Chordl : Single;
Begin
  If ((Depth > 0) AND (Dia > 0)) then
  Begin
    Radius := Dia / 2;
    Temp := radius - Depth;
    ChordL := 2 * sqrt(2 * Depth * radius - depth * depth);
    ang := arccos(temp/radius) * 2;
    arcl := ang * radius;
    seg := ((Arcl * radius - chordl * temp)/2);
    Result := seg;
  End Else
    Result := 0.0;
End;

//Returns gallons when you feed it Diameter, Length, and current Level in inches.
//This is for flat ends like for a fuel or oil tank.  You need to add provisions for spherical heads but honestly, this will be close enough I bet.

Function CalcRoundHorizontalTank (Dia, Len, Depth : Single): Single;
Var    Area, Vol : Single;
Begin
  If (Depth > Dia) then Depth := Dia;  // If tank is over full then it equals full
  Begin
    If ((Dia >1) AND (Len > 1) AND (Depth > 0)) then
    Begin
      Area := SegmentArea(Depth, Dia);
      Vol  := ((area * Len) /1728) * 7.48;
      Result := Vol;
    End Else
      Result := 0.0;
  End;
End;

rwskinner, 2022-08-04:
`
Function SegmentArea (Depth, Dia : Single) : Single;
Var Radius : Single;
    Temp : Single;
    ang, arcl, seg : Single;
    Chordl : Single;
Begin
  If ((Depth > 0) AND (Dia > 0)) then
  Begin
    Radius := Dia / 2;
    Temp := radius - Depth;
    ChordL := 2 * sqrt(2 * Depth * radius - depth * depth);
    ang := arccos(temp/radius) * 2;
    arcl := ang * radius;
    seg := ((Arcl * radius - chordl * temp)/2);
    Result := seg;
  End Else
    Result := 0.0;
End;

Function CalcRoundHorizontalOil (Dia, Len, Depth : Single): Single;
Var    Area, Vol : Single;
Begin
  If (Depth > Dia) then Depth := Dia;  // If tank is over full then it equals full
  Begin
    If ((Dia >1) AND (Len > 1) AND (Depth > 0)) then
    Begin
      Area := SegmentArea(Depth, Dia);
      Vol  := ((area * Len) /1728) * 7.48;
      Result := Vol;
    End Else
      Result := 0.0;
  End;
End;
`

rwskinner, 2022-08-04:
Okay, I tried twice to add code without loosing all the indent but it did anyways.

jgyates, 2022-08-04:
duh, thanks Jon!  I am getting old or just lazy (or both) :)

Thanks for the code @rwskinner 

This makes a lot more sense now. I will code this up and let you know once I have something to test.

skipfire, 2022-08-04:
@jgyates no problem, I know how it is when you see something you aren't expecting and your brain is just in the wrong context.  Start off seeing cos, acos, inverse cos and it puts you into trig mode instead of geometry.

jgyates, 2022-08-04:
I assume that if the tank is vertical the reading from the sensor is the millimeters of fuel from the bottom, but if the tank is horizontal and mounted at the top of the tank the reading is mm of space between the top of the fuel and the tank. Any thoughts?

skipfire, 2022-08-04:
If I found the right sensor it is an ultrasonic sensor, so it should measure the distance from the relative top of the tank to the top of the fuel either way, measuring the air-gap.

rwskinner, 2022-08-04:
No, the sensor always gets mounted at the bottom.  I have the code also for vertical tanks. 

//Everything is passed in inches and returned in gallons.  
//Remember that a Propane tank is considered full at 80% but the Tank Sensor mainly shows a percentage anyways.
//When so when I fill my 250 gal tank, they only put 200 gal in it.  The Tank App shows 80% full, BUT you can easily scale that to //0-100% by using tank size io gallons x .8 
Function CalcRoundVerticalTank (Dia, Len, Depth : Single): Single;
Var    Radius, Area, Vol : Single;
Begin
  If (Depth > Len) then Depth := Len;  // If tank is over full then it equals full  10830
  Begin
    If ((Dia >1) AND (Len > 1) AND (Depth > 0)) then
    Begin
      Radius := Dia / 2;
      Area   :=  ((Radius * Radius) * 3.1416) * Depth;  //Cubic Inches
      Vol    := (area / 1728) * 7.48;         //Cubic Inches to Cubic Feet then to Gallons
      Result := Vol;
    End Else
      Result := 0.0;
  End;
End;

jgyates, 2022-08-04:
OK. I ported / incorporated the code from @rwskinner. Let me know how your testing goes.

rwskinner, 2022-08-05:
I'm having a difficult time finding the mac address on the scan.
NOTE: This program will look for Mopeka Pro Sensors. The SYNC button must be pressed and held for the discovery process to work.

Starting Discovery....
Program Error (main): [Errno 19] No such device mopeka_utility.py:87

Discovery lasts about 1/2 second.

jgyates, 2022-08-05:
sounds like your BT is not enabled. What is the output of this command:

     sudo bluetoothctl

If your BT is disable it will look like this:

[image]

if it is enabled it will look like this:

[image]

If it is disabled the you should validate your /boot/config.txt settings with the /genmon/OtherApps/serialconfig.py program. A reboot is required once a reboot takes place. What is output of serialconfig.py?

jgyates, 2022-08-05:
type 'exit' to exit the bluetoothctl program

jgyates, 2022-08-05:
I added a couple of troubleshooting steps to the wiki.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional

rwskinner, 2022-08-05:
First, thank you for working on this.  I was able to get it to talk using the utility by following your restore then reapply instructions.  I can't tell if it's really working, Even If there is no connection, then my tank shows 100% full ?

Is there a way for me to see the raw output, somewhere?  That would help me validate things.
Raw mm or inches of level, BlueTooth signal, Battery Voltage, Number of received Packets.  Right now I'm kind of running blind.

rwskinner, 2022-08-05:
Okay, weird, I set it to a 500 gallon tank, then it started reading in percent and showed 72% full, then I set it back to a 250 gal tank and it shows like 92% full but the needle on the gauge is like 48% or so.

I looked at the genmopeka log but there wasn't much in there.

jgyates, 2022-08-05:
please post the output of the genmopeka.log.

Also, you can add this line to /etc/genmon/genmopeka.conf:

     debug=True

save the file, then restart genmon, then your genmopeka.log will show a lot more info. When we are done debugging I would recommend that you remove debug=True or set debug=False. If you left that level of debug output on it would likely reduce the life of your SD card over time.

If you open a separate console window, log in an type this:

      sudo tail -f /var/log/genmopek.log

Then restart genmon in another console window you can see the updates to the log file as they happen. 

Note that the default settings will only update the sensor once an hour (Poll Interval = 60 min), for debugging you can lower that to 1 or 2. Also the scan duration is about 15 seconds. Post your log output here once you enable debug.

rwskinner, 2022-08-06:
So seeing the different variations I now understand, sort of.
If the BT reading isn't working, the fuel gauges is ESTIMATED fuel, if the BT is working then it's EXTERNAL Tank.
This evening, the value on the dashboard fuel gauge matches the needle position.  I'm not sure what was going on earlier.

I will enable debug this weekend and collect some data to see how everything aligns, or not.  Again, thanks for the awesome monitor.

rwskinner, 2022-08-06:
[image]

rwskinner, 2022-08-06:
My tank is full, and before the BT deal, it read 192 gallons.  Now, it shows estimated fuel, then 92% but the needle is at 49% ?
I would imagine the 92% is from the Mopeka but the gauges says Estimated?

jgyates, 2022-08-06:
Update your software from the about page again. I forgot to add genmopeka.log to the list of files that get send when logs are submitted.

Regarding your fuel gauge. When ever you change settings or restart genmon you must refresh your browser. The fuel gauges form add on programs change the number of gauges. The web interface requests gauges once when the page is loaded. The picture above could happen, but refreshing your browser should fix it. If it does not then let me know but I will need more info regarding your settings.

rwskinner, 2022-08-06:
Yes, that fixed it.  I saw the label change and other gauge values were updating so I figured it refreshed on restart.  Glad that was an easy fix.  I submitted logs again for grins but I haven't debugged anything yet.

rwskinner, 2022-08-12:
I was able to get time today to stick this in debug and collect some 
data.  Level is off for some reason.
Tank Check monitor is reporting  23.7" of level and at 85% full, or 250 
x .85 = 212.5 gallons.  Which is very close.  It's temperature related 
but I am just a tad overfull.  Tank Gauge shows 84%

GenMopeka is showing 22.95

2022-08-11 15:42:14,472 : Tank Level in mm: 583
2022-08-11 15:42:14,472 : Tank Level in inches: 22.95
2022-08-11 15:42:14,472 : Volume: 231.07
2022-08-11 15:42:14,473 : Tank1 = 92.43

In the library, there are coefficients for Butane, Propane, and I think 
water.  I believe that changes the measurement based on SG.
That will probably make that report the same value afterwards.

To make the volume accurate, on horizontal tanks, use the diameter of 
the tank to make a perfect sphere of the same diameter.
Calculate the total volume of it using a formula like I did. Diameter 
and Level to return gallons.  Each end cap is exactly 1/2 a sphere. so 
both halves is the same as a full sphere.
Tank that amount calculated and add it it the Horizontal Tank 
calculation and total it.

No internet here right now, only email or I would look it up and write 
the formula.  I hope this goes thru.

rwskinner, 2022-08-12:
I was able to get time today to stick this in debug and collect some data.  Level is off for some reason.
Tank Check monitor is reporting  23.7" of level and at 85% full, or 250 x .85 = 212.5 gallons.  Which is very close.  It's temperature related but I am just a tad overfull.  Tank Gauge shows 84%

GenMopeka is showing 22.95

2022-08-11 15:42:14,472 : Tank Level in mm: 583
2022-08-11 15:42:14,472 : Tank Level in inches: 22.95
2022-08-11 15:42:14,472 : Volume: 231.07
2022-08-11 15:42:14,473 : Tank1 = 92.43

In the library, there are coefficients for Butane, Propane, and I think water.  I believe that changes the measurement based on SG.
That will probably make that report the same value afterwards.

To make the volume accurate, on horizontal tanks, use the diameter of the tank to make a perfect sphere of the same diameter.
Calculate the total volume of it using a formula like I did.  Diameter and Level to return gallons.  Each end cap is exactly 1/2 a sphere. so both halves is the same as a full sphere.
Tank that amount calculated and add it it the Horizontal Tank calculation and total it.

jgyates, 2022-08-12:
It could be temp related and a combination of differences in tank sizes. When looking at standard tank sizes I found some small discrepancies on the web that could make a slight difference. The tank dimensions are here in the code:

https://github.com/jgyates/genmon/blob/184cd7825396adf46c524e6cb51bf57bf1bb13e6/genmopeka.py#L165

I used your algorithms above. Let me know if you think they need to be tweaked or if you think there is an error.

rwskinner, 2022-08-12:
I'll take a look but the temperature for what I'm mentioning doesn't have anything to do with temp.  The tank level in MM's reported to your app in comparison to the level reported from the Mopeka app, especially since live samples are being reported at the same time so they should be the same.   I will however go thru and look at the coefficients of the Mopeka api and see if I can reconcile the differences.  Also,  I will work on adding the sphere function tot he code and prove it out some before I send it over to you.  My code was for tanks with flat ends, as in oil and fuel storage tanks, not tanks with concave tank ends.  I will see if I can fix that.

rwskinner, 2022-08-12:
It looks like your calling TankLevelInMM which already does the coefficients so that should work perfect. So I will continue to look into why the factory Mopeka Tank Check App and the Github Mopeka API aren't returning the same value.  For now, I'm going to run the utility program separate and try to figure out what's going on and how to make them match.  Let you know what I find.  Works crazy right now so may be awhile.

rwskinner, 2022-08-12:
Good News!  I had a chance today to verify the readings from the Mopeka Mobile App and the GenMon readings and they both are indeed reading the same when reading mm or inches,  That's good.  I wasn't doubting you, but I never have had experience with the api we used.

So now, I will drill into the calcs a little bit because my tank size functions, which work perfect for oil and water tanks  (very well proven) needs some tweaking for horizontal tanks and semi-spherical ends.  Mopeka Mobile App shows 84% which would be around 221 gallons.   GenMon shows 94.35% and a volume of 235 gal.  Not too bad really, but it doesn't make sense.  So I will play with those calculations some and see if I can fine tune them.

rwskinner, 2022-08-12:
Take a look here:
https://www.omnicalculator.com/construction/tank-volume#elliptical-tank-volume-oval-tank
Using the horizontal capsule tank the Fill Percentage comes out on the money.
94" L x 30" D fill of 23.4"
Tank capacity 80582 cu/in  Fill cap: 67.992 cu/in
67992/80582 = 84.3%  rounded to 84%  which happens to match the Mopeka moble app.
Formulas are there.

jgyates, 2022-08-12:
Are you using a 250 gal tank? Your dimensions (94LX 30 D) make me think you do, but a tank capacity o0f 80582 is 348 gal unless I am calculating things wrong (cubic inches / 231 = gallons). The tank is 250 gal and the fill capacity is 250 * .80 = 

The formula provided in the link uses length, hight, width and filled height. 

I plugged in the formula in the link and used your numbers (94" L x 30" D fill of 23.4") and I get 55606 cu/in (240 gal). I am using the diameter for the width and height in the formula. 

The genmokepa.py add on uses this site for it's dimensions: https://learnmetrics.com/propane-tank-sizes/. This site uses 92 for 250 gal tanks.

Also, using the provided formula and the method described above I get the almost the exact same results as the original formula proved (96.29 vs 96.28)

jgyates, 2022-08-12:
after changing the rounding the two formulas give the exact results, which makes sense because if you make the width and height the same, it is a circle not an ellipse.

jgyates, 2022-08-13:
I checked in an upadate that uses a different approach for tank calculations. This uses the the python [fluids](https://fluids.readthedocs.io/) library.  This allows a tank to be defined with both elliptical and spherical ends in both horizontal and vertical positions. I also limited the use of the add on to python 3.7 as this is requirements of the mopeka library. I tested with a a few different configurations. The larger tanks appear to be more accurate than the smaller tanks. I was able to get 84% by simulating your readings above. My 20lb tank shows 76% with the addon and 84% with the mopeka app.

Let me know how your testing goes. It may take longer to do the update since I added a new library and it has to install.

rwskinner, 2022-08-14:
Great job.  The new fluids library has is on the money!  I never could 
get the online calculators to come out right on the gallons either.  
They all kept showing the tank to hold 350 gallons.

jgyates, 2022-08-14:
If you have a moment can you send your logs via the about page so I can see if there are any unidentified errors occurring?

rwskinner, 2022-08-14:
Sent