jgyates/genmon #742: Outage log times seem inaccurate

bubbaiOS, 2022-07-31:
Notice that the durations where there was fuel used. I'm on 1.18.08 but will upgrade. 

Outage Log : 
        2022-07-27 19:30:11, Duration: 0:00:00, Estimated Fuel: 6.55 gal
        2022-07-25 21:49:00, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-06-17 18:46:20, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-06-17 18:46:05, Duration: 0:00:00, Estimated Fuel: 5.95 gal
        2022-06-17 18:43:23, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-06-17 18:38:16, Duration: 0:00:00, Estimated Fuel: 5.95 gal
        2022-05-21 18:24:37, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-05-20 13:06:18, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-03-28 10:54:31, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-03-08 22:38:54, Duration: 0:00:00, Estimated Fuel: 4.58 gal
        2022-01-29 02:39:38, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2022-01-03 05:36:51, Duration: 0:00:00, Estimated Fuel: 3.82 gal
        2022-01-02 03:28:59, Duration: 0:00:15, Estimated Fuel: 0.00 gal
        2021-12-19 13:41:40, Duration: 0:00:14, Estimated Fuel: 0.00 gal
        2021-05-04 18:29:12, Duration: 0:00:21, Estimated Fuel: 0.00 gal

jgyates, 2022-07-31:
@bubbaiOS 

thanks for the feedback. Not sure what is happening specifically however this is probably room for a quick fix that should disallow this from happening.  There is not a lot of information to go on to tie this to an underlying cause. Not knowing the generator controller model, controller registers, or seeing the logs, power log and raw outage log. If you want to provide the above I would be happy to take a look, otherwise I will keep an eye open for things that may be related.

some background: the fuel estimate in the outage log is taken from the power log. The accuracy of the power log varies by a few things depending on the model of the controller. Also, for performance reasons the power log is reduced to keep the pi from spending too much time parsing the power log. The outage log does not keep outages across a restart of the software or a reboot so this can confuse the software when doing calculations. 

Since your errors are all on an outage of zero duration I can take a look at the code and explicitly make a case for disallowing a fuel calculation.

Also, on a related note, there is a setting on the Advanced Page named "Outage Notice Delay". This will allow you to create a delay before an outage is logged or an outage notification is sent. It looks like you are getting outages of less than a second. At a minimum I would upgrade to the latest and set the "Outage Notice Delay" to a minimum of 5 seconds since the controller typically has about a 5 second delay before an attempt to start the engine on delay.

jgyates, 2022-07-31:
Sorry, there are two outage settings on the advanced page. The other one is "Minimum Outage Duration" this is the one that will not log for x seconds, the other is regarding the outage notice.

closed 2022-07-31 by jgyates

jgyates, 2022-07-31:
I check in an update that ***may*** help.

bubbaiOS, 2022-07-31:
I have submitted the logs and registers to you from the about page. 

Maintenance :
    Model : G0062770

Generator Monitor Stats :
        Monitor Health : OK
        Controller : Evolution, Air Cooled
        Run time : Generator Monitor running for 1:25:50.
        Power log file size : 0.03 MB of 15.00 MB
        Generator Monitor Version : V1.18.13

    Communication Stats :
        Packet Count : M: 162186, S: 162185
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 0
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 62.65
        Average Transaction Time : 0.0313 sec

    Platform Stats :
        CPU Temperature : 130.64 F
        Pi Model : Raspberry Pi Zero 2 Rev 1.0
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 6.05%
        OS Name : Raspbian GNU/Linux
        OS Version : 11 (bullseye)
        System Uptime : 146 days, 18:46:07

I don't see any advanced settings in the web UI unless I'm missing it from my phone.

jgyates, 2022-07-31:
To view the advanced settings page you double click the gear in the top right of the web interface.

jgyates, 2022-07-31:
I have not received your logs. To send the logs your outbound email must be setup and working, then you can submit them from the about page.

bubbaiOS, 2022-07-31:
Apologies. My smtp settings were incorrect but you should have some logs now.

jgyates, 2022-08-01:
I am pretty sure the change I checked in yesterday will fix this issue. Your outage log will not show any change until you have another outage and then only new entries in the log will display correctly. You can edit your outage log if you want to remove any bad entries by editing the file /etc/genmon/outage.txt on your pi.