jgyates/genmon #743: Erroreous Outage Detection

rdb389, 2022-08-04:
### Expected Behavior

Only report outage when actually happening.

### Actual Behavior

Batches of outages showing up in the outage log in quick succession. Unclear if this is a result of Genmon or erroneous data from the controller.
Possibly related utility voltage maximum spikes to unrealistic values as well. 

See screenshot below.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 22kW Generac with Evo 2 Controller
- Genmon Logs: Submitted
- Genmon Version: 1.18.13

jgyates, 2022-08-04:
I received your logs.

This is erroneous data from the controller. Note the maximum until voltage of 14433V. The utility value is read directly from the controller. My guess is that this is due to bugs in the Evolution 2 controller firmware. See number 11 on this page: https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

I can add some code that would bounds check the voltage in a reasonable range ( e.g. ignore anything above 300V for single phase evolution 2.0) that may help the problem.

What is odd is that the minimum utility voltage is not below the threshold voltage. Not sure why that is occurring. I will think on that one. Your outage is from today and your logs state genmon has been running for at least 2 days. Genmon should have logged the utility minimum voltage of a value that is below the threshold. When the utility voltage is every below the threshold voltage the system is considered to be in an outage.

rdb389, 2022-08-04:
I have Ignore Unknown values for Register 0001 set to true in the advanced settings.

I agree it might help mitigate the occurrence by adding a bounds check to the utility voltage parameter for that controller.

Maybe the update interval on the Minimum Utility Voltage value is less that of the erroneous outage? They have all lasted less than 5 seconds according to the time stamps.

The Evo controller currently reports Firmware 1.15.176.

In any case, thanks for your help.

jgyates, 2022-08-04:
Setting the "Ignore Unknown" will only guard agains values for the engine status / switch status registers (0001). The other registers apparently have a similar issue. Another thing you could do to mitigate this is to set the "Minimum Outage Duration" setting on the advanced page to 5. A setting of 5 should ignore fleeting values for the utility voltage. The controller has a delay that will not start the generator after x seconds. This is settable via the dealer menu and defaults to 5.

jgyates, 2022-08-04:
"Minimum Outage Duration" is on the advanced page. You may want to set "Outage Notice Delay" as well so you will not get notices of zero duration outages.

rdb389, 2022-08-04:
Yes. Just noticed the Minimum Duration and Notification Delay settings in the advanced menu. I've set those now. That should mitigate the erroneous notices.

Thanks

rdb389, 2022-08-04:
Also, Voltage Maximum just changed back down to a reasonable value. Maybe a bug on the front end?
[image]

jgyates, 2022-08-04:
I looked at the code, one issue is that the threshold and pickup voltages are read from the controller. If these values are reported by the controller incorrectly then it will cause other things to break. for example you could get an outage reported even if your utility voltage is normal, but your threshold voltage is abnormally high. This would perfectly explain the minimum value in your first screen shot.

The maximum and minimum values are kept by genmon. once you restart (due to settings change) genmon it resets both values.

I will try to add some checks to the code to ignore more values is "Ignore Unknown" is enabled. Thanks for reporting this.

rdb389, 2022-08-04:
That makes sense. A failed read of the threshold voltage value would
definitely explain it.

Hopefully Generac’s firmware gets cleaned up eventually but the best we can
do is work around it for now.

Thanks.

jgyates, 2022-08-04:
I checked in an update that puts some bounds on utility voltage, threshold and pickup voltages read from the controller. Just update from the About page. Version number has not been updated yet.

https://github.com/jgyates/genmon/blob/4e99b9ba7b030d895ab12e2b3ab627b6c186cf2d/genmonlib/generac_evolution.py#L1736

Let me know if you have any issues.

closed 2022-08-04 by jgyates

gmn42, 2023-08-14:
Not sure if I should necrobump this, but the issue is very similar to what I'm seeing on an Evo 1 controller (hardware v1.06, firmware v1.17) with genmon 1.18.18.  About once or twice today at apparently random times, I'll get (almost) always exactly eight spurious outage flaps within a roughly one minute inverval.  I have not enabled the "ignore unknown values" advanced setting because the tooltip indicates it only applies to 2.0 controllers.

I'm using an ESP32 serial bridge (the Pintsize.me board) to genmon running on a proxmox instance.  The wifi signal strength on the ESP32 varies from about -72 to -80dbm, but was at -72dbm during the last two incidents yesterday (see screenshot). 

I do get occasional (a few per hour) "Validation Error: Invalid register length" messages in genmon.log.  Sometimes there is one that seems to correlate with a few seconds before each sequence of spurious outage notifications, but many other errors for identical registers that don't result in the notifications, and some notification sequences don't have corresponding errors.  Nevertheless, here are two that correspond to sequences in the screenshot:

<code>2023-08-12 15:25:50,646 : Validation Error: Invalid register length (base) 0001:0000 4 2</code>
<code>2023-08-12 22:40:57,459 : Validation Error: Invalid register length (base) 0001:0005 4 2</code>

I am submitting the logs via the button on the about page, in case that provides more detail.

Thanks for a great tool!  Setup was a little tricky outside of Raspian, but it runs well!

[image]

skipfire, 2023-08-14:
You can adjust a couple of timeouts on the advanced settings page.  I have increased my "Additional Watchdog Timeout" to 22, and "Outage Notice Delay" to 15.  That has reduced my false alarms by over 90%.  To get to the Advanced Settings, open the settings, click the tool in the top right corner, then click it again on the registers page.

jgyates, 2023-08-14:
I agree with @skipfire on those settings. Also, your are likely seeing modbus timeouts. The monitor page has your communication status and I am guessing you will see a non zero value for timeout errors. See number 16 on the Known Issue list: https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues. Adding a few seconds to the modbus timeout value could help. If your wifi signal is good you may be getting a lot of wifi retries, which increases the round trip of the packet. You may also want to look at your wifi environment, if there are a lot of other wifi access points in the vicinity or one on the same channel (frequency band) this could also cause wifi retries. Retires will also occur if the wifi is bad. The short story is the wifi is a non perfect medium and modbus does not have a built in timeout that the sender and receiver both utilize. The Known issues page give a little more detail.

gmn42, 2023-08-14:
Thanks both of you, I missed those options, I'll try them.  There aren't any other access points, but the wifi signal is showing fairly weak from the AP side right now (-83).  I've got plans to put a better antenna on the bridge, which should help as well, I hope.

It was interesting to me that there were nearly always exactly 8 flaps, which made me wonder if it was a software glitch.

jgyates, 2023-08-14:
If your wifi signal is weak these settings will likely improve your issue, but not completely resolve it. what is happening is this:

genmon sends request 1
controller response to request 1
genmon sends request 2 and does not get a response in the timeout period
genmon sends request 3
due to wifi retry delays the request 2 finally arrives at the controller and the controller responds to request 2
genmon receives the request 2 data but thinks it is the data from request 3

Now the requests and response are out of sync and this results in a lot of invalid data. This invalid data can result in reading a invalid data for the utility voltage and other registers. This behavior is unique to modbus over wifi as this will never happen with modbus serial or modbus serial over Ethernet. The "Validation Errors" you are seeing are genmon checking the number of bytes returned for a request (most request are 4 bytes long, a few are 8). This is the only thing that genmon can do to validate the data as modbus does not support a timeout or the concept of a packet sequence number built into the protocol that you can validate you are not getting out of order responses.

gmn42, 2023-08-14:
Thanks, that helps clarify what's going on.  So it sounds like it might be getting 8 out-of-sequence packets before hitting one that triggers a validation error and allows it to re-synchronize.

And from what I can understand, it sounds like using UDP instead of TCP wouldn't help since the 802.11 layer 2 MAC still requires an ack for each frame and will retry until it gets it.

Does the Modbus-TCP protocol allow for better packet sequencing?  Would something like [this](https://github.com/harihanv/esp32-modbus-gateway) or [this](https://github.com/zivillian/esp32-modbus-gateway) work more reliably, with genmon set to modbus TCP mode?

jgyates, 2023-08-14:
Yep, the retries are taking place at the 802.11 layer 2 level so UDP will not help. Even if Modbus-TCP did help (it does have a per packet sequence number), the Evolution controller only speaks modbus RTU. Genmon can use Modbus TCP, but only some industrial models use it. 

Regarding the two modbus tcp to rtu projects, all I can say is it is *might* work, although I can see it failing the same way if the modbus TCP to RTU converter did not handle multiple requests at the same time. If they are just taking the packet an striping the modbus TCP header and adding back in a CRC (TCP already has a checksum so it is removed form modbus TCP) then it would fail the same way.

gmn42, 2023-08-14:
Interesting.  So it looks like Modbus-TCP packet has a transaction ID field that should match master requests with corresponding slave responses, and from an initial reading of the code for both of those projects (the second seems to actually be an implementation of the [eModbus](https://github.com/eModbus/eModbus) project), it looks like they both maintain the tid of the Modbus-TCP request with the corresponding RTU response.  So as long as genmon sends a transaction ID in the request, and validates it in the received response (and discards it if unexpected/out of order), it would avoid these synchronization errors?

I won't have time for a while, but it would be interesting to try it out sometime.  Of course, then I'd have to give up @skipfire 's slick ESPHome integration with the temp sensor.  Now if we could build the bridge into esphome... Perhaps a project for later... for now, the timeout adjustments should help a lot.  Thanks!