jgyates/genmon #744: Hook for Reboot on Comms failure

jwtaylor310, 2022-08-09:
I am running Genmon V1.18.13. on a Pi located about 100' from the generator.  Connection is via an RS232 to USB adapter (PS2303 chip).  Occasionally (once a month or so) Genmon reports a "Generator Monitor Communication" error which is identified by a red connection failure icon in the lower-left corner of the Genmon web page and generation of a warning email.  When this happens, the only fix that I have found is a reboot.  I believe the USB port has shut down and don't believe there is a problem with Genmon itself.  Is there a clean way of hooking Genmon's error handler to allow me to run a shell script or Python program to force a reboot when this error occurs?   Thanks!

jgyates, 2022-08-09:
@jwtaylor310 

There are a few ways to do this. If you want to use a method that just uses python and not any genmon libraries then this page has the info you are looking for:

https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications

Alternatively you can use the [myclient](https://github.com/jgyates/genmon/blob/master/genmonlib/myclient.py) python library to send the "monitor_json" command (this will get the info displayed on the monitor page. The add on "Genmon GPIO Output to blink LED"  does almost what you are looking for so it could be easily modified to do what you want.

https://github.com/jgyates/genmon/blob/master/gengpioledblink.py

you could modify this line:

https://github.com/jgyates/genmon/blob/8815b56fb631608ebe88460be8e1c890e9b6e53c/gengpioledblink.py#L87

to check for the presence of "Not receiving data" and then do a reboot with this command:

     os.system("sudo reboot")

jwtaylor310, 2022-08-09:
Great.  I’ll take a look at the alternatives and set up the reboot.  Thanks for the quick (and detailed) response!  --John

 

https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications

https://github.com/jgyates/genmon/blob/master/gengpioledblink.py

https://github.com/jgyates/genmon/blob/8815b56fb631608ebe88460be8e1c890e9b6e53c/gengpioledblink.py#L87

closed 2022-08-09 by jgyates

jwtaylor310, 2022-08-11:
I have modified gengpioledblink.py as suggested and when I test by unplugging the USB adapter the Pi reboots as intended.  If the USB connection is functional after the reboot, genmon loads and runs as usual.  However, if the USB connection is not re-established prior to genmon loading on the reboot, genmon apparently exits when it can't connect.  In that case gengpioledblink is never called and the Pi sits idle.  Reconnecting the USB feed does nothing as genmon has already exited and the only way to get genmon running again is to manually start it or manually reboot the Pi.  I'm considering writing an external program to check whether genmon is running and, if not, check the USB port and restart genmon when/if communication is re-established, but that doesn't appear to be a very elegant solution.  Any suggestions?<br><br> BTW, during testing I also identified what might be a bug in genmon.  With gengpioledblink.py disabled, disconnecting the USB feed causes genmon to report "Not receiving data".  Once that happens, it looks like genmon will not recover if the USB feed is re-established.  I have a test utility that polls the generator and reports responses.  It has error trapping that closes the serial connection and re-establishes the connection after a send or receive failure.  That works (at least during testing), so there doesn't appear to be an inherent limitation of the Pi or Python prohibiting automatic reconnection.

jgyates, 2022-08-11:
@jwtaylor310 

One thing to try is to add a delay to genmon loading. To do this you can modify the crontab entry:

        sudo crontab -e

If you are prompted to choose an editor choose nano. Then changer this line:

     @reboot sleep 30 && /bin/bash /home/pi/startgenmon.sh start

Change the "sleep 30" above to some some number larger than 30. This is the number of seconds to delay before loading genmon. Try "sleep 100"

It sounds like your pi is booting faster than your USB serial device. Possibly even try another USB serial device if you have one.

Also, make sure you rename gengpioblink.py or it will get overwritten when you update your software. If you want your renamed file to  be loaded with genmon you can add an entry in /etc/genmon/genloader.conf and it will get loaded along with the other add on programs. If you load it via genloader, your python script needs to be in the same folder as genmon.py (i.e. /hom/pi/genmon).

Regarding the bug you mentioned, I am not surprised. Genmon has not been tested or coded for hot plugging a USB device. In general,  this would need to explicitly programmed for in the serial handing.  I can add this to the feature request list.

jgyates, 2022-08-11:
If you can send your logs that show what happens when you disconnect serial USB that would help.

jwtaylor310, 2022-08-11:
It may not be necessary to add the delay at boot as I expect that in the real world the communications loss is quite brief and (hopefully) would clear before the reboot completes.  If that doesn’t turn out to be the case, I can add the delay in the crontab job or add a ‘time.sleep’ delay to the gengpioblink.py routine immediately before it initiates the reboot command.

 

If in the future you can add exception handling on comm failure that resets/reconnects the serial port and attempt to resume communications, I think that would be a more reliable solution than doing reboots and time delays.   Please add that to the wish list.

 

Which logs would you like me to send you?

 

Also, would it help if I sent you my test program?  I adapted your “modbusdump.py” program to send and receive requests over the serial line to do an end-to-end test of the connection.  I have exception code that closes and re-establishes the connection when there is a comm failure and it recovers gracefully from unplugging/plugging the USB cable.  It should work with an RS232 serial connection as well although I haven’t tested it in that mode.

jgyates, 2022-08-11:
all would be useful if you send them from the about page (assuming outbound email is working) if not then myserial.log and mymodbus.log at a minimum

jwtaylor310, 2022-08-11:
I pulled the USB connection at approximately 1830 and have sent you the logs from the ‘about’ page.

jwtaylor310, 2022-08-11:
I just reconnected the USB cable and have sent you another set of logs….

jgyates, 2022-08-11:
I have not received any logs yet. Your outbound email must be setup and working to submit logs from the about page.

jwtaylor310, 2022-08-11:
That’s odd … I receive alerts via email sent via an external mail server without a problme.  In any event, I have attached the mymodbus and myserial logs.  These include the disconnect at approx 18:31 and reconnect at approx. 18:33.    I noticed this line near the end of myserial.log:

 

2022-08-11 18:33:05,933 : Error in init: Error on open serial port /dev/ttyUSB0: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0' : myserial.py:99

 

The time stamp suggests that error took place after the cable was reconnected.  In my test program I found that if I didn’t explicitly close the serial port in Python after the USB cable was disconnected, when I then reconnecting the USB cable the connection would be assigned to /dev/ttyUSB1 instead of USB0.   That may explain the ‘no such file’ error.

jgyates, 2022-08-12:
I checked in a change that should help with the USB unplug/replug issues. The serial code was doing a close / reopen on the port but it looks like it needed to do it in a couple of places instead of one.

also, I added a setting on the advanced page named "Force Use of Serial Port on Errors". Try enabling this setting. If enabled the software will continue to use the serial port even if it is not found initially on startup.

Just update from the About page to get the latest code. The version number has not been updated yet.

jwtaylor310, 2022-10-11:
The updated program now recovers gracefully when the USB cable is disconnected and then reconnected.  It also emails a ‘Communications Restored’ notification which I have never seen before.  Excellent!  

 

With the “Force Use of Serial Port on Errors” on and my patched gengpioledblink.py program enabled, when the USB cable is disconnected the Pi will repeatedly reboot until the USB cable is reconnected.  Once the USB cable is reconnected, genmon runs normally.  I have a time.sleep() command in gengpioledblink.py which is executed immediately before the reboot which I can use to throttle the frequency of the reboots (or I could add a delay to the crontab entry as you previously suggested).  So, this too works without any problems.

 

I expect that the genpioledblink.py reboot hack won’t be necessary now that the main program handles comm errors gracefully.  I am going to run the program with the add-on disabled for now to see whether there is any need for the modified genpioledblink.py. 

 

THANK YOU for the fast and effective support!