jgyates/genmon #745: No yearly run hours for ver 1.18.13

sgtmaj27, 2022-08-13:
### Expected Behavior
From Change log: Added run hours in the last year to the maint page. Note that this the last 365 days, not the last calendar year.

### Actual Behavior
No yearly run hours listed

### Steps to Reproduce (including precondition)
In checking the log files I found that the controllertype was blank in the genmon.conf.  I tried saving the conf file again from the genmon web page but continued to get the error.  I then edited the genmon.conf file in \etc\genmon folder using the same controller type listed in the tools settings from the genmon webpage (generac_evo_nexus)  this got rid of the error in the log but still not yearly hours listed.

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

- Generator Model:  5870
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.13

jgyates, 2022-08-13:
The data for run hours over a specific period of time is taken from the power log. Your generator is a nexus model and the controller does not report power so the power log is not supported for Nexus controllers. You could get power into genmon if you used the current transformer HAT from [Pintsize.me](https://pintsize.me/store/ols/products/generator-sensor-hat) and the add on for the HAT.

closed 2022-08-13 by jgyates

sgtmaj27, 2022-08-13:
Thanks….I missed that the change was not supported by Nexus controllers….thanks