jgyates/genmon #746: gentankdiy for really external monitor

bdwilson, 2022-08-20:
Hi there. I have a Tank Utility monitor that has been doomed by several water leaks. It is finally in a state that it will not recover. I would like to use the rochester connector from the Tank Utility device, connect it and power it from a Wemos D1 mini (which has analog input, can be put into deep sleep,  and uses minimal power).  I've bench tested this and confirm that I can get data out of the hall sensor.  My question is how to get that data into genmon. I plan to wake the device up from deep sleep every 3 or so hours and will have it post the data to a web server running on the genmon pi perhaps?  I'd like to figure out how to do the right computations to get that data into genmon.  The numbers I get out of the direct analogRead command on ESP32 is ~880.  Can you point me in the right direction for how to get this info into genmon and if you have guidance on calculations (or configurations) to use for genmondiy.conf?  Thanks!

jgyates, 2022-08-20:
Yes, there is a format for importing tank data into genmon. The gentankdiy.py program does this and is documented here:

     https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentankdiypy-optional

Here the wiki article for both approaches that work with the existing add on programs:

     https://github.com/jgyates/genmon/wiki/Appendix-L-Adding-a-Propane-Fuel-Gauge-to-Genmon

The source for this add is here:

     https://github.com/jgyates/genmon/blob/master/addon/gentankdiy.py

The code that actually does the importing of the tank data into genmon is here:

    https://github.com/jgyates/genmon/blob/1600b74bec327ea0d54f5c594121510ea7aa21e9/addon/gentankdiy.py#L136

The library that has the sensor collection and conversions of the data to a percentage are in this module:

     https://github.com/jgyates/genmon/blob/master/genmonlib/gaugediy.py

In short, gemon accepts the tank data via a socket call (myclient.py encapsulates the socket calls and the other add ons use the myclient.py module to communicate with genmon. The socket call sends a python dictionary in JSON format as the data payload.

The python dictionary looks like this:

  dataforgenmon = {
        "Tank Name" : "External Tank",   # Ignored for now
        "Capacity" : 0,                # Ignored for now, tank capacity is taken from the Tank Size setting on the Settings page
         "Percentage" :  80        # decimal percentage (e.g. 80 = 80% full)
}

Let me know if you have any questions.

bdwilson, 2022-08-21:
Thank you! So there is a socket listening that I could send the data directly to from the device when I wake it up? 

I guess I need to figure out how to calibrate it and handle that in arduino land.  

If I figure it out, I will contribute back my findings.

jgyates, 2022-08-21:
The code that sends the socket data to genmon is here:

    https://github.com/jgyates/genmon/blob/master/genmonlib/myclient.py

The server side is here:

    https://github.com/jgyates/genmon/blob/0085b7ff470d59fb60cf0353b27304a22347adf7/genmon.py#L1046

the socket number is 9082 by default. The app /genmon/ClientInterface.py is a command line app that uses the myclient.py library to send commands. The command to send temp data is:

       "generator: set_tank_data='{"Tank Name": "External Tank", "Capacity": 0, "Percentage": 80}'"
       
Send the above string over a TCP socket 9082 to set the tank data. When you open the socket you will read a small amount of data from genmon (base status of genmon). then you can send the above command.

jgyates, 2022-08-21:
I am going to close this issue but feel free to post to this thread with any other questions on this topic.

closed 2022-08-21 by jgyates

bdwilson, 2022-09-30:
@jgyates is there a way to log all external commands? I've got my solution and been testing it, but I've noticed that External Tank percentage falls back to 50% at some point, so I'd like to track successful connections from my ESP monitor to know when it is getting updated remotely.  I'll fully document this when I get it to a state suitable for sharing.

jgyates, 2022-09-30:
Hi @bdwilson,

That functionality is not in code but you cold add one line to temporarily make it log the commands. If you add this line:

     self.LogError(Lookup)

at this line:

https://github.com/jgyates/genmon/blob/27529daf7a906e87613db95a8edc59550b905168/genmon.py#L622

Put it right above the line that looks like this:

   ReturnMessage = ExecList[0](*ExecList[1])

If you want the entire command string, not just the command used then instead of this:

     self.LogError(Lookup)

use this:

    self.LogError(command)

Let me know if you have any questions.

This will print the command (i.e.

bdwilson, 2022-10-01:
Thank you! We are without power from hurricane Ian, so this is a great time to test! Thank you again for genmon, it's crucial in times like this.

bdwilson, 2022-11-22:
Coming back to note that I posted my Hall Effect sensor ESP32 code that will directly send data to Genmon.  Thanks again for the API that makes this possible. 

https://github.com/bdwilson/esp-propane-sensor

jgyates, 2022-11-22:
Very nice. I will add this link to the wiki

jgyates, 2022-11-22:
https://github.com/jgyates/genmon/wiki/Appendix-L-Adding-a-Propane-Fuel-Gauge-to-Genmon#arduino-esp32