jgyates/genmon #747: Update failure

skipfire, 2022-08-22:
I was updating to .14 today from .11 and it stalled, I could not log into the Pi via SSH either.  Power cycled and the Pi loaded, but Genmon isn't.  I am able to access it via SSH for a few minutes and git pull says it is up to date.  Based on logs I am trying `pip install fluids`, but that's not doing much as the CPU is pegged with cc1plus.

### Expected Behavior

Genmon should load

### Actual Behavior

Genmon isn't loading, trying to restart 

### Screenshot or Pictures relating to the problem (if possible)

Web returns ERR_CONNECTION_REFUSED

`
genmonpi@pi14:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts

genloader already running.
`

genloader.log
`
2022-08-22 07:31:40,341 : Installing cmake...
2022-08-22 07:32:04,394 : Error in ExecuteCommandList  : b'debconf: unable to initialize frontend: Dialog\ndebconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)\ndebconf: falling back to frontend: Readline\ndebconf: unable to initialize frontend: Readline\ndebconf: (This frontend requires a controlling tty.)\ndebconf: falling back to frontend: Teletype\ndpkg-preconfigure: unable to re-open stdin: \n'
2022-08-22 07:32:06,948 : Warning: required library mopeka_pro_check not installed. Attempting to install....
2022-08-22 07:32:30,527 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 07:48:07,122 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 08:22:26,570 : Warning: required library fluids not installed. Attempting to install....
`
7:31 corresponds to the upgrade attempt, 7:48 and 8:22 are the hard reboots.

### Your Environment

- Generator Model: 22kW Evo2
- Genmon Logs: not loading
- Genmon Version: V1.18.11 updating to V1.18.14

jgyates, 2022-08-22:
How long are you waiting for the upgrade to take? The release notes do state that this release will take longer to install.

cc1plus is a compiler that the fluids install needs to compile. From the console type this:

     sudo pip3 install fluids

This assumes you are using python3. Do you see any errors?

skipfire, 2022-08-22:
It became unresponsive during the update and I couldn't SSH in so after about 15 minutes I had power cycled it, second round was over 30 minutes.  I'm 15 minutes into this attempt with the manual install of fluids and it appears to be unresponsive again, I'll just let it sit for an hour and see if anything happens.

jgyates, 2022-08-22:
If it takes 15 min that is too long unless you have an exceptional slow internet connection.

Have you tried updating your pi software?

        sudo apt-get update
        sudo apt-get  dist-upgrade

skipfire, 2022-08-22:
I'm not going to be able to do apt update at this point, it becomes unresponsive too fast.  I updated all my Pis somewhere in the last 2 months.

jgyates, 2022-08-22:
That is very strange that an install of a this python package would cause your system to be unresponsive. How old is your SD card? 

There are ways to not run the cron job that starts genmon on boot, but you [would need a keyboard installed to do that](https://askubuntu.com/questions/1197425/how-to-boot-ubuntu-without-triggering-cron-job).

Alternatively, if you are able to log in at all you could type : 

       sudo systemctl disable cron

and then reboot or power cycle. That should stop cron from running. You would need to re-enable it with 

    sudo systemctl enable cron

once this issue is resolved.

skipfire, 2022-08-22:
I'm not sure how old the SD card is, I think probably 9 months.  I did disable CRON and rebooted and pip3 is doing better now with cc1plus not running.

skipfire, 2022-08-22:
I've got that running and it just seems to stall on spicy.  Used the --no-cache-dir option as well.  Download goes fast as I would expect, it's installing build dependencies it just stalls on.  Since the system is responsive with the cc1plus not running I'm going to let it just keep trying to install for now.

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting fluids
  Downloading https://www.piwheels.org/simple/fluids/fluids-1.0.21-py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 2.1 MB/s
Collecting numpy>=1.5.0
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.23.2-cp39-cp39-linux_armv7l.whl (12.3 MB)
     |████████████████████████████████| 12.3 MB 2.2 MB/s
Collecting scipy>=0.9.0
  Downloading scipy-1.9.0.tar.gz (42.0 MB)
     |████████████████████████████████| 42.0 MB 1.2 MB/s
  Installing build dependencies ...
```

jgyates, 2022-08-22:
cc1plus is part of the c++ language complier. I am assuming that some part of fluids (or it's dependancies) is using that.

skipfire, 2022-08-22:
Thanks, I can always grab my configs and rebuild if necessary.  And just in case it becomes relevant at any point, I'm running on a Pi Zero 2.

jgyates, 2022-08-22:
Another thing that *might* be relevant: what version of pip are your running? Type "pip3 -V" to show this. this is my output:

    pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)

skipfire, 2022-08-22:
I also have 20.3.4 python 3.9.

skipfire, 2022-08-22:
Looks like I have a bad Pi Zero 2.  I wouldn't be too bugged if Pis weren't so unavailable right now.

closed 2022-08-22 by skipfire