jgyates/genmon #748: No access to web page

jcherney, 2022-08-22:
### Expected Behavior

Access to web page.

### Actual Behavior

No access to web page. Emails seem to work just fine.

### Your Environment

- Generator Model: {Please write here}  
- Genmon Logs: Submit via the About page  Can't get to about page to send.
- Genmon Version: {Please write here}  Can't see it.

Had a critical RPM error the other day that sent text and emails for notification, thankfully. But I tried to get into the webpage to look at specifics, with no success. Pulled the power twice, but nothing different.

Any suggestions please? 

Thanks.

jgyates, 2022-08-22:
I am assuming that you are trying to access the web interface from your local network using a local IP address. If that is not the case then try that first. 

Next try logging into the pi via ssh. If the pi is not at the IP address you are expecting but you are getting emails form the pi, then your router has changed the IP address of your pi. when you restart your pi you should get an email address with the IP address of the pi in the body of the email so this will let you know the IP address the software detects. Make sure the IP address of the ssh login matches the IP address used on the web interface address.

If you can log into the pi via ssh then type this at the console prompt:

      cd genmon
      ./startgenmon.sh restart

You should see an output like this:

     Restarting genmon python scripts
     Stopping....
     Stopping genserv.py
     Stopping genmon.py
     Starting....
     Starting /home/pi/genmon/genmon.py 
     Starting /home/pi/genmon/genserv.py 

Note that you may also see other lines in the output for the add on programs you have enabled.

If you do not see "Starting genserv.py" then you entry in /etc/genmon/genloader.conf is not enabled to load this. Let me know if that is the case and we can correct it.

If you do see "Starting genserv.py" but you still can not access the web interface then we will need to look at your genserv.log file. To view that type:

         cat /var/log/genserv.log

Copy and paste the output to this thread or attached the file to the gitbhub thread.

Another possibility is that there was a change in your network and your router is blocking the web server on your local network.

rnmpk, 2022-08-22:
I have the same issue. Everything was working, I tried upgrading to V1.18.14, the upgrade hung at the end, waited 30 minutes. Couldn’t reach the web page. Tried pulling the power, same thing. Brought the Pi in, Pi is running, looks like genmon is running, I can ping the address, I can Putty into it, tried running the genmonmaint.sh install script but it stops at random steps every time.

jcherney, 2022-08-22:
Thanks. The IP hasnt changed so I'll check the SSH tonight when I get home.

jgyates, 2022-08-22:
@rnmpk 

It does not sound like the same issue. I don't think @jcherney mentions anything about upgrading to 1.18.14. If genmon is running your can log into your pi via ssh and run the following to send your logs (if your outbound email is setup and working):

      cd genmon
      python3 ClientInterface.py
      
Then at the prompt type:
      
      generator: sendlogfiles

to exit type:

     exit

rnmpk, 2022-08-22:
I wasn't sure whether @jcherney had upgraded and hadn't noticed. Ok, so tried this:

genmonpi@genmon:~/genmon $ python3 ClientInterface.py
Genmon not loaded.
genmonpi@genmon:~/genmon $

but then this:

genmonpi@genmon:~/genmon $ ./startgenmon.sh restart
Restarting genmon python scripts
[sudo] password for genmonpi:
genloader already running.

jgyates, 2022-08-22:
are you using a pi zero?

rnmpk, 2022-08-22:
No, sorry, it's a Pi3, I bought the whole thing with the HAT from PintSize.Me last November.

jgyates, 2022-08-22:
what is the output of /var/log/genloader.log

     cat /var/log/genloader.log

rnmpk, 2022-08-22:
2021-11-10 08:03:53,441 : Updated entry gengpio.conf
2022-01-11 17:38:10,292 : Warning: Missing entry: gencthat , adding entry
2022-01-11 17:38:10,350 : Copying gencthat.conf to /etc/genmon/
2022-02-09 14:45:32,413 : Adding entry: genloader
2022-02-09 14:45:32,426 : Running one time maintenance check
2022-02-09 14:45:47,527 : Found wrong version of pyowm, uninstalling and installing the correct version.
2022-08-05 10:48:50,965 : Warning: required library mopeka_pro_check not installed. Attempting to install....
2022-08-05 10:49:10,297 : Warning: Missing entry: genmopeka , adding entry
2022-08-05 10:49:10,353 : Warning: Missing entry: genmopeka
2022-08-05 10:49:10,360 : Copying genmopeka.conf to /etc/genmon/
2022-08-21 12:58:53,699 : Installing cmake...
2022-08-21 12:59:13,620 : Error in ExecuteCommandList  : b'debconf: unable to initialize frontend: Dialog\ndebconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)\ndebconf: falling back to frontend: Readline\ndebconf: unable to initialize frontend: Readline\ndebconf: (This frontend requires a controlling tty.)\ndebconf: falling back to frontend: Teletype\ndpkg-preconfigure: unable to re-open stdin: \n'
2022-08-21 12:59:15,975 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 13:50:07,149 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 13:59:46,445 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 14:10:57,479 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 14:32:13,154 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:03:37,017 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:27:04,907 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:42:28,519 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:51:37,337 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:53:44,494 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 15:56:07,063 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 16:00:06,141 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 16:10:52,353 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 16:27:15,935 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 16:39:23,707 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 16:57:27,108 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 17:37:57,012 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 18:05:59,173 : Warning: required library fluids not installed. Attempting to install....
2022-08-21 22:48:09,353 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 08:43:05,613 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 08:53:32,508 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 09:09:39,262 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 09:35:10,086 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 09:45:14,490 : Warning: required library fluids not installed. Attempting to install....
2022-08-22 10:00:17,994 : Warning: required library fluids not installed. Attempting to install....
genmonpi@genmon:~/genmon $

jgyates, 2022-08-22:
Looks like your system is having an issue installing the "fluids" library, lets try this:

First lets disable the cron job from loading genmon on boot:

     sudo systemctl disable cron

Note, when we are done we will need to run this command to undo this:

     sudo systemctl enable cron

Once you disable cron, then reboot the pi

     sudo reboot

Once it reboots, then run these commands:

     sudo pip3 install numpy

      sudo pip3 install scipy

     sudo pip3 install fluids

Let me know if any of these lock up or hang the system. You can open another ssh window to see if the system is locked or not.

rnmpk, 2022-08-22:
I get this right from the start:

genmonpi@genmon:~/genmon $ sudo systemct1 disable cron
[sudo] password for genmonpi:
sudo: systemct1: command not found

jgyates, 2022-08-22:
replace the 1 with a lowercase L. The command is systemctl

rnmpk, 2022-08-22:
Sorry about that, I'm not a big Linux guy but should have noticed that. Numpy installed correctly, hangs on installing scipy tried twice:

Collecting scipy>=0.9.0
Using cached scipy-1.9.0.tar.gz (42.0 MB)
Installing build dependencies ... /

jgyates, 2022-08-22:
ok. Lets do this:

1) I checked in an update that does not install fluids (which as scipy as a dependancy)

       ./genmonmaint.sh -u -n

This will update to the latest code.

2) re-enable cron

       sudo systemctl enable cron

3) reboot

        sudo reboot

Then see if everything is running OK.

I am changing the install of "fluids" to be a manual process

rnmpk, 2022-08-22:
Ok, that looks like that fixed it. I have the web interface back. No graphics on the status page, but I'm assuming that's because it's not plugged into the generator. I'm going to re-install it.

jgyates, 2022-08-22:
I am not really sure why there is a problem installing the fluids library on some systems. It is used with the new add on and genloader checks for uninstalled libraries and installs them if needed. For now I am removing the check in genlaoder for the fluids library so if any wants to use the genmopeka add on they will have to manually install fluids via the install script or by had with the pip command.

I am going to close this issue for now but feel free to post here if you have any issues or questions on this topic.

rnmpk, 2022-08-22:
It's all back to working again. I have to thank you firstly for your help and also for an incredible piece of software.

Don331, 2022-08-22:
The update blew mine up too. Pi 3b from PintSize.
Going through the debugging above, at the 
 sudo pip3 install numpy
  sudo pip3 install scipy
 sudo pip3 install fluids

part, install scipy has been running for 30+ minutes so far. PI is still responsive (using another ssh window,   but incrediblly slow. Running a df command takes a couple minutes.

Don331, 2022-08-22:
Update:

Did
 sudo systemctl disable cron
cd genmon
./genmonmaint.sh -u -n
sudo systemctl enable cron
sudo reboot

And it came up. I'll hook it back to up the genset and see!

rnmpk, 2022-08-22:
Great! I'm glad I hijacked the thread!

skipfire, 2022-08-22:
Looks like scipy is a known slow builder.  https://raspberrypi.stackexchange.com/questions/94585/installing-libraries-via-pip-is-taking-forever-raspberry-pi-3b

I'm in the middle of a fresh buildout and installing fresh genmon still does that package, I'm going to just let it for for up to 4 hours and see what happens.

jgyates, 2022-08-22:
interesting, I did my testing on RPi3 and RPi4. I did not see any slow building. I wonder if there are other variables involved. Has anyone tried the suggestion in the link provide by @skipfire? Try using this to install scipy:

   sudo pip3 install scipy --extra-index-url https://www.piwheels.org/simple

Then install fluids

     sudo pip3 install fluids

jgyates, 2022-08-22:
another think that is interesting is that I am seeing multiple versions of scipy on my different test systems

Pi4 with 64bit Raspbian 11 os has scipy V1.9.0
Pi4 with 32 bit Raspbian 11 os has scipv V1.8.1
Pi3 with 32 bit Raspbian 11 os has scipv V1.8.1
Pi3 with 32 bit Raspbian 9 os has scipy V0.18.1

I tried this 

     sudo pip3 install scipy --extra-index-url https://www.piwheels.org/simple
     sudo pip3 install fluids

on a pi3 running Raspbian 11 and it worked, it took about 10 min. Also, there was one error on installing scipy but it reverted to downloading the binaries and finished the install. once I installed scipy, installing fluids took about 3 seconds.

jcherney, 2022-08-22:
I ran the restart thru SSH and it's now working. I guess I was the lucky one!  Thank you!

skipfire, 2022-08-22:
After a couple hours it finally finished and had some output of errors building ninja as part of scipy.  Not sure if that is related to the speed difference.  The most interesting thing is the new build once finished immediately had email working for me which hasn't worked for months so I may have had some other library out of date before.

jgyates, 2022-08-22:
@skipfile,

FYI, you can also update your installed libraries that genmon uses with this command:

      ./genmonmaint.sh -r 

Everyone,

For now the way it works is this:

1) The install script (genmonmaint.sh -i ) will attempt to install fluids (which includes scipy). The assumption is that new installations will not have a problem installing scipy as that was one of my test points. 
2) genloader will not check for fluids or try to install it if the fluids library is not found 
3) genmopeka.py plug in will check if fluids is installed and exit if it is not found
4) the wiki entry for genmopeka.py has been updated to include the installation requirement regarding fluids

Also, fluids is only required for the genmopeka plugin. Other than that it is not used by genmon.

I am going to close this issue for now but if anyone has any additional data

rwskinner, 2022-08-22:
I'm sure sorry that I caused all these headaches....

Richard

jgyates, 2022-08-22:
No worries, it is a process. Thanks for the input everyone

closed 2022-08-22 by jgyates