jgyates/genmon #751: GenMon stops responding periodically

jjmacneil, 2022-09-01:
Hello,

### Expected Behavior

I should be able to visit the GenMon web interface to login to the GenMon application, but after the last GenMon update I noticed I lost access to the web interface after a period of time

### Actual Behavior

No response from GenMon web interface until I reboot the Pi via SSH using terminus. Upon reboot the GenMon web interface is available again for an undetermined amount of time.

### Your Environment

- Generator Model: Generac 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.14

jgyates, 2022-09-01:
Depending on when you updated this issue is likely the one listed in this thread:

https://github.com/jgyates/genmon/issues/748

If you updated to 1.18.14 early then try updating the software again. In the initial version 1.18.14, genloader was trying to load the "fluids" python library but some version of raspbian have issues with this. An update was made to genloader so the "fluids" is not installed if it is missing. You can update form the command line via ssh like this:

      cd genmon
      ./genmonmaint.sh -u

Then type 'y' for yes.

If this does not correct your issue then I would need more info as nothing changed in the web interface.

If you still have an issue make sure that you are accessing the web interface from your local network. Also try disabling HTTPS and see if that makes a difference. I have seen some issues with some routers blocking access on the local network. Some of these are described here:

     https://github.com/jgyates/genmon/issues/737

and here:

     https://github.com/jgyates/genmon/issues/271

Take a look at the two links above and see if any of these suggestions make a difference.

jgyates, 2022-09-04:
@jjmacneil,

I am going to close this issue for now but if you have any additional data or feedback feel free to post to this thread.

closed 2022-09-04 by jgyates