jgyates/genmon #752: Feature Request - Pushover Add-On, Priority Parameter

rnmpk, 2022-09-04:
Would it be possible to implement the Priority Parameter from the Pushover API for each of the notification categories listed in the Pushover Add-On?

For example, a software update notification could be set to low priority, so no vibration or sound on a smartphone using Pushover, but an error or warning could be set to High Priority, overriding a smartphone's do-not-disturb and mute settings and could play a different (or custom) soundfile at a particular volume level. A power outage could be set to Emergency Priority, requiring an acknowledgement to stop the notification.

https://pushover.net/api#priority

skipfire, 2022-09-04:
I'd be curious about priorities and the individual alerts.  If there is a power outage and my generator starts and transfers that's not an emergency to me, just information.  Likewise, if the generator does a self-test and fails to start then it's an error, but still not an emergency.  Combine them (power out and generator fails to start) then that's an emergency to me.  Of course I fully recognize others will likely have different views on the alerts.

jgyates, 2022-09-04:
Let me look into this. Sounds do-able at first glance.

jgyates, 2022-09-05:
I checked in an update that will add priorities for the pushover notices. Once you update to the latest, see the Add page for the settings. Let me know if you have any questions or issues.

closed 2022-09-05 by jgyates

rnmpk, 2022-09-05:
That's incredible. Thank you for that. Is there somewhere I can refer to as to the definitions of Service State, System Health, Run State, etc.? Skipfire, agree with you, although some " if/and/or/then" steps might better be handled through some home auto software (I'm using Omni Pro and HomeSeer) but if System Health or Service State or Run State handles that condition, maybe that will work.

jgyates, 2022-09-05:
These are the conditions that are notified. These are handled via the /genmon/genmonlib/mynotify.py module. These short message notices are used with gensms, gensms_modem, genpushover, genslack and gensyslog add-ons:

Alarm - when alarms or warnings are reported by the controller
Software Update - When a software update is available
System Health - This is the reported when there is a change in the Monitor - > Generator Monitor Stats ->  Monitor Health state. Things like serial communication problems or internal genmon software errors are reported
Fuel - If your controller supports fuel consumption calculation low (below 20%) this state is active
Outage - If your controller supports utility voltage then this is active if there is a change in the outage state. Outage is reported if the utility voltage drops below the threshold voltage and is normal if the utility voltage is above the pickup voltage
Run State - Ready(not running but in Auto mode), Running, Exercise or Running in Manual Mode are the run states. A change in run state is notified 
Service State - Active is service is due with the controller (Evolution and Nexus only)

rnmpk, 2022-09-05:
Thanks for that, very useful.