jgyates/genmon #754: kW Output no longer graphing

skipfire, 2022-09-04:
I had my Evo2 controller replaced a few days ago, and when running a transfer test today I noticed that the graph isn't populating.  The power output and output voltage are both reading correctly so the data is being read from the controller.  The graph does not seem to redraw the content if I change the size of the box or the time period being looked at.

Using the dev tools in the browser there are no console messages and no network entries with anything other than 200 results.  I've tried 2 different browsers and restarting genmon.

### Expected Behavior

kW Output graph to populate

### Actual Behavior

kW Output graph not populating

### Steps to Reproduce (including precondition)

Running a start with transfer

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: G0070432 22kW air cooled Evo2
- Genmon Logs: Submitted
- Genmon Version: V1.18.14

jgyates, 2022-09-04:
Hi @skipfire 

I think I introduced this bug with the fix for this issue: https://github.com/jgyates/genmon/issues/742

I checked in an update that should fix this. Update from the about page and let me know if you have any issues.

Thank for finding this,
Jasno

closed 2022-09-04 by jgyates

skipfire, 2022-09-04:
That did fix it, thanks.