jgyates/genmon #756: Genmon does not connect to Generator

RRoyceus, 2022-09-07:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

ls –l /dev/serial* returns 

lrwxrwxrwx 1 root root 7 Apr 12 01:06 /dev/serial0 -> ttyAMA0
 lrwxrwxrwx 1 root root 5 Apr 12 01:06 /dev/serial1 -> ttyS0

### Actual Behavior
 ls: cannot access '–l': No such file or directory

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

skipfire, 2022-09-07:
That looks like you have a long dash `–` rather than a short dash `-` for the `-l` switch.

`ls -l /dev/serial*`

jgyates, 2022-09-07:
@RRoyceus 

I am going to close this issue as your initial question has been answered however feel free to post to this thread with any other questions on this topic.

closed 2022-09-07 by jgyates

RRoyceus, 2022-09-07:
I do not think it is a matter of the short dash vs long dash: I copied the statement directly from the Wiki, however I will make sure when I get home tonight.

Best Regards,

Jimmy Scruggs

The Scruggs Company
8530 Hansen Rd
Houston, TX 77075

Office: 713-649-2776 x 810
Direct: 832-319-6822
Cell : 713-899-1862
Toll Free: 866-226-8164
Fax: 713-649-1975
***@***.***

This email has been scanned for spam and viruses by Proofpoint Essentials. Click here<https://us1.proofpointessentials.com/index01.php?mod_id=11&mod_option=logitem&mail_id=1662557285-rmOIYsy6s0Gs&r_address=jscruggs%40scruggsco.com&report=1> to report this email as spam.

jgyates, 2022-09-07:
Please check this. This command is not related to genmon, but rather a linux command. If you are running linux (raspbian is a linux operating system for the raspberry pi) you would be able to issue this command.

RRoyceus, 2022-09-08:
Thanks,  I pasted it again with the same result, then erased the dash and re-entered it and it worked.

Best Regards,

Jimmy Scruggs
Office 713-649-2776 x 6
Mobile: 713-899-1862

jgyates, 2022-09-08:
Not sure why the wiki had that strange character in there, it may be that a UNICODE character got embedded in there. I updated the wiki, it should work if you copy and paste now.