jgyates/genmon #757: pip3 install pyopenssl fails with new Raspbian 9-6-2022 Lite version PSA or FYI

liltux, 2022-09-07:
I am just putting this here, not sure if this has been mentioned yet.  I am seeing a build issue with pyopenssl, specifically with cryptography.  Looks like this is a maintainer issue on the cryptography side of things?

[image]

I was able to install rust via rustup.rs.
Then run,
source .bashrc
Then I was able to successfully install pyopenssl via pip3 and did not get errors.
Not sure if this is the recommended way, but it installed without errors this way.  Also I do not believe this is a Genmon issue, but a cryptography issue.
Curious if anyone else has dealt with this yet?

jgyates, 2022-09-08:
I have not seen it however I have not installed a fresh copy in at least a month so it is possible that this is an issue with the latests. It may also be specific to a 32 or4 64 bit version of raspbian.

liltux, 2022-09-08:
I did not even think about it being a 32bit/64bit difference either.  I switched over to using 32bit because I felt there is more support for 32bit with linux.  I just tried the 64bit version and did not get any errors.  Just wanted to share if anyone else ran into it.

jgyates, 2022-09-27:
looks like this issue has been resolved by the maintainer of the package. I just downloaded a fresh copy of raspbian 32bit (lite) and could not reproduce. I am going to close this for now but if anyone has related issues just post here or in a new thread.

closed 2022-09-27 by jgyates