jgyates/genmon #758: Can not access web interface

CHawks78, 2022-09-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

To be able to access the web interface using the pi's ip address

### Actual Behavior

"URL can not be dispalyed"

### Steps to Reproduce (including precondition)

My dsl modem quit working.  I swapped out the modem and verified the new IP address assigned to my pi by looking at the device list while logged into my modem.  I can not log in to the web interface after multiple tries.  Everything else on my network is functioning fine.  The pi is communicating with my generator, I just cant access the web interface.  Im no expert by any means with the pi, and any help will be greatly appreciated.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

skipfire, 2022-09-10:
Are you able to ping the Pi or access it via SSH?

CHawks78, 2022-09-10:
Yes, i can ping the pi with my windows laptop with no errors.  I really am a novice with the pi and linux, so I've never messed with SSH.  My genmon has been working for over a year with absolutely no problems until my modem quit working.  I really didn't have any issues getting it going the first time either.  Thanks for the reply.

liltux, 2022-09-10:
@CHawks78 have you checked your new routers security settings?  The router could be blocking port 8000?  Do you have secure login turned on?  If so you may have to be explicit with https:// before your pi’s up address.

skipfire, 2022-09-10:
Have you rebooted the Pi since the modem replacement?  It is possible that the WiFi reconnected and got a new IP but that GenMon's flask server isn't bound to that IP.

If you want to ssh into it
`ssh pi@ipaddress` or if you got a preloaded unit from PintSize `ssh genmon@ipaddress`.

I also agree with liltux's suggestion of looking at security to see if the port is blocked.

CHawks78, 2022-09-10:
> @CHawks78 have you checked your new routers security settings? The router could be blocking port 8000? Do you have secure login turned on? If so you may have to be explicit with https:// before your pi’s up address.

I am currently logged into my modem looking for anything regarding port blocking....but without much luck.  I've tried logging in with http:// and https:// and neither are working for me.  Thanks for your reply.

CHawks78, 2022-09-10:
> Have you rebooted the Pi since the modem replacement? It is possible that the WiFi reconnected and got a new IP but that GenMon's flask server isn't bound to that IP.
>
> If you want to ssh into it [...]

I have rebooted the pi several times now, but the issue remains.  I tried ssh as you mentioned in my cmd promp on my pc and when it asks for a password, it has a blinking cursor and will not let me enter anything.  I appreciate your input.

skipfire, 2022-09-10:
The password you enter will be hidden, the default password is `raspberry`.  Once you get in you can view the various log files in the `/var/log/` folder.  The first one I would check would be the genmon log `cat /var/log/genmon.log`.  Only the most recent (bottom) entries are likely to be meaningful.

CHawks78, 2022-09-10:
> The password you enter will be hidden, the default password is `raspberry`. Once you get in you can view the various log files in the `/var/log/` folder. The first one I would check would be the genmon log `cat /var/log/genmon.log`. Only the most recent (bottom) entries are likely to be meaningful.

Thanks for the tip about the hidden password (I told you Im a novice!), I tried both the default password, and the user password I created when setting up the pi and I just keep getting "access denied".

CHawks78, 2022-09-10:
> The password you enter will be hidden, the default password is `raspberry`. Once you get in you can view the various log files in the `/var/log/` folder. The first one I would check would be the genmon log `cat /var/log/genmon.log`. Only the most recent (bottom) entries are likely to be meaningful.

Ok, I went back and read what you had said about the ssh being different if ordered from pintsize, and it made me think to not use pi but my user name..again, Im still learning.  Anyway, Im logged in now via ssh on my laptop.  I never could find any info on my modem about port blocking.  Am I correct to assume it's not being blocked if I can log on via ssh?  Any commands that you recommend I run via ssh that would help diagnose?  Thanks again.

skipfire, 2022-09-10:
Being able to log in via SSH would not indicate anything about port 8000 being blocked.

Run `cat /var/log/genmon.log` to see if there are any errors recorded there.  The entries are timestamped so just stuff since your last reboot would be adequate.

For your Pi's IP, is it a public IP or a private IP (starting with 10.x.y.z or 192.x.y.z)?  If you are trying to hit a private IP it isn't likely that the port is being blocked.

jgyates, 2022-09-10:
@CHawks78 

what is the IP of the pi and what URL have you been using in your browser to attempt to access the web interface?

Do you have a password setup for the genmon web interface (i.e. is HTTPS web security enabled)?

It is not uncommon for routers to block local web servers on your network if you have high security settings. Are you using a DSL modem / router or do you have a separate modem and router. Knowing the model of your router may help.

CHawks78, 2022-09-10:
Its a 192.168.x.xx address, and when I ran the command you mentioned I dont see anything that looks weird to me.  I'll also admit that I was wondering If I somehow screwed something up on my pi entering the new wifi password, so I formatted the sd card and did a fresh load of genmon.  Same issue remains.

CHawks78, 2022-09-10:
> @CHawks78
>
> what is the IP of the pi and what URL have you been using in your browser to attempt to access the web interface?
>
> Do you have a password setup for the genmon web interface (i.e. is HTTPS [...]

Pi ip is 192.168.0.20 and I've tried using "http://192.168.0.20" and "https://192.168.0.20".  I did a fresh reload of genmon after this issue started today just to make sure I hadn't accidently fat fingered something when entering the new wifi password.  It didn't resolve my problem so I haven't been able to change anything within genmon because Im only use to making changes through the web interface...and I can't get that to work.  I am using a dsl modem-router and it may very well be the culprit, I just am not having any luck finding the problem.  It is a centurylink technicolor c1100t if that helps any?  I appreciate any help!

jgyates, 2022-09-10:
The default method for accessing genmon using this IP address is http://192.168.0.20:8000

If you do not specify the 8000 at the end of the URL the browser will try to access port 80. Genmon defaults to port 8000 so it will not interfear with any other sites possibly hosted on the pi

Can you access genmon with this address:   http://192.168.0.20:8000

CHawks78, 2022-09-10:
> The default method for accessing genmon using this IP address is http://192.168.0.20:8000
>
> If you do not specify the 8000 at the end of the URL the browser will try to access port 80. Genmon [...]

You found my problem!  Im not sure why it wouldn't work the first time I changed my wifi password, but after the reload, I completely forgot about the :8000 after the ip address.  I remembered it from setting it up the first time last year as soon as you mentioned it.  I am truly thankful for everyones help on this issue!

jgyates, 2022-09-10:
I am glad it is working. Thanks for the help everyone.

closed 2022-09-10 by jgyates

rwskinner, 2022-09-10:
Try adding the port to it like below
http://192.168.0.123:8000/