jgyates/genmon #760: 14KW Model 7223 Ambient Temp Incorrect

dmjpixel, 2022-09-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Ambient temperature would be close to temp reported in MobileLink as well as current weather API input

### Actual Behavior

Ambient temp has been higher - may have started to display incorrectly (not positive) after new firmware 1.15 was pushed to generator this evening

### Steps to Reproduce (including precondition)

Ambient temp in genmon showing as 82 F on maint screen; current temp is 71.53 F on Monitor screen; MobileLink app showing 73F; Register showing Lo 082

### Screenshot or Pictures relating to the problem (if possible)

N/A

### Your Environment

- Generator Model: 7223
- Generator Firmware: 1.15.176
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.14

dmjpixel, 2022-09-11:
Activated new generator today - installed pre-built pi cap that I purchased along with this software...it is awesome!  Thank you.

dmjpixel, 2022-09-11:
Thinking about this a little more - I have the PI in a case sitting between battery and gas intake..could heat from the Pi raise the temp sensor...no idea where it is located.  If MobileLink uses a local weather source as opposed to the register that could explain why it is showing a more accurate temp.

jgyates, 2022-09-11:
Yes, the ambient temp is not the outside temp. The ambient temp typically goes up if the engine is running as this sensor is in the wiring harness inside the generator enclosure close to the engine. You can use openweather map (configure this on the settings) page to get the outside temp.

jgyates, 2022-09-11:
going to close this for now but feel free to post here if you have any other questions on this topic.

closed 2022-09-11 by jgyates