jgyates/genmon #761: Constant serial issues

danodemano, 2022-09-12:
This has been a re-occurring problem for the past few months (it wasn't an issue in the ~2 years prior). I get constant "not receiving data" alerts in Pushover. They are followed almost instantly by "OK" notices. I've swapped hardware, confirmed cabling, swapped from pin 4 to pin 2 for the ground, even played with the timeouts in the advanced settings. No matter what I do this issue persists. I'm unsure if it's even a problem or just an erroneous error. So this is either a bug report or a troubleshooting help request, lol. :D Any ideas appreciated. I was inside the generator today and all the connections are solid, no corrosion or anything. Can provide more detail if needed. Thanks!!!

### Expected Behavior

Stable serial connection

### Actual Behavior

Unstable serial connection

### Steps to Reproduce (including precondition)

N/A

### Screenshot or Pictures relating to the problem (if possible)

N/A

### Your Environment

- Generator Model: Generac 22kw with evolution 2.0 controller
- Genmon Logs: Done
- Genmon Version: V1.18.14

jgyates, 2022-09-12:
I received your logs. One thing that I noticed is that your comms stats show not only timeout errors, but CRC errors. Also noticed that your pi is reporting some voltage issues. This is from your log submission and you should be able to see this on your Monitor page:

  Pi CPU Frequency Throttling : Has occurred. 
   Pi ARM Frequency Cap : OK
   Pi Undervoltage : Has occurred. 

How are you powering your pi? Under-voltages and CPU throttling are signs of power issues.

when you said you changed hardware, does that include your power source?

danodemano, 2022-09-12:
Interesting, I didn't manage to find that. I swapped the serial adapter but the Pi is powered via PoE. The switch hasn't caused any other issues so perhaps it's my PoE adapter (that takes the -48v to 5v). I'm going to swap that and report back, thanks for digging into it and for the amazing software!

jgyates, 2022-09-12:
A 2.5 Amp 5V power supply is recommend for the  pi 3 B  so make sure your PoE covers that range. I am going to close this issue but feel free to post to this thread with follow up questions or data.

closed 2022-09-12 by jgyates

danodemano, 2022-09-17:
I swapped the PoE adapter on Wednesday and this issue has yet to resurface. I guess their lifespan is about 2 years, lol. I have not gotten any communication errors and no undervolting or CPU throttling has occurred. Wouldn't have guessed that to be the problem but here we are. :D Thanks so much for the assistance and awesome software!!

jgyates, 2022-09-17:
I am glad it is working now. Thanks for the follow up. This helps others who might have similar problems.