jgyates/genmon #762: Add ACME and "Let's Encrypt"

MiddleSiggy, 2022-09-12:
NEW FEATURE:

It would be nice to have the integration of free web certificates using the Automated Certificate Management Environment ("ACME") and "Let's Encrypt".

I set up my generator monitoring site at genmon.<mydomain>.com and have recently turned on HTTPS, but now I get the self signed error message.

I have recently utilized the ACME and "Lets Encrypt for my pfSense firewall, which allows for a seamless integration of certificate provisioning and integration with the web server. (for a reference point) https://www.youtube.com/watch?v=Tc_8PAE8S28

This would be a very nice integration for those of us that run the site under HTTPS and don't want the error messages or to pay for the certificates for home use.

As always, thank you for all the hard work, and I love what you have done.

Let me know if I can help in any way.

Thank you.

jgyates, 2022-09-12:
There is a wiki entry on using Let's Encrypt with genmon: https://github.com/jgyates/genmon/wiki/Appendix-K-Adding-LetsEncrypt-certificates-to-Genmon

If you have any updates to the wiki entry I would be glad to incorporate them.

jgyates, 2022-09-13:
I am going to close this issue for now but if you have additional questions or followup please feel free to post them here.

closed 2022-09-13 by jgyates