jgyates/genmon #763: Genmon not recognizing current engine state

skipfire, 2022-09-12:
Genmon did not show the generator as running during my weekly idle test, it showed as Auto, Off-Ready.  The screenshot was taken while the generator was running.

Using Start Generator from the maintenance page it did show the generator as running.

The monitor page shows 0 errors.

In this past it shows the correct status in Genmon during weekly idle tests, this is the first weekly test since I got my new controller.

[image]

### Expected Behavior

Genmon should see the correct state

### Actual Behavior

Genmon didn't recognize the generator was running

### Steps to Reproduce (including precondition)

My weekly 5-minute maintenance is when I detected this.

### Your Environment

- Generator Model: 22kW Generac Evo2
- Genmon Logs: Submitted
- Genmon Version: v1.18.14

jgyates, 2022-09-12:
Hi @skipfire ,

A few questions

1) Is there any relevant info in the logs of errors?
2) How is your system wired (direct serial, modbus TCP, is the pi close to the generator, etc)
3) Can you duplicate this or is this just a single instance 
4) when you noticed this, was the monitor time at the bottom of the status page updating every 5 seconds (if not then your web page is not updating which would indicate a browser issue)
5) were any of the gauges moving at all
6) did you get any notices regarding generator changing state (emails or txt messages) if you have that setup to do so

skipfire, 2022-09-12:
1. none of the logs show anything around that time
2. wired direct serial with a 1M cable
3. I'll look at duplicating, it happened with the weekly test but not when manually started with Genmon
4. I did not notice, but I did do a full refresh and I switched computers to take the screenshot so I can rule out anything client specific.
5. CPU temp, WiFi signal, and battery moved a bit, but none of the running gauges moved.  Usually I see 0.02kW output when it is idle and did not see that.
6. No messages about state change from the weekly test, but when I started with the maintenance page the status text, gauges, and notifications all worked correctly.

closed 2022-09-12 by skipfire