jgyates/genmon #764: Modbus

BillyEll, 2022-09-13:
If the generator already talks modbus RS-485 can I just control it via Modbus? Will you share the modbus registers?

jgyates, 2022-09-13:
@BillyEll 

I guess I need more info to answer your first question. This project supports a few different controllers. All of the modbus registers are in the source code of this project. 

What controller are you interested in controlling? Withtout knowing that I don't know how to answer your questions.

All of the code contains register definitions. Depending on the controller, some of that information is available online, other controllers are not published by the manufacturer.

BillyEll, 2022-09-13:
I have a Generac RG10090KNAC with a RTSN400K3 transfer switch. And also an older Eaton generator, haven’t looked at it yet, but I’m sure it’s a brand labeled Generac.

But I’m doing is using Ignition Automation and monitoring our small data center via ModBus. I have 2 Generac MG200’s & P&L panel, 4  Eaton Transfer switches, 2 Eaton 93M UPS systems, and 2 Eaton RPP panels, 2 Stulz Air conditioner system.
We also have several stores that have generators and I thought as long as I’m watching the Datacenter, why not watch these generators and make sure they are being exercised, generating power etc. I have another older Generac that has I think an R-Panel? It’s pretty dumb, so I use a Shelly relay I can control to bypass the manual/auto/off switch to start it when I want to and an Eaton Power Expert meter on it so I can see that it is running and generating.
I ran across you Genmon and thought that was pretty cool. But it dawned on me if the generators are already talking Modbus via RS-485 I would probably talk to them directly? So any guidance you can give me would be helpful.
I can talk to the MG200’s directly, but I can’t figure out how to start/stop them via modbus? I can tell the transfer switches to start them via modbus, and the P&L panel doesn’t seem to want to talk Modbus either?
Can I talk to the Genmon via Modbus? Any suggestions are appreciated!

thanks

jgyates, 2022-09-13:
This software communicates with the controller so knowing which controller is used is the first step in know if it is supported.  This page describes the controllers supported: https://github.com/jgyates/genmon/blob/master/README.md

It looks like the RG10090 uses the Evolution controller so yes genmon should work with this. If you are asking if you can just issue modbus commands directly to the generator yourself, yes you can. The evolution specific code is here: https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py

The code for starting and stoping each controller is in the controller specific code. For example for various generac controllers it would be in these three modules:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_HPanel.py#L2124
https://github.com/jgyates/genmon/blob/master/genmonlib/generac_powerzone.py#L1916
https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L1120

I am not sure what controller the MG200 uses. If it uses the Evolution, H panel, G panel or PowerZone controllers then this project will communicate with the controller. 

If you have a specific question about a specific controller let me know.

jgyates, 2022-09-13:
No, genmon does not accept modbus commands. It acts as a modbus slave device, only a modbus master device.

Sorry, you will have to give me more info. I don't know what controller you are using so register 708 does not mean anything without context. What controller are you using? Also, what software is giving you the permission error?  

You can give genmon other types of commands via a socket interface to start and stop the controller and you can also use socket commands to issue modbus register reads and writes, however this is only single reads or writes, not writing multiple registers. An example of this is the ClientInterface.py program:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy

The commands are: 

      generator: readregvalue=0100

to read modbus register 100 hex

     generator: writeregvalue=01f4,aa

to write an AA hex to regsiter 1F4 hex

BillyEll, 2022-09-13:
Sorry, I was trying to write to the Generac MG200 via modbus. The modbus map I have for Power Zone Control Platform shows register 708 is the _remote_start and is 0=stop, 1=start.
But when I try writing to it, Inductive Automation give me some kind of permission error. I haven’t seen that issue talking to other devices via modbus so I was wondering if Generac had some kind of access level? So anyone could read, but needed some kind of authentication to write to certain registers. It’s probably something I’m doing in Inductive Automation?
I’ve tried accessing the Permissive & Load Shed panel but it doesn’t seem to like Modbus? I can access it via the built in web site.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy

BillyEll, 2022-09-13:
> If memory serves you have to enable external modbus commands. I am not 100% sure about that. The Powerzone controller stuff I have only tested myself in a limited capacity. This is one that I had help testing with someone who has access to a powerzone controller.
>
> assuming the 708 is decimal then this look right, but again I have not tested the remote stuff on a powerzone so it may not work.

So I guess that’s something you would have to enable in the generator?
I hate to keep bugging you, do you happen to have a molex connector pin out? I see your using the RS232, but isn’t there a RS485 port as well? Figured I might try a wifi to RS485 converter and see if I could control these other generators directly?

jgyates, 2022-09-14:
Unfortunately I do not have a pin out. I would like to have it to add to the wiki for the powerzone. This is the current wiki:

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

BillyEll, 2022-09-14:
If I can find it, I’ll send it to you

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

jgyates, 2022-09-14:
To my knowledge the only way the powerzone has been tested with genmon is with modbus over TCP via the Ethernet port.

jgyates, 2022-09-14:
I talked to someone with some more knowledge than I have about this and apparently Generac has added modbus over RS485 support to the powerzone sync display. You have to explicitly enable it in the UI of the display. I believe there is a 4 pin header and pins 3 and 4 are RS485A and RS485B respectively.

BillyEll, 2022-09-14:
Someone just sent me this. I don’t know if there is such a connector on the powerzone. But I’m currently logged into the powerzone generators and I can see I have it enabled. I also see under the ModBus TCP there is an option for Authentication. I don’t have it turned on but apparently over TCP it’s an option.

Here is the pinout of the 8-Pin molex for the Evolution
***@***.***

jgyates, 2022-09-16:
I am going to close this thread but feel free to post here with any follow up.