jgyates/genmon #766: Cannot connect through Port

RRoyceus, 2022-09-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Be able to access Genmon remotely through an open port
{Please write here}

### Actual Behavior
I have had Genmon working for many years.  In the past I use port forwarding to view Genmon.  I also use the same method for several other devices.  This past weekend I decided to do a fresh install on a new PI with a new SD card.  The setup went flawless and I am able to view and control the generator using a local IP, but I no longer can access it through port forwarding either locally or remotely.  I made no changeS to my router.  All my other devices can be accessed using DYN.

Any suggestions on what might need to be changed in my setup?

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

griffths, 2022-09-15:
> Links:

RRoyceus,

Has a new local IP address been assigned to your rebuilt Pi?  If so, you 
likely will need to establish a new forward through your router.

That is all I can offer....

Regards,

Jay

jgyates, 2022-09-15:
@RRoyceus 

Genmonn does not control anything related to port forwarding and the socket code has not changed in quite a while (years). The problem is very likely with your router or your routers interaction with the pi / linux. It is possible that your routers is retaining some knowledge about your old pi's IP address and MAC address.  If your pi is using a static IP address, is your router aware that the MAC address has changed since the pi is new. If you need help troubleshooting this issue you will need to proviede some info about your router and how the IP is assigned to the pi (static vs dynamic vs reserved (assigned by the router via DHCP but fixed address every assignment).

RRoyceus, 2022-09-15:
Jay,

Thanks for the reply.  I used the same static IP and port.

Best Regards,

Jimmy Scruggs
Office 713-649-2776 x 6
Mobile: 713-899-1862

—

This email has been scanned for spam and viruses by Proofpoint Essentials. Click here<https://us1.proofpointessentials.com/index01.php?mod_id=11&mod_option=logitem&mail_id=1663206803-CwCM6jqz7uOh&r_address=jscruggs%40scruggsco.com&report=1> to report this email as spam.

jgyates, 2022-09-16:
@RRoyceus 

I would try to remove the port forwarding rule in your router, save the configuration, reboot the router and then re-create the rule.  

I am going to close this issue but feel free to post to this thread with any follow up questions or status

closed 2022-09-16 by jgyates

RRoyceus, 2022-10-11:
Problem solved.  I checked the IP and MAC address in router and they were correct, but I decided to start over, so went into my Pi, changed the address, then went into the router and changed the address in my forwarding pool, rebooted  the router and everything works.

Thanks for the (as usual) fast responses.  I love the program and will be sending a PayPal donation.

Best Regards,

Jimmy Scruggs
Office 713-649-2776 x 6
Mobile: 713-899-1862