jgyates/genmon #767: error while loading shared libraries: libhogweed.so.4

bdwilson, 2022-09-15:
### Actual Behavior

pi@genmon:~/genmon $ ./startgenmon.sh start
Starting genmon python scripts
Error in ExecuteCommandList  : b'cmake: error while loading shared libraries: libhogweed.so.4: cannot open shared object file: No such file or directory\n'
Starting....
Starting /home/pi/genmon/genmon.py 
Starting /home/pi/genmon/genserv.py 
Starting /home/pi/genmon/addon/gentankutil.py 
Starting /home/pi/genmon/addon/genpushover.py 
Starting /home/pi/genmon/addon/genmqtt.py 
pi@genmon:~/genmon $ sudo apt-cache search hogweed
libhogweed6 - low level cryptographic library (public-key cryptos)
libnettle8 - low level cryptographic library (symmetric and one-way cryptos)
pi@genmon:~/genmon $ sudo apt-get install libhogweed
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libhogweed
pi@genmon:~/genmon $ sudo apt-get install libhogweed6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libhogweed6 is already the newest version (3.7.3-1).
0 upgraded, 0 newly installed, 0 to remove and 134 not upgraded.
pi@genmon:~/genmon $ cat /etc/issue
Raspbian GNU/Linux 11 \n \l

### Steps to Reproduce (including precondition)

Start Genmon. 

I’ve run ./genmonmaint.sh -r to refresh libs. 

Logs have been submitted via about; I’m on V1.18.14 on a pi zero 2.

skipfire, 2022-09-15:
You might try running apt update and upgrade then rerun the -r.
`sudo apt update -y`
`sudo apt upgrade -y`
That should ensure all the APT libraries are up to date.

jgyates, 2022-09-16:
According to your logs, your pi is having problems running these commands:

     sudo apt-get -yqq update
     sudo DEBIAN_FRONTEND=noninteractive apt-get -yqq install cmake

I would validate that this is an issue by running the above, then try running the update like @skipfire  suggested. Let us know the results.

Unfortunately this is a linux / open source issue. A lot of people make a lot of libraries and you can run into issues from time to time due to minor bugs in packages that do not play well together.

jgyates, 2022-09-16:
@bdwilson 

did these suggestions help?

bdwilson, 2022-09-16:
> @bdwilson
> 
> did these suggestions help?

No. upgrading everything is really a hammer approach, and broke my wifi driver, so I have to pull the box and connect to ethernet to repair it.

I just noticed this issue because my primary SD card was toast, so I moved my rpi-clone from last Sunday. I don’t recall this error before the primary drive died. Unfortunately I have a non-standard wifi card, so doing the suggested upgrade of all packages left the network dead.

bdwilson, 2022-09-17:
Well this escalated quickly. I'm back in business. Turns out rpi-clone has an issue where files with the same time/date stamp and length won't get updated. https://forums.raspberrypi.com/viewtopic.php?p=1965273&sid=1f60f9cad9620f8474d33d2a0db1d2d4#p1965273

I also filed a bug report with rpi-clone suggesting they use the -c option in rsync to resolve the issue, but I suspect that will increase clone operation times. https://github.com/billw2/rpi-clone/issues/146

closed 2022-09-17 by bdwilson