jgyates/genmon #768: Pi Upgrade

tcflint, 2022-09-16:
I've been using a 3b+ for a few years now and it was working fine.  Kept up with Genmon updates but now the pi is dead.
I have a pi4, questions is can I just put my microSD card in and roll?
If not what is the easy way to get going again?
I have a SD card backup

TIA

tcflint, 2022-09-16:
> I've been using a 3b+ for a few years now and it was working fine. Kept up with Genmon updates but now the pi is dead. I have a pi4, questions is can I just put my microSD card in and roll? If not what is the easy way to get going again? I have a SD card backup
> 
> TIA

Update it seems the pi itself is ok but the card is corrupt, so the backup I had is bad as well.
I do have the back ZIP from Genmon.
How do I get back from here.  Installed a new card and have OS but that's as far as I am.

jgyates, 2022-09-16:
From a fresh install of raspbian follow the setup instructions here:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Once installed, then to restore from a backup follow these instructions:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#restoring-settings-from-a-backup

General tips and tricks that may be useful are here:
https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#tip-and-tricks

tcflint, 2022-09-17:
Back to good.  Thank You sir
Excellent support as usual.

closed 2022-09-17 by jgyates