jgyates/genmon #769: Problem after enabling password

RRoyceus, 2022-09-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
Log in using Https
{Please write here}

### Actual Behavior
I wanted to require a login to Genmon   and thought all I had to do was enable HTTPS via my browser.  I was connected by my port forward "webhop, ***. webhop.org:8000".  Now I cannot access Genmon through the port forward or locally.  I can use Putty to get in.  How do I undo it?  Curiously, at the time all the devices on my home network lost the internet connect.  May or may not be related.  Rebooting the router appears to have fixed that problem

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2022-09-17:
@RRoyceus when you enable https and login the port is no longer 8000, but uses the standard https port 443.  Try that or no port designation when adding https://genmonipaddress.  You will also have to update your port forwarding.  On the advanced page I believe you can change to a non standard https port as well.

RRoyceus, 2022-09-17:
Jared,
Thank you for the fast reply.  When I tried reaching Genmon using port 443, I got the security warnings, then when I advanced, it asked for a user name and password, which obviously, I don’t have.   Is there a default user name and password?

Best Regards,

Jimmy Scruggs
Office 713-649-2776 x 6
Mobile: 713-899-1862

This email has been scanned for spam and viruses by Proofpoint Essentials. Click here<https://us1.proofpointessentials.com/index01.php?mod_id=11&mod_option=logitem&mail_id=1663380179-uxocVSYuJY4k&r_address=jscruggs%40scruggsco.com&report=1> to report this email as spam.

RRoyceus, 2022-09-17:
Jared,

it has been a long day.  I forgot that I set a username and password.  I went in to /etc/genmon\genmon.conf, found the username and password.  I got in and turned the security off until I get some rest and set it up correctly.

Thank you for the help!!!

Best Regards,

Jimmy Scruggs
Office 713-649-2776 x 6
Mobile: 713-899-1862

closed 2022-09-17 by jgyates