jgyates/genmon #772: Pi Temperature

tcflint, 2022-09-18:
Sorry if I'm in the wrong wrong area.
This is more of a request than issue.

I think it would be great to have an option in the settings to report a high temp (user set) for the Pi.
I use Pushover and that would work beautifully with that.

Again if I'm in the wrong place or this exists somewhere else I apologize.

Thank You

jgyates, 2022-09-18:
That is a good idea. Possibly a less subjective (how hot is too hot?) implementation would be to monitor / notify the CPU Frequency Throttling, Frequency Cap and Undervoltage that is displayed on the monitor page. internally the pi will throttle the frequency if the pi CPU temp is 80C or higher. This info is already displayed, but I can possibly add this to the things reported by short notifications like pushover. I will update this thread when completed but feel free to add any comments or feedback to this thread.

closed 2022-09-18 by jgyates

jgyates, 2022-09-18:
I added this feature. This is the new notice that applies to genpushover, genslack, gensms, gensms_modem, genemail2sms and gensyslog: 

Pi Hardware Sensor State - OK if all is good, otherwise it will report if CPU Throttling, CPU Undervoltage or CPU Frequency cap conditions have occurred on the pi. These are indicators of overheating or power supply voltage problems and can cause issues with the operation of your pi. 

These messages will not be sent on non pi platforms.

Let me know if you have any questions or issues.

tcflint, 2022-09-18:
Wow you're quick!

Thanks, should be useful.  
And thanks again for your hard work and continuous improvement to this project it is fantastic.