jgyates/genmon #773: Understanding evolution pin out

mikerenna, 2022-09-19:
opening a new issue to move the conversation from this thread:
https://github.com/jgyates/genmon/issues/16#issuecomment-762490183

However, in reference to the picture there for the evolution pin out, i had the following question:

i'm wondering about the difference in pin out. The referenced thread looks like my generac wifi connection. Elsewhere here it talks about evolution 2 using pins 3456 i think, but this looks like they are using 1256 (or it could be 3478 since I have no idea which direction the pin out diagrams are supposed to be viewed as). What implications does this have in creating a repinned connector to use both genmon and their existing wifi? (for the record, i've been trying to implment gregmac's esp32 wifi streamer version here that uses a 5v lead in the connector to power the ESP32 but in either case, understanding this pinout is my current sticking point). Thank you in advance for any guidance. This looks like awesome stuff. Is there a pinout somewhere that shows those four existing wires on the right pin numbers? And does anyone know the impact it has if i'm trying to use 1278 for data +/1, ground a 5v power in gregmacs thing? Thanks -Mike

jgyates, 2022-09-19:
Hi @mikerenna 

Here is the wiki entry with the molex pinout with diagrams:

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

The Evo 2 has some extra pins for the wifi module (it uses RS-485). There is a note on the wiki page about Evo 2 and the wifi module.

Main wiki page:
https://github.com/jgyates/genmon/wiki

mikerenna, 2022-09-19:
Thank you. That's the page causing me the confusion. in that page it says that evolution 2.0 adds pin 3,4,5,6 but based on the pinout diagram those don't seem to be the same pins used here: 
[image]

i was expecting to use 1,2,7,8 for my ESP32 connection and 3,4,5,6 for the existing wifi module in a repinned connector. I'm sure I'm missing something here though and hoping you can help me understand where I'm going wrong. Thank you for taking the time to assist btw. -Mike

skipfire, 2022-09-19:
The wiring diagram is the picture of the bottom of the controller not the plug, so you have to mirror the pin numbers when looking at the cable.

mikerenna, 2022-09-19:
would that not indicate that they are using pins 1,2,5,6 for the wifi module?

liltux, 2022-09-19:
@mikerenna Yes your are correct.  In the picture you shared that red wire is in pin 1, black is pin 2, blue is 5, and white is 6.  7 and 8 are RS 232 the mobile link cellular and genmon use.

mikerenna, 2022-09-19:
Aha, so i think my confusion is that it looks like GregMac's ESP32 implementation wants to also use pins 1,2. I was hoping they could coexist. 
[image]

skipfire, 2022-09-19:
You can share power pins, though you will have to watch for total power draw.  If it the setup pulls too much power it throws off the voltage calculations in the controller and makes it think that the lines are undervoltage and the generator will startup in outage mode.  You could look at adding a 12v->5v converter or powering from the battery.

skipfire, 2022-09-19:
Either from the pin or the battery.  The power limitation from the controller applies to the 12v on pin3 as well, but not if you power from the battery.  That said, I'm guessing an ESP would not be too close to the limit.

mikerenna, 2022-09-19:
ok thank you. that helps. Maybe i'll try doubling up on the supposition that the esp would draw very little and if i end up with unexpected behavior, i'll switch over to an option to power from the battery. thank you again for the clarity on this

closed 2022-09-19 by mikerenna