jgyates/genmon #774: Genmon Version V1.18.15 sends constant Raspberry Pi health messages

mxplea10, 2022-09-19:
I just updated from Genmon Version V1.18.14 to V1.18.15 and immediately starting receiving text messages about the Raspberry Pi health from Genmon.  How can I stop Genmon from monitoring the Raspberry Pi and/or sending texts about the Raspberry Pi health?  Or, how can I revert back to Genmon Version V1.18.14?  I'm receiving so many text messages that I had to shut down the Genmon Raspberry Pi.

- Generator Model: Generac 22kW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.15

mxplea10, 2022-09-19:
[image]

I thought that I had attached this but don't see it.

jgyates, 2022-09-19:
@mxplea10 

You can disable the Pi hardware sensor notifications on the add on page. Each add on has it's own disable for each notification catagory.

[image]

Also, note that these sensors indicate a problem with your pi's temperature or power source. Your attachment looks like the CPU freq throttling so this could be a heat issue. If the pi's internal temp is over 80 degrees C then CPU throttling is enabled. Maybe add a fan or better air flow may also correct the issue.

mxplea10, 2022-09-19:
Ok, it seems strange that this occur immediately after the update.  I'll troubleshoot the issue and updated accordingly.  Thanks for the fast response.

mxplea10, 2022-09-19:
@jgyates

Is there another setting to change?  I'm still receiving messages.
[image]

jgyates, 2022-09-19:
This is a new feature in 1.18.15. 

https://github.com/jgyates/genmon/blob/master/changelog.md

It is very likely you had this issue all along but you were not monitoring this info. The new update allows genmon to monitor this information and notify you. This feature was added by request form this thread:

https://github.com/jgyates/genmon/issues/772

I am going to close this issue now but feel free to post here with follow up questions or comments.

closed 2022-09-19 by jgyates

mxplea10, 2022-09-19:
Ok, but how is this helping me with temporarily disabling this until I have time to troubleshoot it.

mxplea10, 2022-09-19:
Again, I have disable "Notify for Pi Hardware Issues" in the Twilio Add-on but I'm still receiving the text messages.

jgyates, 2022-09-19:
have you saved you settings once you disable this? I will try to duplicate this later tonight.

mxplea10, 2022-09-19:
Yes, I saved it.  I disabled Twilio and the text messages stopped.  I'll enable it again and leave "Notify for Pi Hardware Issues" disabled to verify if I receive the message again.

[image]

mxplea10, 2022-09-19:
Ok, the Pi Hardware issue messages started again when I enabled Twilio.

jgyates, 2022-09-20:
@mxplea10 

sorry about the confusion in the earlier reply. I glossed over the fact that you were unable to disable the pi hardware sensor monitoring. I was able to duplicate and found the issue. I checked in an update so just update from the About page and then you can enable the SMS add on and also disable the pi sensor monitor. Let me know if you have any issues or questions.

mxplea10, 2022-09-20:
It works with the update.  Thanks...