jgyates/genmon #775: Feature request : Manually start / Stop / Transfer generator from Dashboard

fencing49, 2022-09-20:
Maybe it's not on my dashboard but i don't see a way to manually start / stop / transfer the generator on my own.

It would be a nice addition for also running along side with the exercise as sometimes i just want to turn my generator on to monitor outputs.

If it's already in Genmon let me know where and feel free to close this.

Either way, love this software so damn much you have no idea.
Down with Generac corporate greed, all hail genmon!

skipfire, 2022-09-20:
@fencing49 it is near the bottom of the maintenance page.

fencing49, 2022-09-20:
That's odd, appears to be missing on mine
[image]

> @fencing49 it is near the bottom of the maintenance page.

jgyates, 2022-09-20:
Do you have "Smart Transfer Switch" enabled on the Settings page? If you have a smart switch then the controller can not support these commands as your controller would be in Two Wire mode. If you do not have a smart switch then you should disable this settings and you will see the controls to start, stop and start/transfer on the Maintenance page.

If that does not answer your question I would need to see you logs, which you can submit on the about page.

fencing49, 2022-09-20:
> Do you have "Smart Transfer Switch" enabled on the Settings page? If you have a smart switch then the controller can not support these commands as your controller would be in Two Wire mode. If you [...]

That was it, i have a smart transfer switch so i had it enabled."

Thankyou!

closed 2022-09-20 by fencing49