jgyates/genmon #777: Outage log not showing outages

farberm, 2022-10-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Outage Panel not showing recent iutages

### Actual Behavior

it stopped showing outages for the past year

### Steps to Reproduce (including precondition)

I have updates to the latest software, rebooted etc

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: latest

jgyates, 2022-10-01:
Is the outage page not offered in the menu on the left or is your outage page blank. More info would help. Also, do you have "Smart Transfer Switch" enabled on the settings page. Outages are not reported if this is enabled.

If your outage menu item is showing up but you outage page is not showing past outages then check the contents of the file /etc/genmon/outage.txt

      cat /etc/genmon/outage.txt

farberm, 2022-10-01:
The outage page is visible but only shows outages from 2019. Nothing for the past 2 years.Smart transfer switch option is disabled I will need to check the file when I get home in 2-3 daysSent from my iPhone

jgyates, 2022-10-01:
also, there is a setting on the settings page called "Do Not Check for Outages" make sure that is not enabled.

farberm, 2022-10-01:
Got it. That was the problem…Sent from my iPhone

jgyates, 2022-10-01:
k, closing this issue now. Let me know if you have any other issues.

closed 2022-10-01 by jgyates