jgyates/genmon #778: Placement of ambient temperature and engine temperature sensors

FlaMike, 2022-10-05:
I am in the process of building a D1 Mini Pro serial bridge version of genmon based on gregmac's ESP32 serial bridge (I happen to have an unused D1 Mini Pro so I am planning to use it in place of the ESP).

I would like to include in the build a Dallas temperature sensor for ambient temperature of the generator case and a second Dallas sensor for engine temperature.  My question is, where should I place the sensors and how should I secure them in place?

My generator is an air cooled Generac 24KV unit with an Evolution controller.

Thank you for your help & suggestions!

jgyates, 2022-10-05:
@FlaMike 

For the ds18b20: GPIO4, 3.3V and ground. once connected the pi OS software should pick up the one wire if it is enabled via the raspi-config program. For genmon support, I would start here:  https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentemppy-optional

and the links in the wiki will take you to this project:

https://github.com/lmamakos/RPi-pHat-Thermocouple

which has schematic:

https://github.com/lmamakos/RPi-pHat-Thermocouple/blob/genmon/RPi-pHat-Thermocouple.pdf

jgyates, 2022-10-05:
I am going to close this issue but feel free to post any followup questions to this thread.

closed 2022-10-05 by jgyates