jgyates/genmon #779: Unable to SSH into pi Zero W when connected to controller (since last update of genmon V1.18.15)

rhorrell, 2022-10-05:
### Expected Behavior

To be able to SSH into GENMON pi while system is running

### Actual Behavior

SSH prompts for password and freezes.  Sometimes it will eventually give you a prompt but usually give a disconnect:

ssh pi@192.168.x.x
pi@192.168.x.x's password: 
Linux genmon 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Oct  2 15:20:07 2022 from 192.168.x.x
pi@genmon:~ $ client_loop: send disconnect: Broken pipe

### Steps to Reproduce (including precondition)

I was always able to SSH into the monitor prior to the last update.   The web interface works fine but just can't SSH into it while it is attached to the Generator.   I thought that maybe the Micro SD card may have gone bad again (happened a year ago) so I built out a new setup using the Pintsize.me pHAT instead of my old configuration.   I also switched from using Molex supplied power to using power cable connected directly to the battery (supplied/w pHAT).

New setup has the exact same behavior; works perfectly via the web interface but SSH hangs while connected to Evolution controller.   Both old and new run perfectly fine both web and SSH when not connected to controller.   

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070431 
- Genmon Logs: Submitted  
- Genmon Version: V1.18.15

Monitor :

    Generator Monitor Stats :
        Monitor Health : OK
        Controller : Evolution, Air Cooled
        Run time : Generator Monitor running for 0:49:31.
        Power log file size : 0.09 MB of 15.00 MB
        Generator Monitor Version : V1.18.15
        Update Available : No

    Communication Stats :
        Packet Count : M: 43155, S: 43153
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 1
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 29.05
        Average Transaction Time : 0.0384 sec

    Platform Stats :
        CPU Temperature : 127.40 F
        Pi Model : Raspberry Pi Zero W Rev 1.1
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 72.5%.  ### Kind of high here but usually less than 50% 
        OS Name : Raspbian GNU/Linux
        OS Version : 11 (bullseye)
        System Uptime : 0:30:51
        Network Interface Used : wlan0
        WLAN Signal Level : -55 dBm
        WLAN Signal Quality : 55/70
        WLAN Signal Noise : -256 dBm
        WLAN ESSID : xxxxxx
        System Time : Wednesday October 5, 2022 16:36:10

    Weather :
        Current Temperature : 59.4 F
        Conditions : Mist

rhorrell, 2022-10-05:
BTW: SSH output was from old config (Buster) Monitor details are from new setup (Bullseye)

jgyates, 2022-10-05:
@rhorrell 

I can't think of anything that has changed recently that would cause the issue you are describing.

I can suggest a few things to try:

1) make sure "Optimize for slower CPUs" is enabled on the settings page. This adjusts the serial received timing to be kinder to slower CPUs like the Pi Zero. If enabled this should reduce the CPU usage.
2) Disable any unused / unneeded add on programs. Based on the logs you sent it looks like you have genlog (log to CSV file), gensyslog (log to linux syslog), and genpushover enabled. There is no issue running all of these but each one does contribute to the CPU requirements.
3) Look at your disk space to make sure you have adequate space on your SD card.  Low disk space can make your system very slow.
4) If you log files are being written a lot the log rotation built into the OS could be causing you higher CPU utilization. Older log are written frequency the system will compress / archive older files.

rhorrell, 2022-10-06:
@jgyates 
Looks like the issue was with the access point it was connecting too resulting in ssh packets getting corrupted...  it had nothing to do with the pi or your software, Sorry for the fire drill.

PS: I am a long time user and huge fan of this software.... Keep up the great work!!!!!

closed 2022-10-06 by rhorrell