jgyates/genmon #780: Compatibility with Ethernet Mobile Link

rcmurphy, 2022-10-06:
Hi! I'm using the pintsize.me HAT and splitter. Genmon works fine, but appears to interfere with the operation of the MobileLink Ethernet; MobileLink displays an error that its not getting updates from the controller, and reports the occasional random fault. If I unplug the GenMon RPi, MobileLink is happy.

skipfire, 2022-10-06:
I have updated the PintSize.Me site providing some compatibility clarification.

rcmurphy, 2022-10-06:
We should probably add it to the wiki page on creating your own splitter cable too…

@jgyates  completely unrelated, but let me know if you’d be interested in getting some CI working; build pipelines and infrastructure is my specialty!

jgyates, 2022-10-06:
I am going to close this thread but feel free to post any follow up here.

closed 2022-10-06 by jgyates

rcmurphy, 2022-10-07:
@jgyates great! I'll open a PR to get some linting working on PRs this weekend. In the meantime, I would recommend creating a separate GitHub organization (unfortunately `genmon` is already taken as a github username/ group name) and moving the repo to it. There is a lot of functionality that only works with groups, not individual accounts, and it would allow me to assist with configuration. Let me know if you want any assistance with this; my email is `rebecca@[github username].me`.