jgyates/genmon #782: json issues

andrewvanlagen, 2022-10-09:
Hi 
I am trying  to inport the json file
I can import all the  (status- Engine) but the (External Temperature Sensors) wont import. 

http://192.168.20.10:8000/cmd/status_json
{"Status": [{"Engine": [{"Switch State": "Auto"}, {"Engine State": "Off - Ready"}, {"Battery Voltage": "14.00 V"}, {"RPM": "0 "}, {"Frequency": "0.00 Hz"}, {"Output Voltage": "0 V"}, {"Output Current": "0.00 A"}, {"Output Power (Single Phase)": "0.00 kW"}, {"Active Rotor Poles (Calculated)": "0 "}]}, {"Line": [{"Utility Voltage": "242 V"}, {"Utility Max Voltage": "245 V"}, {"Utility Min Voltage": "230 V"}, {"Utility Threshold Voltage": "156 V"}]}, {"External Temperature Sensors": [{"Main DS18B20": "23.00 C"}, {"External DS18B20": "17.31 C"}]}, {"Last Log Entries": {"Logs": {"Alarm Log": "11/26/21 19:11:02 ESTOP Pressed : Alarm Code: 2800", "Service Log": "", "Run Log": "10/02/22 14:00:28 Stopped - Auto "}}}, {"Time": [{"Monitor Time": "Saturday October 8, 2022 21:05:11"}, {"Generator Time": "Saturday October 8, 2022 21:05"}]}]}

 temp = resp1.data.Status[1].External Temperature Sensors[0].'Main DS18B20'
 sendEvent(name: "Main DS18B20", value: temp, isStateChange: true)

jgyates, 2022-10-09:
I think it should be 

   temp = resp1.data.Status[2].External Temperature Sensors[0].'Main DS18B20'

instead of 

    temp = resp1.data.Status[1].External Temperature Sensors[0].'Main DS18B20'

In the Status list is is the 3rd entry (Engine, Line, then External Temperature Sensors.

Let me know if that does not correct your issue.

jgyates, 2022-10-10:
I am going to close this issue for now but feel free to post any follow up questions to this thread.

closed 2022-10-10 by jgyates

andrewvanlagen, 2022-10-10:
both these statement doesn't work. 

-    temp = resp1.data.Status[2].External Temperature Sensors[0].'Main DS18B20'
                   sendEvent(name: "Main DS18B20 Number", value: temp, isStateChange: true)

-   temp = resp1.data.Status[2].External Temperature Sensors[1].'External DS18B20'
                 sendEvent(name: "External DS18B20 Number", value: temp, isStateChange: true

this is the mqtt statement

- Genmon/generator/Status/External Temperature Sensors/Main DS18B20 22.19 C
- Genmon/generator/Status/External Temperature Sensors/External DS18B20 17.06 C

are you able to remove the C. 
i am wondering if that is the issue

liltux, 2022-10-11:
@andrewvanlagen I was doing a project with NodeRed and importing these values.  I was doing it two ways, one via HTTP request and another via the clientinterface.py method.  HTTP I had issues with formatting in NodeRed and have to do some JS tweaks to use some of the values.  When using the clientinterface.py method I used a modified clientinterface.py and bash script to run the python file.  Then, using bash, I would parse out the json of all spaces.  This solved some format issues I was having.  I can't recall exactly what I did, but maybe that will help?

jgyates, 2022-10-11:
Try this:

Your javascript is not dereferenceing the object correctly. you can copy and past the code below into this site and it will work:

https://www.programiz.com/javascript/online-compiler/

```javascript
test = {"Status": [{"Engine": [{"Switch State": "Auto"}, {"Engine State": "Off - Ready"}, {"Battery Voltage": "14.00 V"}, {"RPM": "0 "}, {"Frequency": "0.00 Hz"}, {"Output Voltage": "0 V"}, {"Output Current": "0.00 A"}, {"Output Power (Single Phase)": "0.00 kW"}, {"Active Rotor Poles (Calculated)": "0 "}]}, {"Line": [{"Utility Voltage": "242 V"}, {"Utility Max Voltage": "245 V"}, {"Utility Min Voltage": "230 V"}, {"Utility Threshold Voltage": "156 V"}]}, {"External Temperature Sensors": [{"Main DS18B20": "23.00 C"}, {"External DS18B20": "17.31 C"}]}, {"Last Log Entries": {"Logs": {"Alarm Log": "11/26/21 19:11:02 ESTOP Pressed : Alarm Code: 2800", "Service Log": "", "Run Log": "10/02/22 14:00:28 Stopped - Auto "}}}, {"Time": [{"Monitor Time": "Saturday October 8, 2022 21:05:11"}, {"Generator Time": "Saturday October 8, 2022 21:05"}]}]}

temp = test.Status[2]['External Temperature Sensors'][1]['External DS18B20']
console.log(temp)
```