jgyates/genmon #783: 22KW Generac Air Cooled

Barrygordon, 2022-10-09:
I have a Generac 22KW air cooled generator. I believe its is usuing an Evolution 1.0 controllrtr

I have installed the software on an RPi 4 with a Pintsizeme HAT.  The HAT has a barrel connector for power and a molex connector for connection to the generator.  I Have the cables.  The box is mounted outside on the back of the generator with a 1 inch close nipple and bushings. The nipple is large enouch to pass the molex cable and the barrel cable through. I have tested the operation of the firmware and it appears to be working, but it is not yet connected to the generator.

I am ready to connect the RPi to the generator.  Can someone tell me how to get to the Molex plug on my model generator?  I want to avoid taking off too many covers and parts searching.

TIA, Barry Gordon

[image]

[image]

[image]

skipfire, 2022-10-09:
The front panel locks in place with 2 pegs going down from the bottom and 2 pegs sideways at the top that hold it to the sides, look at this before removing so you have an idea how it goes back, then just pull straight up on that front panel and it will come out.  Then on the right side panel there will be 2 screws holding it at the top, and then there is typically 1 screw on an L-bracket at the bottom, remove the 2 screws at the top and either screw for the L-bracket that holds the bottom front on (whichever is easier to get at), and this panel also has 2 pegs that on the bottom.  Once the screws are out this panel will likely just try to fall sideways, just move the top out a couple inches, then pull the whole thing up.

At this point you should have full access to the underside of the control panel, this is also where the battery is.  On the bottom near the front towards the engine side you the 8-pin Molex will either have a Mobilelink cable plugged into it, or a sticker over it, this is towards the front of the controller on the engine side.  Remove the sticker or Mobilelink cable, and then you can plug into the cable that goes to the HAT.  You will want a socket wrench for the battery cable, and be sure that your socket wrench doesn't contact anything else when loosening the positive terminal, especially the silver shroud you will see that protects the upper back section.

To close it back up, get the bottom pegs of the side panel positioned, put all 3 screws back in.  Get the bottom of the front panel between the sides but about 2 inches above the base, then rotate it so the top is lined up, then slide the whole thing down so that the bottom pegs go into the holes and the top pegs go into the slots on the side, I find this the hardest part.

Barrygordon, 2022-10-09:
Having nothing to do I decided to look inside the generator.  The front panel came off quite easily (Thanks Skipfire).  I was unable to remove the side panel. I can get at the two screws attaching the side panel at the front but can not see any screws attaching the side panel at the rear?

I am pretty sure I have an evolution 1.0 controller for which the mobilink is not an option.  I looked with a mirror and I can not see any 2x4 molex connector. There are several cables connected with molex type connectors but none are a 2x4 molex.  

There is one molex (1x3) with three black wires entering it and a black stub connected to it.  The stub plugged into the connector is marked as "OK0258BB"  Perhaps this the serial connection (Tx, Rx , gnd)?  If so does anyone know which wires are TX, Rx and Ground? Or it might be a software access point for testing.
 
[image].

There are two labels on the bottom of the controller marked as Accessory port 1 and Accessory port 2.  I assume they might be covering the 2x4 molex connectors.  Is that what I am looking for, and if so which one is the one I want to use.  I live in FL and it has just started to pour so my investigations will continue tomorrow.

[image]

I can generally figure this stuff out.  I am fairly expert in digital software and hardware and my whole home runs on a Raspberry Pi. I was on the original GPS team in the early1960s, but am now 83 yo dont like to putter around taking things on and off.  The only disadvantage to getting old is that the alternative is really bad.

TIA,  Barry

skipfire, 2022-10-09:
Your enclosure must be different than mine, but you are in the right spot.  It is under the Accessory Port 1 or 2 sticker.  One of them will have an 8-pin connector.

madisdev, 2022-10-09:
Hey Barry,

I have a similar model and it should be under accessory port 1. 

But kudos to you for working on such cool projects as gps. I’m sure you have some cool stories. 

Just a heads up you don’t need to remove any covers except for the front. I am able to reach under and plug it in without removing other panels.

grsthegreat, 2022-10-10:
Removal of side panel is only necessary on Evolution 2 systems. If yours is evolution 1, you can access port 1 from the front

jgyates, 2022-10-11:
I am going to close this issue for now but feel free to post here with any followup questions or status.

closed 2022-10-11 by jgyates