jgyates/genmon #785: Fuel calculations incorrect

rwskinner, 2022-10-11:
Using Version 1.18.14

When using the Mopeka sensor, the amount of fuel remaining on the Maintenance is incorrect.
Down below, it shows 80.89%  (Really should just round that), but anyways, 250 gal tank x 80.89% = 202.2 gallons.

I double checked, and the Mopeka add-on shows 25 gallon tank selected.  The percentage remaining is correct, but for some reason it's using a 200 gallon tank x 80.89% = 161.78,   
This also affects Hours of fuel remaining.

**From the maintenance page:**
Fuel Level Sensor : 80.89 %
    Fuel In Tank (Sensor) : 161.78 gal
    Hours of Fuel Remaining (Estimated 0.50 Load ) : 63.94 h

jgyates, 2022-10-11:
In your comments above you mention mopeka is setup to use a 25 gallon tank. Do you mean 250 gallon?

What size tank do you have on the settings page? This size needs to be the same as your mopeka tank or set to zero to disable any calculations of remaining fuel. If you are using an addon gauge (one of the four types supported), the maintenance page gets the percentage from the add on and the rest of the info (in this case tank size) from the settings page. What you are showing would indicate that the settings page only has a value of 200, not 250. 

If that is not your issue I would need to see your logs, know your model number and for your to be running the latest version of the software.

rwskinner, 2022-10-11:
Sorry, it was a typo.
Add-ons page - Mopeka tank size = 250 gallons.
In the settings page - Yes, it was set to 200.  I changed to 250 and it 
resolved the issue.
For some reason, I overlooked it and thought that the Mopeka defined it 
when used.

jgyates, 2022-10-11:
Great, I am glad it is working for you now. I am going to close this issue but feel free to post related questions or comments to this thread.

closed 2022-10-11 by jgyates