jgyates/genmon #787: All Done; some questions

Barrygordon, 2022-10-13:
All Done.  Took me longer than I expected but it all worked out in the end.  The back of my 22KW generator had an unused connection point for a 3/4" conduit or cable entry.  I enklarged that opening to fit a 1" EMT connector using a chasis punch..  That size passes the molex plug nicely.  I did not have to remove the end panel.  It took me a minute or two to figure out how the control panel came out of the case. The nottom of my EVO v1 contol panel had a sticker over Accesory port 1 on the bottom of the controller. I removed the sticker and attached the molex.  I attached a 1" LB condulet to  the hole I enlarged in the rear of the generator using a chase nipple inside the generator. Just enough room to get my hand in to tighten it up. I drilled a proper size hole in the case (amazon (5.9 x 5.9 x 3) and attached it to small wood spacer I painted black.  The spacer is attached to the back of the generator.  Attached the power cable to the battery and everything worked!!  I have attached two pictures.

The only question is that after a few minutes (generator Idle or running) I get an exclamation point and a message that the genmon could not ceonnect to the server and a count of the number of messages lost. After a while it goes away.  What server is it unable to connect to?  I assume the Raspberry Pi.

Thanks to All who helped me with this project!!

[image]

[image]

[image]

jgyates, 2022-10-13:
@Barrygordon 

The message you are getting means that the javascript running in your browser is failing to communicate with the web server running on the raspberry pi. The code in the browser gets updates from the server to continuously (every 5 seconds) update the info and controls on the web page. This could be caused by poor wifi signal on either the system running the browser or the pi. To troubleshoot, you could try the browser on a system with a wired (Ethernet) connection and see if the problem persist.

Barrygordon, 2022-10-13:
Thanks,  I will look into that.

I guess I need to use one of the add-ons to have notifications sent to my iPhone.  Which one would you suggest?

TIA, Barry

PS  Genmon is a really nice piece of work!!

jgyates, 2022-10-14:
if you want messages sent to an app on your phone then pushover is nice.

Email works great also.

Both are free.

I am going to close this thread but feel free to post follow up here.

closed 2022-10-14 by jgyates