jgyates/genmon #788: No exercise or run/stop options on maintenance screen

atarijeff, 2022-10-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Exercise and run/stop options available at bottom of maintenance screen 

### Actual Behavior

No options displayed

### Steps to Reproduce (including precondition)

Load maintenance page

### Screenshot or Pictures relating to the problem (if possible)

[Untitled.pdf](https://github.com/jgyates/genmon/files/9793239/Untitled.pdf)

### Your Environment

- Generator Model: Generac 18Kw air cooled. Hardware Version : V1.01. Firmware Version : V1.15
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.15
- Connected via pintsize.me pi hat https://pintsize.me/store/ols/products/generator-hat/v/PGI-Gc

jgyates, 2022-10-16:
See this issue:

https://github.com/jgyates/genmon/issues/775

If that does not solve your problem let me know. I received your logs and it looks like you likely have the "Smart Transfer Switch" setting enabled. If you actually have a smart transfer switch the functionality of the run / stop buttons will not work with a smart transfer switch as the switch is in control of that. A Smart transfer switch puts the switch (instead of the controller) of monitoring the utility lines and deciding when the generator should start.

I am going to close this issue but feel free to respond here if you have additional questions or status.

closed 2022-10-16 by jgyates

atarijeff, 2022-10-16:
The setup does use an automatic transfer switch (same as smart?) Is there a way to edit the exercise times when using an automatic transfer switch?

jgyates, 2022-10-16:
An automatic transfer switch is not the same as a Smart or Intelligent Transfer Switch. The Smart Switch is programmable and controls the ability to set parameters like exercise time. If your switch does not have that is is just a simple automatic transfer switch then do not set the Smart Transfer Switch Setting to the enabled position. Once you change this setting and save your settings you should have the ability to set the exercise time and have start / stop buttons in the web interface. Setting this to "Smart Switch" just disables these settings since the transfer switch controlling this functionality with a Smart Switch. With a smart switch the generator is in two wire mode so the exercise and start stop functions do not work over mod bus so this setting removes them from the user interface.

atarijeff, 2022-10-16:
Very good. I have disabled "smart switch" settings and everything looks good