jgyates/genmon #789: Unable to access web GUI interface.

DC072012, 2022-10-16:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Should be able to access genmon via web interface at http://192.168.57.170:7500

### Actual Behavior

Safari can't connect to server.  I searched and searched the issues but can't figure out what to check.

### Steps to Reproduce (including precondition)

Genmon was working well.  I intended to install the Pushover system today.  Before I did that I wanted to get a backup image of the SD card.  
Apparently that was a mistake.

I shut down the Pi 3, brought the card in.  Used Apple Pi Baker on the Mac to clone the card to my hard drive.  Apple Pi Baker said it worked but I'm not sure.  Properly ejected the card, put it back in the Pi.

Genmon starts up, I get email messages saying it is active on 
IP 192.168.57.170.  If I turn my switch off on the controller genmon sends an email notifying me.  Turn switch back on I get email notification.

Somehow cloning the card must have messed something up with the web settings - I have no idea.

Can SSH into the Pi (that is how I sent the logs using [ClientInterface.py].  
Nothing changed with my router - it is pre-assigning the proper IP address.  

Can ping 192.168.57.170, median ping time 11ms.

I cannot figure out how to fix the web interface access problem since I don't know what caused it to get messed up.

Thanks in advance for any help.
### Screenshot or Pictures relating to the problem (if possible)

generator: monitor

Monitor : 

    Generator Monitor Stats : 
        Monitor Health : OK
        Controller : Nexus, Air Cooled
        Run time : Generator Monitor running for 0:30:31.
        Generator Monitor Version : V1.18.15
        Update Available : No

    Communication Stats : 
        Packet Count : M: 57266, S: 57265
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 0
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 62.54
        Average Transaction Time : 0.0307 sec

    Platform Stats : 
        CPU Temperature : 110.30 F
        Pi Model : Raspberry Pi 3 Model B Plus Rev 1.3
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 3.0%
        OS Name : Raspbian GNU/Linux
        OS Version : 9 (stretch)
        System Uptime : 0:31:12
        Network Interface Used : wlan0
        WLAN Signal Level : -59 dBm
        WLAN Signal Quality : 51/70
        WLAN Signal Noise : -256 dBm
        WLAN ESSID : djnet-1
        System Time : Sunday October 16, 2022 13:45:28

### Your Environment

- Generator Model: 005882-0
- Genmon Logs: Submit via the About page (sent using manual interface)
- Genmon Version: current

DC072012, 2022-10-16:
Found another issue thread, thought this might add info.  Still have the problem however:

Ran:
pi@genpi [~] cd genmon
pi@genpi [~/genmon] ./startgenmon.sh restart

Restarting genmon python scripts
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Stopping....
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py 
Starting /home/pi/genmon/genserv.py 

Then ran (some of these output messages seem old):
cat /var/log/genserv.log

Output:
2019-09-05 19:41:55,328 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-09-05 20:38:43,393 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-09-05 21:21:03,058 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-09-06 08:14:22,007 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-09-06 08:17:40,517 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-07 14:02:39,477 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-07 14:06:27,463 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-10 15:47:48,042 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-12 10:54:00,028 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-27 16:12:08,906 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-10-27 16:12:16,392 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-10-27 16:12:41,191 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-10-27 16:12:48,577 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-11-10 12:17:22,369 : Warning: Reading Config File (value not in list): controllertype : 
2019-11-10 14:45:23,678 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2019-11-10 14:45:27,880 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-11-10 14:45:38,253 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-11-10 14:45:38,378 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2019-11-10 14:45:53,274 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-11-10 14:46:02,268 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-11-10 19:03:36,499 : Warning: Reading Config File (value not in list): controllertype : 
2019-11-15 12:33:08,540 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2019-11-15 12:33:10,187 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-11-15 12:33:17,500 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2019-11-16 11:56:04,886 : Warning: Reading Config File (value not in list): controllertype : 
2019-11-16 11:56:12,826 : Warning: Reading Config File (value not in list): controllertype : 
2019-11-16 11:56:58,415 : Warning: Reading Config File (value not in list): controllertype : 
2019-11-19 14:22:29,166 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2019-11-19 14:22:30,896 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2019-11-19 14:22:38,696 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-01-16 18:54:05,398 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2020-01-16 18:54:07,478 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2020-01-16 18:54:14,844 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-01-16 21:58:55,721 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2020-01-16 21:59:03,125 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-01-17 15:41:40,975 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-01-17 16:05:02,155 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2020-01-17 16:05:06,440 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2020-01-17 16:05:15,197 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-01-20 11:01:39,523 : Warning: Reading Config File (value not in list): controllertype : 
2020-01-22 11:12:23,896 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-02-16 15:37:22,604 : Warning: Reading Config File (value not in list): controllertype : 
2020-02-21 13:52:50,178 : Warning: Reading Config File (value not in list): controllertype : 
2020-02-26 12:40:18,883 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh -u -n
2020-02-26 12:40:23,137 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart /etc/genmon/
2020-02-26 12:44:59,600 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2020-03-15 17:02:31,640 : Warning: Reading Config File (value not in list): controllertype : 
[... cut ...]
2022-08-22 09:28:49,907 : Error in ExecuteCommandList  : b'debconf: unable to initialize frontend: Dialog\ndebconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)\ndebconf: falling back to frontend: Readline\ndebconf: unable to initialize frontend: Readline\ndebconf: (This frontend requires a controlling tty.)\ndebconf: falling back to frontend: Teletype\ndpkg-preconfigure: unable to re-open stdin: \n'
2022-08-22 09:28:51,302 : Warning: Missing entry: genmopeka , adding entry
2022-08-22 09:28:51,374 : Warning: Missing entry: genmopeka
2022-08-22 09:28:51,394 : Copying genmopeka.conf to /etc/genmon/
pi@genpi [~/genmon]

jgyates, 2022-10-16:
I received your logs. Nothing outstadning there. I would not assume that just because nothing has changed in your router that that is not the issue. The fact that your pi rebooted may be triggering some protection / blocking in your router. Do you have an ASUS router. They have some features in some models that have "learning" security that I have seen block web servers on your local network.

Try to ping the pi with the network name you have assigned (from the logs abvove)

    ping genpi

or 

     ping genpi.local

If either of these work then subustitue the named value (e.g. genpi.local) in the URL 

      http://genpi.local:7500

If that does not work then I would start looking at your router settings.

Also, do you have remote access to your pi setup via your router? If you do try disabling it.

The logs are showing that everything is working as expected.

You could also try rebooting your pi.

jgyates, 2022-10-16:
Also, you can search past issues with the "Router" tag and see a few issues similar to what you are describing.

DC072012, 2022-10-16:
OK, thanks for the info so far.  Another data point.

If I reboot the Pi, the web interface will not work.  If I do this:
 cd genmon
  ./startgenmon.sh restart

the web interface works immediately after the restart.  Not sure the router would cause this based on reading other posts.  What else might cause this?  thanks again.

jgyates, 2022-10-16:
Try this:

1) reboot your pi
2) log in via ssh
3) issue this command:

        curl -I  http://192.168.57.170:7500/

what is the output of the above command? 

4) try to access genmon with ClientInterface.py. Is this successful?

5) now do this:

      cd genmon
     ./startgenmon.sh restart

Then repeat steps 3 and 4

Also I noticed this issue from the past?

https://github.com/jgyates/genmon/issues/271

Generally when troubleshooting an issue like this try disabling any external access  or port forwarding at the router and try to access the pi. Since you can access the pi on your network, just not the web server, and the server appears to be functioning

DC072012, 2022-10-17:
Thanks for the help.  I re-read #271.  It looks like that was a problem caused by trying to access genmon via remote access through the router.  After that problem I took your advice and set up a VPN on Digital Ocean where I can access items on my local network without port forwarding the devices.  

I ran the following tests as you outlined with no port forwarding turned on at all at the router.  Rebooting the Pi results in no connection, restarting genmon after that fixes the problem.  Seems like the reboot isn't starting something right away???  

Thanks again.

Linux genpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l

  _______ _______ ______      _______ ___ 
 |   _   |   _   |   _  \    |   _   |   |
 |.  |___|.  1___|.  |   |   |.  1   |.  |
 |.  |   |.  __)_|.  |   |   |.  ____|.  |
 |:  1   |:  1   |:  |   |   |:  |   |:  |
 |::.. . |::.. . |::.|   |   |::.|   |::.|
 `-------`-------`--- ---'   `---'   `---'
                                          

Last login: Sun Oct 16 21:16:26 2022 from 192.168.57.93
pi@genpi [~] sudo reboot
Connection to 192.168.57.170 closed by remote host.
Connection to 192.168.57.170 closed.
 ~ % ssh pi@192.168.57.170
pi@192.168.57.170's password: 
Linux genpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l

  _______ _______ ______      _______ ___ 
 |   _   |   _   |   _  \    |   _   |   |
 |.  |___|.  1___|.  |   |   |.  1   |.  |
 |.  |   |.  __)_|.  |   |   |.  ____|.  |
 |:  1   |:  1   |:  |   |   |:  |   |:  |
 |::.. . |::.. . |::.|   |   |::.|   |::.|
 `-------`-------`--- ---'   `---'   `---'
                                          

Last login: Sun Oct 16 21:28:41 2022
pi@genpi [~] curl -I  http://192.168.57.170:7500/
curl: (7) Failed to connect to 192.168.57.170 port 7500: Connection refused
pi@genpi [~] python3 ClientInterface.py -a 192.168.57.170
python3: can't open file 'ClientInterface.py': [Errno 2] No such file or directory
pi@genpi [~] cd genmon
pi@genpi [~/genmon] ./startgenmon.sh restart
Restarting genmon python scripts
/usr/local/lib/python3.5/dist-packages/OpenSSL/crypto.py:14: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release of cryptography. Please upgrade your Python.
  from cryptography import utils, x509
Stopping....
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py 
Starting /home/pi/genmon/genserv.py 

pi@genpi [~/genmon] curl -I  http://192.168.57.170:7500/
HTTP/1.0 200 OK
Content-Length: 2245
Content-Type: text/html; charset=utf-8
Last-Modified: Tue, 20 Sep 2022 01:47:29 GMT
Cache-Control: no-cache, no-store, must-revalidate, public, max-age=0
Expires: 0
ETag: "1663638449.1661212-2245-3473673333"
Date: Mon, 17 Oct 2022 01:29:56 GMT
Pragma: no-cache
Server: Werkzeug/1.0.1 Python/3.5.3

pi@genpi [~/genmon] python3 ClientInterface.py -a 192.168.57.170
OK : Auto, Off - Ready
>exit
pi@genpi [~/genmon]

jgyates, 2022-10-17:
One thing to keep in mind. Once your system boots, genmon will wait 30 seconds before starting. Try it again but wait about 30 seconds after the initial login via ssh.

You can check the script that starts genmon like this:

      sudo crontab -e

you may be promoted to chose an editor, if you are press the number that uses the nano editor

This will bring up a script file and it should have the following line in it:

     @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

If it does not then add this line, but only if it is not there already. Starting it twice would be problematic.

if you do change the contents of the cron job then you can exit by typing Ctrl+X. If you changed the file it will ask you if you want to save it. Press Y then Enter to save

DC072012, 2022-10-17:
This line was there:
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

I changed it to 60 seconds:
@reboot sleep 60 && /bin/bash /home/pi/genmon/startgenmon.sh start

I rebooted the Pi 2 times and both times the web interface became available without needing to restart genmon.  

Not sure why the 30 seconds isn't enough.  Are there any Pi system updates I should run?  I'm not sure the system has been updated since I bought it.

thanks again.

jgyates, 2022-10-17:
I run this once a week:

     sudo apt-get update
     sudo apt-get dist-upgrade

This makes sure that security updates and bug fixes are applied to the operating system. Once it is updated it is a good idea to reboot. There are some other utilities that you can run that will tell if you a reboot is required but rebooting if you have not run the update in a long time is probably needed,

closed 2022-10-17 by jgyates